libusb/libusb #681: program crashes in windows_clock_gettime_threaded when operating a camera

gmmybu, 2020-01-22:
I was using libuvc to receive frames from a camera on Win10.  The program is a libuvc demo which receives frames at 30FPS and runs for less than a minute.  It crashes in libusb _windows_clock_gettime_threaded_ occasionally.  The debugger tells me that request->tp is NULL.
```
		case WM_TIMER_REQUEST:
			// Requests to this thread are for hires always
			// Microsoft says that this function always succeeds on XP and later
			// See https://msdn.microsoft.com/en-us/library/windows/desktop/ms644904.aspx
			request = (struct timer_request *)msg.lParam;
			QueryPerformanceCounter(&hires_counter);
			request->tp->tv_sec = (long)(hires_counter.QuadPart / hires_frequency);
			request->tp->tv_nsec = (long)(((hires_counter.QuadPart % hires_frequency) / 1000) * hires_ticks_to_ps);
			if (!SetEvent(request->event))
				usbi_err(NULL, "SetEvent failed for timer request: %s", windows_error_str(0));
			break;
```

I modified _windows_clock_gettime_
```
			tp->tv_sec = 0;
			tp->tv_nsec = 0;

			if (!pPostThreadMessageA(timer_thread_id, WM_TIMER_REQUEST, 0, (LPARAM)&request)) {
				usbi_err(NULL, "PostThreadMessage failed for timer thread: %s", windows_error_str(0));
				CloseHandle(request.event);
				return LIBUSB_ERROR_OTHER;
			}
```

```
			if (r == WAIT_OBJECT_0) {
				if (tp->tv_sec == 0 && tp->tv_nsec == 0) {
					printf("windows_clock_gettime is wrong\n");
				}
				return LIBUSB_SUCCESS;
			}
```

Before _PostThreadMessageA_ tp is assigned to a special value. After _WaitForSingleObject_, if tp is not changed, the program shows some message.

After that it shows
```
windows_clock_gettime is wrong
libusb: error [windows_clock_gettime_threaded] SetEvent failed for timer request: [6] 句柄无效。
```
or shows
```
windows_clock_gettime is wrong
```
and crashes.

I think it's weired but it happens

dickens, 2020-01-22:
This should no longer be an issue with commit eb66a6573d.

closed 2020-01-22 by dickens

gmmybu, 2020-01-23:
> This should no longer be an issue with commit [eb66a65](https://github.com/libusb/libusb/commit/eb66a6573daad7eae2d058c51a7e31bf7911e5ad).

After a lot of trying I found the bug came from the old _ExtendWaitForMultipleObjects_.
```
		SetEvent(pThread[i].wait_event[0]);
		while (pThread[i].bexit == 0); //wait for thread exist;
```
the right code should be
```
		SetEvent(pThread[i].wait_event[0]);
		WaitForSingleObject(pThread[i].thread, INFINITE);
```

It is fixed in [df61c0c](https://github.com/libusb/libusb/commit/df61c0c3a3de070c7f7a714ef8758a248bf97263) Windows: Improve poll abstraction