libusb/libusb #686: Can't communicate with USB device libusb_open() returns LIBUSB_ERROR_ACCESS

keenan-reimer-haptx, 2020-01-27:
I have a libusb device connected to this [USB hub](http://www.ti.com/product/TUSB2077A), running Windows 10 x64, and using the latest version of WinUSB as the driver. It is possible to get my device in a state such that I can no longer interface with it without restarting my computer or reinstalling the WinUSB driver: typically just a power cycle can get me here. If I use my device without this specific USB hub (with another USB hub, chained USB hubs, or no USB hub), it works just fine every time.

Now, another type of USB device I have (one built with entirely different software) that also uses libusb exhibits similar failure modes. Yet another device that doesn't use libusb works every time without fail.

I have tried the other two drivers (libusbk and libusb32) and they also fail, but in different ways.

Here is the code that's failing:
```C++
libusb_device* device;  // Populated above
libusb_device_handle* libusb_handle = nullptr;
int rc = libusb_open(device, &device_handle);
if (rc != LIBUSB_SUCCESS) {
  // Returns LIBUSB_ERROR_ACCESS
}
```

I believe the error is coming from: https://github.com/libusb/libusb/blob/e782eeb2514266f6738e242cdcb18e3ae1ed06fa/libusb/os/windows_winusb.c#L2333

Below is the complete debug output. Notice this error appears a lot:
```
[ 0.120874] [00001f3c] libusb: error [winusbx_claim_interface] could not access interface 0: [122] The data area passed to a system call is too small.
```

As far as I can tell this error has something to do with the libusb <--> WinUSB or WinUSB <--> Windows interfaces. Please advise.
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000007] [00001f3c] libusb: debug [libusb_init] created default context
[ 0.000074] [00001f3c] libusb: debug [libusb_init] libusb v1.0.22.11347
[ 0.000106] [00001f3c] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.000131] [00001f3c] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000139] [00001f3c] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.000153] [00001f3c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.001038] [00001f3c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003553] [00001f3c] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.003562] [00001f3c] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.004233] [00001f3c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.004240] [00001f3c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.004381] [00001f3c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.004989] [00001f3c] libusb: info [windows_init] UsbDk backend is not available
[ 0.005003] [00001f3c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.005032] [00001f3c] libusb: debug [libusb_get_device_list] 
[ 0.010919] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [83]
[ 0.010966] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [84]
[ 0.011007] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [85]
[ 0.011046] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [86]
[ 0.011086] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [87]
[ 0.011126] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [88]
[ 0.011166] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [89]
[ 0.011205] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [8A]
[ 0.011246] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [8B]
[ 0.011291] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [8C]
[ 0.011437] [00001f3c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.011472] [00001f3c] libusb: debug [get_api_type] lower filter driver(s): RtkBtFilter
[ 0.011478] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [9]
[ 0.011701] [00001f3c] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.011726] [00001f3c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.011730] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [20]
[ 0.012010] [00001f3c] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.012036] [00001f3c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.012040] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [50]
[ 0.012204] [00001f3c] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.012230] [00001f3c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.012234] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [61]
[ 0.012330] [00001f3c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.012356] [00001f3c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.012360] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [62]
[ 0.012450] [00001f3c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.012475] [00001f3c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.012480] [00001f3c] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[... 1357 lines cut ...]
[ 0.121394] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.4
[ 0.121398] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.10
[ 0.121401] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.121404] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.3
[ 0.121426] [00001f3c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.121431] [00001f3c] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.121435] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.18
[ 0.121438] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.13
[ 0.121442] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.21
[ 0.121445] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.19
[ 0.121448] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.15
[ 0.121452] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.17
[ 0.121455] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.7
[ 0.121458] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.6
[ 0.121461] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.121465] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.16
[ 0.121468] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.12
[ 0.121471] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.9
[ 0.121474] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.5
[ 0.121477] [00001f3c] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.122329] [00001f3c] libusb: debug [libusb_exit] 
[ 0.122337] [00001f3c] libusb: debug [libusb_exit] destroying default context
[ 0.122345] [00001f3c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.122827] [000010f0] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

dickens, 2020-01-27:
The WinUsb_Initialize() function is provided by the WinUSB DLL.  All libusb does it call it and report back any errors.  Some searching turned up [this](https://community.osr.com/discussion/103920/winusb-initialize-returns-with-error-insufficient-buffer) which suggests that the WinUsb_Initialize() function may be interacting with a buggy device.  Can you verify the validity of the descriptors provided by the device?

keenan-reimer-haptx, 2020-01-28:
Device descriptors look good in usbview.exe (from Windows SDK debugging tools) in all cases; however, I am getting unexpected values for `libusb_device_descriptor::iProduct` and `libusb_device_descriptor::iSerialNumber` only when I cannot communicate with my device.

dickens, 2020-01-28:
Those values should be static.  This further suggests that the issue lies with the device.

keenan-reimer-haptx, 2020-01-28:
Yeah I've been debugging USB packets with Wireshark I'm not longer convinced that either libusb or WinUSB is the source of the problem so I'll close the issue.

I still don't understand how usbview gets good data when the USB packets are clearly malformed, but that's not your problem. Thanks for the help @dickens.

closed 2020-01-28 by keenan-reimer-haptx