libusb/libusb #687: On Android 7 (rooted), libusb_init fails with LIBUSB_ERROR_OTHER

snake984, 2020-01-30:
Hi!

I am trying to use libusb in an Android app, it is embedded in an Android native module and built with cmake. The code only call libusb_init and returns a string to the activity and it systematically fails with LIBUSB_ERROR_OTHER.

The thing is I can get it to work when it is launched from a small command-line program doing only libusb_init that we launch with adb shell. In that case, it returns success.

The device runs Android 7 and is rooted, with Magisk Manager I give my app root permission but it's not like doing "su" in command-line.

I mainly tried with version 20 and 22 (we use mostly version 20 in our other projects).

I already tried many stackoverflow solutions and workarounds but they were either ineffective or not possible to implement (due to rom restrictions for example)

mcuee, 2020-07-25:
Try the method mentioned in the following ticket.
https://github.com/libusb/libusb/issues/623

And as per the ticket, 1.0.22 should work but 1.0.23 does not work.

mcuee, 2020-07-25:
And maybe you can provide more information about the workaround you have tried and what are the problem associated with the workaround.

mcuee, 2020-08-26:
#760 
Not so sure if the above will help. Users may want give the above pull request a try.

mcuee, 2021-02-05:
Close this due to inactivity.

closed 2021-02-05 by mcuee