libusb/libusb #688: libusb fails to retrieve the configuration descriptor from a root hub on Windows

jonasmalacofilho, 2020-02-01:
Tracking down a bug reported to PyUSB I noticed that `libusb_get_config_descriptor` fails to get configuration index 0 from some devices on Windows, even though `bNumConfigurations` is reported as being 1 for them.  This happens when these devices are the USB root hubs.

The problem can be reproduced with `testlibusb.exe -v`:

```
> .\testlibusb.exe -v
Dev (bus 1, device 0): 1B21 - 2142
  Couldn't retrieve descriptors
Dev (bus 2, device 0): 8086 - A2AF
  Couldn't retrieve descriptors
[... other devices ...]
```

_(for additional debug information, check the edited version of this issue on GitHub)_

For reference, this is the corresponding [testlibusb](https://github.com/libusb/libusb/blob/67e6816264e35a1f22b43a78b4be7481652bc51a/examples/testlibusb.c#L203-L210) code:

```c
for (i = 0; i < desc.bNumConfigurations; i++) {
	struct libusb_config_descriptor *config;
	ret = libusb_get_config_descriptor(dev, i, &config);
	if (LIBUSB_SUCCESS != ret) {
		printf("  Couldn't retrieve descriptors\n");
		continue;
	}
```

Additionally, Device Manager shows these root hubs as USB devices, but locates them on the PCI bus, which seems somewhat related (in the sense that they appear to be handled in special way).

---

Downstream issue: pyusb/pyusb#186
libusb version: 67e6816264e35a1f22b43a78b4be7481652bc51a (also confirmed with 1.0.21 and 1.0.23)
OS: Windows 10 Pro 10.0.18363 Build 18363

dickens, 2020-02-11:
This has always been the behavior, so I'm not sure this is truly a bug.  Windows does not provide any mechanism for retrieving a descriptor from a root hub directly, and in fact they are more representative of the host controller itself rather than an actual hub device.  Linux, for example. provides fake descriptors for root hubs.  The Windows implementation exposes the root hub for the completeness of the hierarchy, but it does not represent an actual underlying USB device.  Notice also how the device address of these root hubs is 0, which does not represent a configured and addressable device.

jonasmalacofilho, 2020-02-11:
I agree that not being able to retrieve the descriptor isn't in itself a problem.

But isn't `bNumConfigurations == 1` unexpected in these cases, since there aren't any configurations to retrieve?

> Notice also how the device address of these root hubs is 0, which does not represent a configured and addressable device.

I hadn't noticed that, thanks for pointing it out.

dickens, 2020-02-11:
> But isn't `bNumConfigurations == 1` unexpected in these cases, since there aren't any configurations to retrieve?
Good point.  It might be better to just have this field as 0.

commit d42410b2 (2020-08-17): Windows: Improve how root hubs are reported

Microsoft provides very little information about the actual root hub
characterstics beyond the number of ports. Determining the maximum
supported speed of the root hub is not directly possible but instead
requires that a device matching the highest speed of the root hub is
connected. Additionally, Windows 8 or later is required for _any_
successful detection of SuperSpeed devices.

One other inconvenience is that there are no descriptors exposed for
root hubs. This does not fit well with the structure of the library
because root hubs are considered  [... cut]
closed 2020-08-17 by dickens