libusb/libusb #691: libusb_open failed on Win10 Dell XPS 15 laptop

garfungiloops, 2020-02-10:
Windows 10, Dell XPS 15 (9550), different USB sticks

1. libusb_get_device_descriptor - works fine!

2. libusb_open - fails with the code "LIBUSB_ERROR_NO_DEVICE = -4" when usb-stick is connected to the Dell Thunderbolt TB16 dock.

3. libusb_open - fails with the code "LIBUSB_ERROR_NOT_SUPPORTED = -12"  when the same usb device is connected to the Dell XPS 15 laptop usb directly

dickens, 2020-02-10:
Please provide a debug log.

mcuee, 2020-03-10:
Hmm, is this USB Flash Drive?

garfungiloops, 2020-03-11:
yes this is USB flash.
But I have already realized that the problem was perhaps that I had no WinUSB driver installed.
For my project I need a solution that works without any extra drivers.
So I decided to switch to libusbp (which works in my case).
I think you can close the issue.

closed 2020-06-02 by mcuee