libusb/libusb #692: Cannot cross compile for windows

davidsteed, 2020-02-10:
I used to be able to cross compile libusb for windows, but this is no longer working.

./configure --host=i686-pc-mingw32
make

and I get

Making all in libusb
  CC       libusb_1_0_la-core.lo
In file included from core.c:23:
In file included from ./libusbi.h:82:
./os/poll_windows.h:47:2: error: unknown type name 'OVERLAPPED'
        OVERLAPPED *overlapped;         // what will report our I/O status
        ^
In file included from core.c:23:
In file included from ./libusbi.h:89:
./os/threads_windows.h:27:9: error: unknown type name 'LONG'
typedef LONG usbi_mutex_static_t;
        ^
./os/threads_windows.h:30:9: error: implicit declaration of function 'InterlockedExchange' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        while (InterlockedExchange(mutex, 1L) == 1L)
               ^
./os/threads_windows.h:30:9: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:31:3: error: implicit declaration of function 'SleepEx' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
                SleepEx(0, TRUE);
                ^
./os/threads_windows.h:31:3: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:31:14: error: use of undeclared identifier 'TRUE'
                SleepEx(0, TRUE);
                           ^
./os/threads_windows.h:35:2: error: implicit declaration of function 'InterlockedExchange' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        InterlockedExchange(mutex, 0L);
        ^
./os/threads_windows.h:38:9: error: unknown type name 'CRITICAL_SECTION'
typedef CRITICAL_SECTION usbi_mutex_t;
        ^
./os/threads_windows.h:41:2: error: implicit declaration of function 'InitializeCriticalSection' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        InitializeCriticalSection(mutex);
        ^
./os/threads_windows.h:41:2: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:46:2: error: implicit declaration of function 'EnterCriticalSection' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        EnterCriticalSection(mutex);
        ^
./os/threads_windows.h:46:2: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:50:2: error: implicit declaration of function 'LeaveCriticalSection' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        LeaveCriticalSection(mutex);
        ^
./os/threads_windows.h:50:2: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:54:10: error: implicit declaration of function 'TryEnterCriticalSection' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        return !TryEnterCriticalSection(mutex);
                ^
./os/threads_windows.h:54:10: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:58:2: error: implicit declaration of function 'DeleteCriticalSection' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        DeleteCriticalSection(mutex);
        ^
./os/threads_windows.h:58:2: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:76:9: error: unknown type name 'CONDITION_VARIABLE'
typedef CONDITION_VARIABLE usbi_cond_t;
        ^
./os/threads_windows.h:79:2: error: implicit declaration of function 'InitializeConditionVariable' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        InitializeConditionVariable(cond);
        ^
./os/threads_windows.h:79:2: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:83:8: error: implicit declaration of function 'SleepConditionVariableCS' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        (void)SleepConditionVariableCS(cond, mutex, INFINITE);
              ^
./os/threads_windows.h:83:8: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:83:46: error: use of undeclared identifier 'INFINITE'
        (void)SleepConditionVariableCS(cond, mutex, INFINITE);
                                                    ^
./os/threads_windows.h:89:2: error: implicit declaration of function 'WakeAllConditionVariable' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        WakeAllConditionVariable(cond);
        ^
./os/threads_windows.h:89:2: warning: this function declaration is not a prototype [-Wstrict-prototypes]
./os/threads_windows.h:96:9: error: unknown type name 'DWORD'
typedef DWORD usbi_tls_key_t;
        ^
./os/threads_windows.h:99:9: error: implicit declaration of function 'TlsAlloc' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        *key = TlsAlloc();
               ^
./os/threads_windows.h:99:9: warning: this function declaration is not a prototype [-Wstrict-prototypes]
fatal error: too many errors emitted, stopping now [-ferror-limit=]
11 warnings and 20 errors generated.
make[2]: *** [libusb_1_0_la-core.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

dickens, 2020-02-10:
You didn't specify what distribution you are using and what the version of the toolchain is.  I suspect that the toolchain doesn't provide a definition for `_WIN32_WINNT_VISTA`.

davidsteed, 2020-02-11:
I get the same result on Darwin and Linux (4.19.76-linuxkit).  What do I do to fix this?

davidsteed, 2020-02-11:
Think I have solved the problem.  It works if I use host=x86_64-w64-mingw32

dickens, 2020-02-23:
Since you have a solution, I'm going to close this issue.  When I cross-compile, I specify either "i686-w64-mingw32" or "x86_64-w64-mingw32".

closed 2020-02-23 by dickens