libusb/libusb #693: macOS Catalina: wrong detection of parent devices

IgorAnohin, 2020-02-14:
I'm working with `libusb` on my laptop. I need to get the whole USB location of my device. Device (_0781:5597_) is connected to MacBook over the USB hub. So, I expect that it will have more than one USB port in its location. But when I run `examples/listdevs` I got the following result:
```shell
➜  libusb git:(a36862b) ✗ ./examples/listdevs
05ac:027b (bus 128, device 5) path: 5
-> 0781:5597 (bus 20, device 14) path: 4
05e3:0608 (bus 20, device 13) path: 4
05e3:0608 (bus 20, device 12) path: 2
05e3:0751 (bus 20, device 6) path: 1
2109:0102 (bus 20, device 3) path: 4
1a40:0801 (bus 20, device 2) path: 2
0bda:8153 (bus 1, device 2) path: 3
2109:2817 (bus 20, device 1) path: 7
2109:0817 (bus 1, device 1) path: 3
```

If I checkout to this changes https://github.com/libusb/libusb/pull/282, I'll get the following output:
```shell
➜  libusb git:(871eb29) ✗ ./examples/listdevs
05ac:027b (bus 128, device 5) path: 5
-> 0781:5597 (bus 20, device 14) path: 7.2.2.4.4
05e3:0608 (bus 20, device 13) path: 7.2.2.4
05e3:0608 (bus 20, device 12) path: 7.2.2
05e3:0751 (bus 20, device 6) path: 7.2.1
2109:0102 (bus 20, device 3) path: 7.2.4
1a40:0801 (bus 20, device 2) path: 7.2
0bda:8153 (bus 1, device 2) path: 3.3
2109:2817 (bus 20, device 1) path: 7
2109:0817 (bus 1, device 1) path: 3
```
The second path looks much better, but I have a question. Does the Usb ports path should be equal the LocationID of the device? 
The second output returned 7.2.2.4.4 ports path, but device LocationID is `14222440` (2.2.2.4.4 is the ports path, right?)

Here is the output of `ioreg` for my device:
[image]

I'm a newbie in MacOs and don't really understand what is going wrong. Could you help me, please?

Output of `uname -a` on my machine:
Darwin Igors-MacBook-Pro.local 19.3.0 Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64 x86_64

IgorAnohin, 2020-02-14:
If I change `kIOUSBPlane` to `kIOServicePlane` in `get_device_parent_sessionID` function on master, I'll get the same output as in #282 

[image]

But it still not fixes the first port, that is still `7`, but I think that it should be `2`
```bash
➜  libusb git:(master) ✗ ./examples/listdevs
-> 0781:5597 (bus 20, device 11) path: 7.2.2.4.4
05e3:0608 (bus 20, device 10) path: 7.2.2.4
05e3:0608 (bus 20, device 9) path: 7.2.2
2109:0102 (bus 20, device 4) path: 7.2.4
05e3:0751 (bus 20, device 3) path: 7.2.1
1a40:0801 (bus 20, device 2) path: 7.2
0bda:8153 (bus 1, device 2) path: 3.3
2109:2817 (bus 20, device 1) path: 7
2109:0817 (bus 1, device 1) path: 3
05ac:027b (bus 128, device 5) path: 5
```

IgorAnohin, 2020-03-10:
@dickens , Could you help me please with this issue? I still get no response from you.

dickens, 2020-03-10:
> @dickens , Could you help me please with this issue? I still get no response from you.

Unfortunately I'm not the Darwin maintainer and have limited experience on that platform, so I can't say much on this topic.  However based on the output, every device on bus 20 has the same leading port number of 7, which seems reasonable if device 2109:2817 (bus 20, device 1) is the parent of them all.

seanm, 2020-03-20:
@hjelmn might be able to help...

IgorAnohin, 2020-04-08:
Ok, I think I found the solution.

The reason for that behavior is `kIOUSBDeviceClassName`. In previous releases, `libusb` used `"IOUSBDevice"`. But when I checked `kIOUSBDeviceClassName`, it had `"IOUSBHostDevice"` value.

044a7eeb7566dacade75b31476b46dcc6bfe7efc has changed explicitly used `IOUSBHostDevice` to `IOUSBDevice` that was defined as `kIOUSBDeviceClassName`, because `libusb` has been working incorrectly with `IOUSBHostDevice`
`
This change fixes the incorrect behavior and `libsub` works as expected:
```shell
➜  libusb git:(master) ✗ git diff
diff --git a/libusb/os/darwin_usb.c b/libusb/os/darwin_usb.c
index 7fdf88f..39a6153 100644
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -87,7 +87,7 @@ static CFRunLoopSourceRef libusb_darwin_acfls = NULL; /* shutdown signal for eve

 static usbi_mutex_t darwin_cached_devices_lock = PTHREAD_MUTEX_INITIALIZER;
 static struct list_head darwin_cached_devices = {&darwin_cached_devices, &darwin_cached_devices};
-static const char *darwin_device_class = kIOUSBDeviceClassName;
+static const char *darwin_device_class = "IOUSBDevice";

 #define DARWIN_CACHED_DEVICE(a) ((struct darwin_cached_device *) (((struct darwin_device_priv *)((a)->os_priv))->dev))
```

Another solution is to add `AC_DEFINE(MAC_OS_X_VERSION_MIN_REQUIRED, 101400, [Custom build version])` to `configure.ac`

May someone approve it?

IgorAnohin, 2020-04-08:
@hjelmn . could you take a look, please?
@dickens , @seanm , Do you know someone who could also help?

seanm, 2020-04-08:
@hjelmn is the expert I believe.

But setting MAC_OS_X_VERSION_MIN_REQUIRED to 10.14 is no good, older version of macOS still need to be supported.

mvp, 2020-04-08:
> But it still not fixes the first port, that is still `7`, but I think that it should be `2`

I concur with @dickens, port 7 seems correct here to me.
My patch should work for all cases https://github.com/libusb/libusb/pull/711

IgorAnohin, 2020-04-08:
My bad. I did not describe what exactly is happening.

`IOUSBLIb.h` on my PC has the following code.
[image]

By default `MAC_OS_X_VERSION_MIN_REQUIRED` on my PC is `10500`. That's why `libusb` uses `kIOUSBHostDeviceClassName`(`"IOUSBHostDevice"`). https://github.com/libusb/libusb/commit/044a7eeb7566dacade75b31476b46dcc6bfe7efc commit described, that `libusb` unable to work with that kind of devices.

So, `MAC_OS_X_VERSION_MIN_REQUIRED= 101400` just returns the _old_ behavior and _old_ `kIOUSBDeviceClassName`. @seanm , I agree, that it isn't clear and might be confusing. Custom define isn't a very good solution.

I suppose, that `darwin_device_class = "IOUSBDevice"` is ok. All MacOs version should support it. And `libusb` has been working with it from the very beginning

Should I write a mail directly to  @hjelmn or just mention him here.

IgorAnohin, 2020-04-09:
> > But it still not fixes the first port, that is still `7`, but I think that it should be `2`
> 
> I concur with @dickens, port 7 seems correct here to me.
> My patch should work for all cases #711

The answer is that version, built on Mojave, returns the different output
Here is the screen, that I compare #711 patch and `libusb-1.0.23.mojave`:

[image]

#711 patch will not work correctly on Mojave and older macOS versions.

IgorAnohin, 2020-04-09:
I created PR #712

commit bb377309 (2020-06-12): os/darwin: use IOUSBDevice as darwin_device_class explicitly

kIOUSBDeviceClassName define from IOUSBLib.h file was changed from
IOUSBDevice to IOUSBHostDevice in macOS Catalina.

In previous macOS versions, it was always defined as IOUSBDevice:
```
 #define kIOUSBDeviceClassName		 "IOUSBDevice"
```
In macOS Catalina it looks as follows:
```
 #if MAC_OS_X_VERSION_MIN_REQUIRED <= MAC_OS_X_VERSION_10_14
 #define kIOUSBDeviceClassName           "IOUSBDevice"
 #define kIOUSBInterfaceClassName        "IOUSBInterface"
 #else
 #define kIOUSBDeviceClassName           kIOUSBHostDeviceClassName
 #define [... cut]
closed 2020-06-12 by hjelmn