libusb/libusb #694: Can't use usb webcam in Windows remote host (Windows 10) from ubuntu vm

myjimmy, 2020-02-18:
**Describe the bug**
I installed Ubuntu 18.04 64 bit virtual machine in VMware 15.0.4 and Windows 10 64bit.
Also, I installed Windows 10 64bit virtual machine as a remote host.
I plugged the USB webcam on my computer and connected it to ubuntu virtual machine.
[image]

In ubuntu vm, I run the following command:
`$ sudo xfreerdp /v:192.168.77.134 /u:ptc /dvc:urbdrc,dev:1908:2311`
192.168.77.134 is Windows 10 virtual machine.
[image]

**Error**
Then, I get the following error:
```
... ... ...
VID: 0x1908, PID: 0x2311
libusb_set_configuration: error LIBUSB_ERROR_OTHER [-99]
Set interface altsetting get error num -99
libusb_control_transfer LIBUSB_ERROR_PIPE [-9]
... ... ...
```

[image]

I can see the webcam item on the camera page of Chrome web browser in about 20s.
[image]

But I can't use this device in the webRTC sample pages such as the following pages:
https://simpl.info/getusermedia/sources/
https://webrtc.github.io/samples/src/content/getusermedia/gum/
[image]

** Application details
* Version of FreeRDP : latest version of master branch
* Command line used : sudo xfreerdp /v:192.168.77.134 /u:ptc /dvc:urbdrc,dev:1908:2311

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04 64bit (libusb 1.0.21)

Thanks for any help.

diabolo38, 2020-02-19:
I had functional issue using several devices (most usb3 bulk ) on VM under Window  (VMware and Virtual Box)
Device detection works so it's fine for device detection and  driver iinstall test  but i can't run most functional test they fail  while they  run just fine on real machines.
So it is likely a VM issue not  libusb one

mcuee, 2020-03-10:
I think you have to ask the VM vendor. Traditionally there are multiple issues with VM's USB driver.

closed 2020-03-10 by mcuee