libusb/libusb #695: Can't compile due to /ld.exe: cannot find -libusb or /ld.exe: cannot find -lusb? You're not alone.

theshawnz, 2020-02-18:
The instructions for adding this library to your Visual Studio Code (VSCode) project can be quite skinny. Some people mention "add -lusb" or "add -libusb-1.0" to your tasks.json to make it work; well this may not be specific enough for some. 

for reference: 

Windows 10, MinGW64 and VSCode. 

Try removing both of those commands above and adding the following to your arguments section where I have both mingw 32 and 64 installed on the same machine within my MinGw folder): 

                "-LI",
                "<path to MinGW Lib files>\\libusb-1.0.a"

your compiler command looking something like this: gcc -Wall -o ${fileDirname}\\${fileBasenameNoExtension}.exe ${file} -LI C:\\MinGW\\mingw64\\lib\\libusb-1.0.a

I was able to compile the included examples testlibusb.c, listdevs.c, etc after doing so.

dickens, 2020-02-23:
Good information, thanks.  What do you suggest be done within libusb?  Do you have an improvement to the Wiki or documentation to contribute?

mutl3y, 2020-02-23:
Was having the same sort of issues in Goland, thanks to this post I found I needed to modify the lib-usb-1.0.pc file

rename without the .txt
[libusb-1.0.pc.txt](https://github.com/libusb/libusb/files/4242093/libusb-1.0.pc.txt)

prefix=/mingw64
exec_prefix=${prefix}
libdir=${exec_prefix}/usr/lib
includedir=${prefix}/include

Name: libusb-1.0
Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace
Version: 1.0.23
Libs: -L${libdir}\libusb-1.0
Libs.private: 
Cflags: -I${includedir}/libusb-1.0

theshawnz, 2020-02-28:
> Good information, thanks. What do you suggest be done within libusb? Do you have an improvement to the Wiki or documentation to contribute?

Honestly, i am still quite new to using GCC and programming in general so it's hard to say...

I was expecting for an approach similar to adding 3rd party includes in Visual Studio where you point to the folder for the includes, reference a Lib file and that's it. 

After looking at it further; the Readme for MinGW/cygwin would benefit greatly (for the novice user ) mentioning differences between static and dynamic lib files (although one could always just google) and how to (verbatim) link the libusb-1.0.a files in the task Json (commented in my original post above).   Other than that; greatly appreciative for libusb. :-)

closed 2020-02-28 by theshawnz

Gustavomurta, 2021-04-11:
Windows 10 X64

- Install msys2     ([https://www.msys2.org/])

- Run msys2 and run command:

**pacman -S mingw-w64-x86_64-libusb**
Reference : [https://packages.msys2.org/package/mingw-w64-x86_64-libusb](url)

- check version of libusb that installed

pacman -Q | grep libusb
mingw-w64-x86_64-libusb 1.0.24-2

mcuee, 2021-04-12:
Yes if you use MSYS, libusb-1.0.pc is already there. It is recommended to use pkg-config for libusb if you use MinGW GCC.

https://packages.msys2.org/package/mingw-w64-x86_64-libusb
/mingw64/bin/libusb-1.0.dll
/mingw64/include/libusb-1.0/libusb.h
/mingw64/lib/libusb-1.0.a
/mingw64/lib/libusb-1.0.dll.a
/mingw64/lib/pkgconfig/libusb-1.0.pc

mcuee, 2021-04-12:
https://github.com/libusb/libusb/wiki/Windows

Updated the wiki page.