libusb/libusb #696: functionfs gadget multiple endpoints

nandra, 2020-02-19:
Hi,

I'm playing with functionfs and use configfs + functionfs + ffs-test from kernel to test connection between my device and host. ffs-test example provide 1 configuration with 1 interface and with 2 bulk endpoints. 

I'm writing application which will write/read data to usb device but can be accessed from multiple application. So as read in some other thread I'm planning to write kind of gatekeeper which will serialize data to endpoint and notify reader about received data.

I was thinking about other possibility to extend gadget to provide more endpoints (like 2). I was able to extend ffs-test and also I can send/receive data over original endpoints but not over new created one (I've run 2 instances of same application which basically create transfers for 2 different endpoints) but when running second application I got this error:
```
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
```  
I'm not USB expert but multiple endpoint should be possible (aybe I need to put them to other interface?). Thanks for any pointers and advice.

dickens, 2020-02-23:
Two different processes on Linux (generally) cannot use the same interface.  There are exceptions if the device is opened and the process forks, but I'll assume that's not happening here.  The value of errno 16 is EBUSY, and the value of errno 2 is ENOENT.  The former is self-explanatory, the latter means the endpoint for the URB wasn't found.

mcuee, 2020-03-10:
Closing this as this is related to gadget device (device side) and not libusb (host side).

closed 2020-03-10 by mcuee