libusb/libusb #698: Invalid pid vid on Raspberry Pi 4

jhihn, 2020-02-21:
I have some code that I've been using on MacOS and Linux(x86). I recently moved it to a Raspberry Pi 4 and the VID:PID values are all over the place.

Inserting a known device with a known PID:VID only works the first time 50% of the time. Subsequent insertions are always wrong.
I contually get 
libusb: error [udev_hotplug_event] ignoring udev action bind

This happens directly on the Pi USB3 port, and two hubs by different manufacturers. 
My application logs the insert/remove as:
insert F0E0:A7E0 @ 1-1 (01)
remove CFE8:0143 @ 1-1

However the linux kernel sees the correct VID:PID

dickens, 2020-02-23:
Can you provide a debug log?

jhihn, 2020-02-24:
Error in dependent library.

closed 2020-02-24 by jhihn