libusb/libusb #702: Integrating with OSS-Fuzz

Google-Autofuzz, 2020-03-03:
Greetings libusb developers and contributors,

We’re reaching out because your project is an important part of the open source ecosystem, and we’d like to invite you to integrate with our [fuzzing](https://www.owasp.org/index.php/Fuzzing) service, [OSS-Fuzz]( https://opensource.googleblog.com/2016/12/announcing-oss-fuzz-continuous-fuzzing.html ). OSS-Fuzz is a free fuzzing infrastructure you can use to identify security vulnerabilities and stability bugs in your project. OSS-Fuzz will:

- Continuously run at scale all the fuzzers you write.
- Alert you when it finds issues.
- Automatically close issues after they’ve been fixed by a commit.

Many widely used [open source projects]( https://github.com/google/oss-fuzz/tree/master/projects ) like OpenSSL, FFmpeg, LibreOffice, and ImageMagick are fuzzing via OSS-Fuzz, which helps them find and remediate [critical issues]( https://bugs.chromium.org/p/oss-fuzz/issues/list?can=1&q=status%3AFixed%2CVerified+Type%3ABug%2CBug-Security+-component%3AInfra+ ). 

Even though typical integrations can be done in < 100 LoC, we have a [reward program]( https://www.google.com/about/appsecurity/patch-rewards/ ) in place which aims to recognize folks who are not just contributing to open source, but are also working hard to make it more secure.

We want to stress that anyone who meets the eligibility criteria and integrates a project with OSS-Fuzz is eligible for a reward.

To help you getting started, we can attach our internal fuzzer for your project that you are welcome to use directly, or to use it as a starting point.

If you're not interested in integrating with OSS-Fuzz, it would be helpful for us to understand why—lack of interest, lack of time, or something else—so we can better support projects like yours in the future.

If we’ve missed your question in our [FAQ]( https://google.github.io/oss-fuzz/faq/ ), feel free to reply or reach out to us at oss-fuzz-outreach@googlegroups.com.

Thanks!

Tommy
OSS-Fuzz Team

hjelmn, 2020-03-03:
Might be worthwhile to do this. I am hoping to find the time to get some fuzzing tests up internally some time next quarter. I think there is an internal Google bug tracking this but I do no know who the current owner is.

Google-Autofuzz, 2020-04-27:
Do you have a google email address so that I can use as the [primary contact](https://google.github.io/oss-fuzz/getting-started/new-project-guide/#primary)? It is needed so that after OSS-Fuzz can email you the statistics and crash reports after some fuzzing tests are done.

dickens, 2020-04-27:
I am fine with my address being used.  We don't have an e-mail address for the project, but we could also make one if that will work better.

hjelmn, 2020-04-28:
@Google-Autofuzz Mine is hjelmn@.

@dickens We probably should have a project email. Not sure what the best way to do that is. I can receive the info at from my Google email and forward them to libusb-devel.

Google-Autofuzz, 2020-04-30:
Thanks for the emails. Just wondering if libsub has any fuzzers that we can use as a starting point. If not, no worries and we can start one.

dickens, 2020-04-30:
There are no existing fuzzers within the project, and I haven't seen any mention of outside ones.

Google-Autofuzz, 2020-05-04:
I was able to make build and install libusb. But I am running into building/linking issues with the libusb library _./libusb/.libs/libusb-1.0.a_. I was thinking if any of you can help me figure the problem behind this. I can also email the files to you too if you need them. Thanks :) 

Here is the build code:
````
# build project
./autogen.sh
./configure
make -j$(nproc) all

# build fuzzers
for fuzzers in $(find $SRC -name '*_fuzzer.cc'); do
  fuzz_basename=$(basename -s .cc $fuzzers)
  $CXX $CXXFLAGS -std=c++11 -I. \
  $fuzzers $LIB_FUZZING_ENGINE ./libusb/.libs/libusb-1.0.a \
  -o $OUT/$fuzz_basename
done
````
I got an error like the following:
````
./libusb/.libs/libusb-1.0.a(linux_udev.o): In function `linux_udev_start_event_monitor':
/src/libusb/libusb/os/linux_udev.c:49: undefined reference to `udev_new'
/src/libusb/libusb/os/linux_udev.c:55: undefined reference to `udev_monitor_new_from_netlink'
/src/libusb/libusb/os/linux_udev.c:61: undefined reference to `udev_monitor_filter_add_match_subsystem_devtype'
/src/libusb/libusb/os/linux_udev.c:67: undefined reference to `udev_monitor_enable_receiving'
/src/libusb/libusb/os/linux_udev.c:121: undefined reference to `udev_monitor_unref'
````

dickens, 2020-05-04:
If you are statically linking to libusb, you'll need to specify libusb's library dependencies.
Adding `-pthread -ludev` to the compilation command should do the trick.

Google-Autofuzz, 2020-05-08:
Libusb has been integrated to OSS-Fuzz as shown above. You can take a look and make changes as you like. Please feel free to add more fuzzer or edit the fuzzer I wrote for the initial integration. Thanks. Feel free to close this issue.

closed 2021-03-12 by mcuee