libusb/libusb #706: Communicate real-time data between Android and Windows?

WimVriend, 2020-03-22:
Hello experts,

I have created an App on Android and now I would like to add USB support to it. Via several routes I ended up here, but I never find a clear answer to my question: can I use libusb (on both Android and Windows) to send data from Android to Windows?

If so: are there any examples that show how this can be implemented?

dickens, 2020-03-23:
libusb is a _host_ side library for interacting with USB _devices_.  What you are describing is something running on the _device_ side to help interact with a Windows _host_, in which case libusb is of no use to on the Android side.

I'm going to close this since this isn't an issue.  For technical support questions please use the libusb mailing list.

closed 2020-03-23 by dickens

WimVriend, 2020-03-25:
Hello Chris,

 

I thought there might also be an Android version of libusb, so the App could communicate via the USB port of the device, with the USB port on the host.

Is that such a strange idea?

 

Kind regards,

Wim

mcuee, 2020-08-11:
This is a question, better to ask in libusb-devel mailing list. 

If you want to use the following, yes, it is possible to write a Windows host app with libusb to communicate with Android device if your Android device act as a USB slave device.
Windows host with libusb based Windows app <--> Android slave

On the other hand, now there are Windows devices (not many, mainly ARM based) which can act as a slave USB device. In that case, Android can act as the host. Then you can write an Android app with libusb.
Android host with libusb based Android app <--> Windows slave.

You can see libusb is always on the host side.

mcuee, 2020-08-11:
If you have further questions, please ask in libusb-devel mailing list and not here.