libusb/libusb #707: libusb_get_port_numbers() is broken on MacOS Catalina

mvp, 2020-03-27:
This commit https://github.com/libusb/libusb/commit/044a7eeb7566dacad seems to have broken support for `libusb_get_port_numbers` on MacOS Catalina (10.15).
For example, on my Mac, building libusb from master branch and running `examples/listdevs` gives on my system:
```
0930:6545 (bus 20, device 6) path: 1
05ac:1460 (bus 20, device 5) path: 2
```
which is incorrect - hub `11` is skipped in port path.

Compiling libusb-1.0.22 (or anything before aforementioned commit) gives following output:
```
0930:6545 (bus 20, device 6) path: 11.1
05ac:1460 (bus 20, device 5) path: 11.2
```
which is correct.

Applying this patch fixes this problem, and everything works from master tree:

```
diff --git a/libusb/os/darwin_usb.c b/libusb/os/darwin_usb.c
index 4639a7e..d488639 100644
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -979,7 +979,7 @@ static bool get_device_parent_sessionID(io_service_t service, UInt64 *parent_ses

   /* Walk up the tree in the IOService plane until we find a parent that has a sessionID */
   parent = service;
-  while((kresult = IORegistryEntryGetParentEntry (parent, kIOUSBPlane, &parent)) == kIOReturnSuccess) {
+  while((kresult = IORegistryEntryGetParentEntry (parent, kIOServicePlane, &parent)) == kIOReturnSuccess) {
     if (get_ioregistry_value_number (parent, CFSTR("sessionID"), kCFNumberSInt64Type, parent_sessionID)) {
         /* Success */
         return true;
```
This breaks my utility `uhubctl` on MacOS, which relies on `libusb_get_port_numbers` to work.
Please let me know if I should create pull request for this.
Thanks!

mvp, 2020-03-27:
I have documented this in e3734e0. Please use workaround mentioned in README.md, it should work for you now.

IgorAnohin, 2020-04-08:
Hello @mvp ! I opened the same issue #693 .
I tried changing `kIOUSBPlane` to `kIOServicePlane`, but it does not work in all cases. If I add a USB hub, `libusb` shows the wrong first port.
Have you tried changing `MAC_OS_X_VERSION_MIN_REQUIRED` or `darwin_device_class`? (Which `MAC_OS_X_VERSION_MIN_REQUIRED` are you using during the build)?

I'm new to `MacOs`, but I think this might be the solution for everyone.

mvp, 2020-04-08:
Well, libusb-1.0.22 is using kIOServicePlane: https://github.com/libusb/libusb/commit/044a7eeb7566dacad#r38287626,
and it works perfectly in all cases that I tried.
No, I did not change anything else - why would I want to? We want it to work for any MacOS version.

> I tried changing kIOUSBPlane to kIOServicePlane, but it does not work in all cases. If I add a USB hub, libusb shows the wrong first port.

Why do you think it shows wrong first port?

IgorAnohin, 2020-04-08:
> Why do you think it shows wrong first port?

Because the [Pyserial](https://github.com/pyserial/pyserial/blob/master/serial/tools/list_ports_osx.py#L218) (link to the usage) project uses `LocationID` to get the device ports. When I use usb-hub, `kIOServicePlane` returns a different value for the first port.

> No, I did not change anything else - why would I want to? 

Because I think that you have the same version of `IOUSBLib.h` as mine. In this case, your `libusb` uses `"IOUSBHostDevice"` as `kIOUSBDeviceClassName` by default. 
https://github.com/libusb/libusb/commit/044a7eeb7566dacad commit says, that `libusb` unable to work with `"IOUSBHostDevice"`.

You may check the class, that you are using, by printing `darwin_device_class` value.

> We want it to work for any MacOS version.

In your case, `MAC_OS_X_VERSION_MIN_REQUIRED=10400` should return the _old_ behavior.

mvp, 2020-04-08:
> usb-hub, kIOServicePlane returns a different value for the first port. (I have "14222440" locationID for "20-7.2.2.4.4" libusb value)

locationID is just internal MacOS id which should have no meaning as to which port chain was actually used.
Instead, you should use path in following format: b-x.y.z, where b is USB bus number, and x, y, z... are port numbers for all hubs in chain, starting from root hub for a given USB bus (basically what is returned by `libusb_get_port_numbers()`). This format is used by Linux kernel, and I also use this in my tool uhubctl. This is how I retrieve device path in C:

https://github.com/mvp/uhubctl/blob/fa524cf7dffdc0a8967815141e82d962cb14b61c/uhubctl.c#L399-L409

IgorAnohin, 2020-04-08:
I checked `libusb-1.0.22` output. Thank you for the link to `libusb-1.0.22`. I found it in https://github.com/mvp/uhubctl repo.

Output after your fix:
```bash
1005:b113 (bus 20, device 10) path: 6.1.4.4
```
Output from `libusb-1.0.22`:
```bash
1005:b113 (bus 20, device 10) path: 1.1.4.4
```
As you can see, the outputs are different. Could you please try to run the same test with USB-hub?
I'm using the hub to convert type-C to USB (it has 3 USB ports + hdmi) + DLink hub. That's why my port path is pretty long

mvp, 2020-04-08:
* libusb-1.0.22, `examples/listdevs` outputs on my system:
```
0930:6545 (bus 20, device 18) path: 5.1
05ac:1460 (bus 20, device 4) path: 5.2
05ac:100f (bus 20, device 2) path: 5
```
* libusb-1.0.23 (or master branch):
```
0930:6545 (bus 20, device 18) path: 1
05ac:1460 (bus 20, device 4) path: 2
05ac:100f (bus 20, device 2) path: 5
```
* libusb master branch + my patch:
```
0930:6545 (bus 20, device 18) path: 5.1
05ac:1460 (bus 20, device 4) path: 5.2
05ac:100f (bus 20, device 2) path: 5
```

mvp, 2020-04-08:
Note that my tool uhubctl _only_ works with USB hubs. it works correctly on libusb-1.0.22 or on master + my patch.

IgorAnohin, 2020-04-08:
Could you append 1 more USB hub, please?

IgorAnohin, 2020-04-08:
Here is my setup.
[image]

And my output:
[image]

mvp, 2020-04-08:
Added intermediate hub:
* libusb-1.0.22:
```
058f:6364 (bus 20, device 21) path: 5.1.3
0930:6545 (bus 20, device 20) path: 5.1.2
058f:6254 (bus 20, device 19) path: 5.1
05ac:1460 (bus 20, device 4) path: 5.2
05ac:100f (bus 20, device 2) path: 5
```
* libusb master branch (1.0.23):
```
058f:6364 (bus 20, device 21) path: 3
0930:6545 (bus 20, device 20) path: 2
058f:6254 (bus 20, device 19) path: 1
05ac:1460 (bus 20, device 4) path: 2
05ac:100f (bus 20, device 2) path: 5
```
* libusb master + my patch:
```
058f:6364 (bus 20, device 21) path: 5.1.3
0930:6545 (bus 20, device 20) path: 5.1.2
058f:6254 (bus 20, device 19) path: 5.1
05ac:1460 (bus 20, device 4) path: 5.2
05ac:100f (bus 20, device 2) path: 5
```

IgorAnohin, 2020-04-08:
I installed `brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/5314f1d/Formula/libusb.rb` to the system. So, `examples/.libs/listdevs` uses it. `examples/listdevs` uses  compiled version.

mvp, 2020-04-08:
Using libusb from `brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/5314f1d/Formula/libusb.rb`
gives:
```
058f:6364 (bus 20, device 21) path: 4.1.3
0930:6545 (bus 20, device 20) path: 4.1.2
058f:6254 (bus 20, device 19) path: 4.1
05ac:1460 (bus 20, device 4) path: 4.2
05ac:100f (bus 20, device 2) path: 4
```
I have no idea why first port is different

IgorAnohin, 2020-04-08:
Sorry for the confusion. Instead of building `libusb-1.0.22` I used `brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/5314f1d/Formula/libusb.rb`.

That's why you got the same output for `libusb-1.0.22` and `master+fix` branch.
The reason for the different outputs is `darwin_device_class` var. I tied to explain it earlier.

IgorAnohin, 2020-04-08:
Please, look at this file.
[image]
I think that `https://raw.githubusercontent.com/Homebrew/homebrew-core/5314f1d/Formula/libusb.rb` version uses `darwin_device_class=="IOUSBDevice`, but you PC (and my) uses `darwin_device_class=="IOUSBHostDevice"`

mvp, 2020-04-08:
> I think that `https://raw.githubusercontent.com/Homebrew/homebrew-core/5314f1d/Formula/libusb.rb` version uses `darwin_device_class=="IOUSBDevice"`, but you PC (and my) uses `darwin_device_class=="IOUSBHostDevice"`

Generated binaries should work on any version of OS, and should not depend on whatever headers happen to be installed on your particular system - that may change as you upgrade MacOS, and should not break already installed utilities.

At any rate, my patch does not make it any worse - it simply returns back libusb-1.0.22 behavior, which at least gives some working port path.

IgorAnohin, 2020-04-08:
> my patch does not make it any worse - it simply returns back libusb-1.0.22 behavior, which at least gives some working port path.

I do not agree with you. Look, If you change it, the code will not work correctly with `IOUSBDevice` system. It will work with only `IOUSBHostDevice`. https://github.com/libusb/libusb/commit/044a7eeb7566dacad commit says:
_"The problem is that using the new class[IOUSBHostDevice] misses some devices in more recent versions of macOS."_ This commit reverts all changes to return to `IOUSBDevice` system.

The reason for incorrect behavior - `IOUSBHostDevice`, that was picked during the build by default. Your patch fixes incorrect behavior for `IOUSBHostDevice`, but `libusb` tries to work with `IOUSBDevice` system (If I understand all right). To support MacOs Catalina, we should explicitly pick `IOUSBDevice` system. What do you think about that?

mvp, 2020-04-08:
Well, in current 1.0.23 implementation `libusb_get_port_numbers()` simply does not work at all.
In 1.0.22 it kinda works - gives more or less legit port path.

I agree that there should be working `libusb_get_port_numbers()` that gives out actual port numbers for all hubs in chain, starting from root hub for given USB bus. I don't know enough about MacOS internals how to get that information. When it gives me port number 11, I am highly suspicious (as most hubs rarely have more than 7 ports). Or if it gives port 4 for old libusb version and 5 in another version running on the same hardware. Port numbers are physical USB topology property which is independent of OS.
E.g. if I have certain chain of USB devices connected via one or more hubs, `listdevs` must be able to print identical physical port path whether it it is connected to Mac or Linux (only bus number might be different).

mvp, 2020-04-09:
@hjelmn, can you please evaluate my pull request https://github.com/libusb/libusb/pull/711. 
It fixes this issue on all MacOS versions that I have access to.
It is also consistent to `get_device_port()` code retrieving parent device using `kIOServicePlane`:

https://github.com/libusb/libusb/blob/17143e307b11075e9fd6f77eebb7483a6f635b86/libusb/os/darwin_usb.c#L957-L966

Thanks!

IgorAnohin, 2020-04-09:
> It fixes this issue on all MacOS versions that I have access to.

You may manipulate version of you used via `MAC_OS_X_VERSION_MIN_REQUIRED` define. By default, you use `10500`. Try to change it to `10400` for example. (_"We want it to work for any MacOS version"_, right?). Your PR will return different output ._.

mvp, 2020-04-09:
@IgorAnohin, feel free to submit your own pull request. Changing compiler defines will affect on which MacOS version libusb could be compiled - we never want to limit that.

I stand by my statement that my fix #711 is the only working solution until better one is offered or found.

IgorAnohin, 2020-04-09:
Thanks for the advice!
I opened PR #712 . It works for Catalina and Mojave.
You can check it out!)

mvp, 2020-07-10:
Fixed by bb37730.

mvp, 2020-07-10:
@hjelmn, what are the expectations of having 1.0.24 release cut anytime soon?
master branch has a fix, but I wonder when it will be official, preferrably in homebrew. Thanks!

closed 2020-07-10 by mvp

hjelmn, 2020-07-10:
Probably by the end of the month. I will take a look at outstanding issues and PRs and look at cutting an rc0 in a couple of weeks.

mvp, 2020-08-30:
@hjelmn, any chance to release 1.0.24 soon? It's been over a year since 1.0.23 release. Asking because libusb 1.0.23 is broken on MacOS Catalina.

mcuee, 2020-09-01:
@mvp, initially the goal of 1.0.24 is to have libusb hotplug support in Windows. I think it will take some time. Probably you can comment on the https://github.com/libusb/libusb/issues/618 to get the notics.

Or you can use libusb mailing list to request a new release.

mcuee, 2020-09-01:
I have commented on 1.0.24 milestone ticket.