libusb/libusb #708: Add vcpkg installation instructions

NancyLi1013, 2020-04-01:
`libusb`  is available as a port in [vcpkg](https://github.com/microsoft/vcpkg), a C++ library manager that simplifies installation for libusb and other project dependencies. Documenting the install process here will help users get started by providing a single set of commands to build libusb, ready to be included in their projects.

We also test whether our library ports build in various configurations (dynamic, static) on various platforms (Windows: x86, x64) to keep a wide coverage for users.

I'm a maintainer for vcpkg, and [here is what the port script looks like](https://github.com/microsoft/vcpkg/blob/master/ports/libusb/portfile.cmake) We try to keep the library maintained as close as possible to the original library.

I try to add vcpkg installation instructions but I couldn't find the related docs in github except for on https://github.com/libusb/libusb/wiki page.

Could you please help add the following contents to this port?

Installing and building libusb via vcpkg

You can download and install libusb using the vcpkg dependency manager:

     git clone https://github.com/Microsoft/vcpkg.git
     cd vcpkg
     ./bootstrap-vcpkg.sh
     ./vcpkg integrate install
     vcpkg install libusb

The libusb port in vcpkg is kept up to date by Microsoft team members and community contributors. If the version is out of date, please [create an issue or pull request](https://github.com/Microsoft/vcpkg) on the vcpkg repository.

Thanks.

LudovicRousseau, 2020-04-01:
I don't think we should document how to use vcpkg in the libusb project.
vcpkg documentation should be in the vcpkg project as it is now as you indicate.

libusb already refused to provide configuration files to support this or that software distribution. The problem is that the documentation/files provided by the libusb project would soon become out of date.

gregsskyles, 2020-04-01:
But it wouldn't hurt to document that, for example, "While we do not maintain this, Windows users might find it convenient to install `libusb` via `vcpkg` ([https://github.com/microsoft/vcpkg](url))"

dickens, 2020-04-15:
Can you provide a PR with this addition?

NancyLi1013, 2020-04-16:
@dickens 
Do you mean that let me make a PR or someone else?
Since I have no idea about which document is suitable to be added in.
I am happy to add this if some suggestions are given.

dickens, 2020-04-17:
> @dickens
> Do you mean that let me make a PR or someone else?
> Since I have no idea about which document is suitable to be added in.
> I am happy to add this if some suggestions are given.

I was suggesting that you make a PR if you wanted this information in the repo itself.  I would imagine INSTALL_WIN.txt is the best place for such an addition.

You can also consider the [Windows](https://github.com/libusb/libusb/wiki/Windows) page on the Wiki.

NancyLi1013, 2020-04-17:
Since I have no permission to add this on WiKi.
I add the instructions to INSTALL_WIN.txt in PR #715.

Thanks for your suggestion.

dickens, 2020-04-17:
Thanks for the PR.  If you have a suggested edit to the Wiki, I can facilitate that as well.

closed 2020-04-17 by dickens

mcuee, 2020-08-03:
vcpkg added to the Wiki Windows page.
https://github.com/libusb/libusb/wiki/Windows#vcpkg_port

NancyLi1013, 2020-08-03:
Hi @dickens and @mcuee 
Thanks for your support and help.

I have noticed the info that you added to Wiki Windows page. It looks good.

Actually, it might be better if we can also add how to use `libusb `via vcpkg. In this case, the users may find that it is more simple to use libusb via vcpkg.

We can add the instructions to https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows.
The contents can be like this:

```
Installing and building libusb via vcpkg

You can download and install libusb using the vcpkg dependency manager:

    git clone https://github.com/Microsoft/vcpkg.git
    cd vcpkg
    ./bootstrap-vcpkg.bat
    ./vcpkg integrate install
    ./vcpkg install libusb

The libusb port in vcpkg is kept up to date by Microsoft team members and community contributors. If the version is out of date, please create an issue or pull request(https://github.com/Microsoft/vcpkg) on the vcpkg repository.
```
In fact, vcpkg supports multiple platforms(Windows, linux and MacOS). But only adding to windows is enough.

mcuee, 2020-08-03:
Done. Please check if the following is okay or not.
https://github.com/libusb/libusb/wiki/Windows#vcpkg_port

NancyLi1013, 2020-08-03:
It's so nice. 
LGTM now. Thanks for your hard work. @mcuee

NancyLi1013, 2021-10-12:
Thanks for your correction. @Gustavomurta `"./" ` should be used on Unix platform. For Windows, it should be `".\"`.

mcuee, 2021-10-14:
> > I believe you instruction will not work under Windows Terminal.
> 
> I tested it in Windows 10, in the **Visual Studio 2019 Developer Command Prompt** and then I tested it again in the **command prompt window**.

I know your instruction works under the cmd.exe command prompt. I am just saying it does not work with Windows Terminal if you use Powershell. So the wiki instruction is more generic and it should work under both cmd.exe and powershell.

I use Powershell under Windows Terminal now much more often than command prompt.

commit be57525e (2020-04-17): Add vcpkg installation instructions

Closes #708, Closes #715