libusb/libusb #709: [windows]use libusb to control usb camera(composite device)

WangQunM, 2020-04-02:
My usb camera is a composite device, include 1 configuration, 2 interfaces and some endpoints. I wants to get video and control camera.The device driver is UsbDK, which is supported by Libusb-1.0.23。In windows device manager, I can see two device: USB camera and USB composite device.Both of their VID and PID are same.

Device Information：
Configuration Descriptor
bNumInterface 0x02
bConfigurationValue 0x01

Interface Descriptor
bInterfaceNumber 0x00
bNumEndpoints 0x01
bInterfaceClass 0x0E---->Video Interface Class
bInterfaceSubClass 0x01---->Video Control Interface Subclass

Interface Descriptor
bInterfaceNumber 0x01
bNumEndpoints 0x01
bInterfaceClass 0x0E---->Video Interface Class
bInterfaceSubClass 0x02---->Video Streaming Interface Subclass

My steps:
1.use ffmpeg or directshow to get video, it's ok.
2.use libusb to open device, it's ok.
3.after open successfully, the device disappear from windows device manager, so I can't use libusb_control_transfer.

Pseudocode:
libusb_init(ctx);
libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);
libusb_open_device_with_vid_pid(ctx, vid, pid);
libusb_claim_interface(handle, 0);
libusb_control_transfer(....);

What's wrong with my step?
I think, when I use libusb_open, the kernel driver is detached, so the device is disappeared,but...if I don't open device, how can I control my camera?

Forgive my poor English,^-^.

mcuee, 2020-06-11:
What if you use WinUSB driver? This does not seem to be a libusb problem, rather it may be a device problem or usbdk problem. The device should not disappear from windows device manager.

WangQunM, 2020-06-15:
Thanks for your reply.
when I use libusb_open, the device is disappeared, When I use libusb_close, the device is back,This make me confused.
This device can work well in Linux with LibUSB.
By the way, should I not make UsbDK as my firset choice？
Maybe I should try another driver.

diabolo38, 2020-06-15:
libusbk  temporary replace  the existing os driver  to handle it so you can access it. 
so it does not really disappear  but rather change subclass where it appear in  the device manager
initially it's likely under camera/webcam then it shall appear under usbdk category

if you to want access  the device with libusb and do not want to add a specific driver for it you do not have quite the choice than  to use usbdk. 
camera are usualy  quite complexe  composite device  you may have some related issue here.

mcuee, 2020-06-24:
All in all, I do not think this is a libusb issue, probably a usbdk issue or device issue.

mcuee, 2020-09-15:
Close this for now. Please reopen if you have new info.

closed 2020-09-15 by mcuee