libusb/libusb #717: Matching from android.hardware.usb.UsbDevice to a libusb_device

DiegoBM, 2020-04-19:
Hi everyone,

I'm trying to port a library that internally uses libusb from Linux to Android, the library managed to build apparently fine after previous build of all its dependencies, and while I have success obtaining permission for the USB device in my android application. Calling the library endpoints simply won't do anything. Then I came across this paragraph in libusb:

> The difficulties in this method is that there is no guarantee
> that it will continue to work in the future Android versions, it
> requires invoking Java APIs and running code to match each
> android.hardware.usb.UsbDevice to a libusb_device.

Could someone please describe how is that match done? I've searched around and for the love of Yoda that I cannot find anything useful or anyone describing such kind of example in some sort of detail, or some example of a working usage of the same case stated above.

Thank you in advance

XanaduAPI, 2020-12-05:
首先，你需要 用  Android Studio 构建一个 Android 项目
然后，你需要申请此App 运行时的权限

[image]
[image]

在主窗口的 OnCreate 中注册回调函数
[image]

在回调函数中 申请 USB 连接权限
[image]

然后打开设备

	private void DeviceConnect(UsbDevice vDevice)
	{
		UsbManager			vManager = (UsbManager) getSystemService(Context.USB_SERVICE);
		//vManager.requestPermission(vDevice, PendingIntent.getBroadcast(MainActivity.this, 0, new Intent(ACTION_USB_PERMISSION), 0));

		UsbEndpoint			vPointRead = null;
		UsbEndpoint			vPointWrite = null;
		int				vInterfaceNumber = 0;

		if (false == vManager.hasPermission(vDevice))
		{
			vManager.requestPermission(vDevice, PendingIntent.getBroadcast(MainActivity.this, 0, new Intent(ACTION_USB_PERMISSION), 0));
		}
		INFO("USB 权限 : " + (vManager.hasPermission(vDevice) ? "有" : "无"));

		INFO("打开设备");
		UsbDeviceConnection		vConnection = vManager.openDevice(vDevice);
		if(vConnection != null)
		{
			INFO("查找设备接口 : 0xFF-0xFE-0x02");
			for (int vIndex = 0; vIndex < vDevice.getInterfaceCount(); ++vIndex)
			{
				UsbInterface intf = vDevice.getInterface(vIndex);
				INFO("接口 : " + vIndex + " : " + intf.toString());
				if (intf.getInterfaceClass() == INTERFACE_CLASS && intf.getInterfaceSubclass() == INTERFACE_SUBCLASS && intf.getInterfaceProtocol() == INTERFACE_PROTOCOL)
				{
					vInterfaceNumber = vIndex;
					//break;
				}
			}

			INFO("使用设备接口编号 : " + vInterfaceNumber);
			UsbInterface		vInterface = vDevice.getInterface(vInterfaceNumber);
			//UsbInterface		vInterface = vDevice.getInterface(0);

			INFO("查找设备输出输出端口");
			for (int vIndex = 0; vIndex < vInterface.getEndpointCount(); ++vIndex)
			{
				UsbEndpoint point = vInterface.getEndpoint(vIndex);
				//if (point.getType() == UsbConstants.USB_ENDPOINT_XFER_BULK)
				if (point.getType() == UsbConstants.USB_ENDPOINT_XFER_BULK)
				{
					if (vPointRead == null && point.getDirection() == UsbConstants.USB_DIR_IN)
					{
						vPointRead = point;
						INFO("找到端口 : IN READ >> " + vPointRead.toString() + "\n");
					}
					else if (vPointWrite == null && point.getDirection() == UsbConstants.USB_DIR_OUT)
					{
						vPointWrite = point;
						INFO("找到端口 : OUT WRITE >> " + vPointWrite.toString() + "\n");
					}
				}
			}

			//vConnection.setConfiguration();
			//检查通信权限
			if(false == vConnection.claimInterface(vInterface, false))
			{
				INFO("通信权限 : FAILURE");
				return;
			}
			INFO("通信权限 : SUCCESS");

			//INFO("IOCTL LOCAL : " + new String(vConnection.getRawDescriptors()));

			INFO("开始写入");

			//int			vFileHandle = vConnection.getFileDescriptor();
			//INFO("IOCTL HANDLE : " + vFileHandle);

			//Integer			vIoctlResult = XanaduDevice.IoctlSend(vConnection.getFileDescriptor());
			//INFO("IOCTL SEND : " + vIoctlResult);

			String			vSend = XanaduDevice.BuildSend();
			Integer			vLength = vSend.length();
			int			vResult = 0;
			String			vStringLength = XanaduDevice.ToNetwork(vLength);
			byte[]			vByteLength = vStringLength.getBytes();

			vResult = vConnection.bulkTransfer(vPointWrite, vByteLength, 4, 1000);
			//vResult = vConnection.controlTransfer(SET_REPORT, 0x09, 0x00, 0, vByteLength, 4, 1000);
			INFO("发送长度数据 : " + vLength + "_" + StringToHex(vStringLength) + (vResult == -1 ? "失败" : "成功"));
			if(vResult == -1)
			{
				return;
			}

			vResult = vConnection.bulkTransfer(vPointWrite, vSend.getBytes(), vSend.length(), 1000);
			INFO("发送XML数据 : \n" + vSend + "_" + (vResult == -1 ? "失败" : "成功"));
			if(vResult == -1)
			{
				return;
			}

			INFO("开始读取");
			vLength = 0;
			byte[]			vRecvLength = new byte[8];
			byte[]			vBuffer = new byte[1024];
			byte[]			vRecv = new byte[1024];

			vResult = vConnection.bulkTransfer(vPointRead, vBuffer, 1024, 3000);
			vRecvLength[0] = vRecv[0];
			vRecvLength[1] = vRecv[1];
			vRecvLength[2] = vRecv[2];
			vRecvLength[3] = vRecv[3];
			INFO("接收长度数据 : Buffer.Length = " + vBuffer.length);
			System.arraycopy(vBuffer, 4, vRecv, 0, 1020);
			INFO("接收长度数据 : " + new Integer(4) + "_" + (vResult == -1 ? "失败" : "成功"));
			if(vResult == -1)
			{
				return;
			}

			vLength = FromNetwork(vRecvLength);
			vResult = vConnection.bulkTransfer(vPointRead, vRecv, vLength, 1000);
			INFO("接收数据内容 : " + (vResult == -1 ? "失败" : "成功"));
			if(vResult == -1)
			{
				return;
			}
			INFO("接收数据内容:" + vRecv[0] + vRecv[1] + vRecv[2] + vRecv[3] + vRecv[4] + vRecv[5] + vRecv[6] + vRecv[7]);
			INFO(new String(vRecv));
		}
		else
		{
			INFO("打开设备失败");
		}
	}

mcuee, 2020-12-19:
Try this tip as well from Peter Stoiber.
https://marc.info/?l=libusb-devel&m=160832455725344&w=2

```
perhaps Libusb does not work, because of this modified version you are using.
Please use the official version of Libusb and try out the following:
The latest approach of using Libusb on Android is over the method \
libusb_wrap_sys_device. There is no documentation up to now...,  I wanted to make a \
pull request about the missing doc, but hadn't time to finish my testings in the past \
on several android devs.

Any way, try out the following for using Libusb on Android:

    libusb_context *ctx;
    enum libusb_error rc;
    rc = libusb_set_option(&ctx, LIBUSB_OPTION_WEAK_AUTHORITY, NULL);
    if (rc != LIBUSB_SUCCESS) {
        __android_log_print(ANDROID_LOG_ERROR, TAG,"libusb_init failed: %d\n", ret);
        return -1;
    }
    ret = libusb_init(&ctx);
    if (ret < 0) {
        __android_log_print(ANDROID_LOG_INFO, TAG,
                            "libusb_init failed: %d\n", ret);
        return 1;
    }
    libusb_device_handle *devh = NULL;
    ret = libusb_wrap_sys_device(NULL, (intptr_t)FD, &devh);
    if (ret < 0) {
        __android_log_print(ANDROID_LOG_INFO, TAG,
                            "libusb_wrap_sys_device failed: %d\n", ret);
        return 2;
    }
    else if (devh == NULL) {
        __android_log_print(ANDROID_LOG_INFO, TAG,
                            "libusb_wrap_sys_device returned invalid handle\n");
        return 3;
    }

(FD = FileDescriptor from Android.USBManager<http://Android.USBManager>)
From here on you should be able to use all LibUsb functions as decribed. (claiming \
interfaces, detach drivers,...).
```

mcuee, 2020-12-24:
https://github.com/libusb/libusb/pull/830
Please check the above pull request as well.

XanaduAPI, 2020-12-24:
This operation can indeed connect successfully.

However, not all interfaces can be opened.

For some interfaces, The claimInterface operation will fail

XanaduAPI, 2020-12-29:
这个问题什么时候关闭呢？

mcuee, 2020-12-29:
> 这个问题什么时候关闭呢？
Tranlation added by mcuee: when can you close this issue?

@XanaduAPI Please use English here as others may not understand Chinese. Thanks.

So you think that the issue can be closed as your solution is already working? Or do you think there are works to be done?

Peter-St, 2020-12-30:
Hi,

problems with claiming the interfaces on native side could be solved by claiming them before on Java/Kothlin side.

XanaduAPI, 2020-12-31:
@Peter-St 
Do you have any sample code or references? It's a bit difficult to find information in this way. Please help me.

Peter-St, 2020-12-31:
Sure,

- start from your applied code above and proceed with the following: --> I made a PullRequest for updating the doc for Android Libusb support (read through ..) [https://github.com/libusb/libusb/pull/830](url)
- Download my fork or apply the pullRequest at your libusb Repo.
- Execute the ndk-build at the LibUsb Repo
- Add the created libs folder to your project (to the Main dir)
- In your Apps build.gradle you add the following:
`sourceSets {
        main {
            jniLibs.srcDir 'src/main/libs'
            jni.srcDirs = []
        }
    }`

- Add the JNA dependencie: `implementation 'net.java.dev.jna:jna:5.6.0@aar'` and include this to your probuard-rules.d:
  `-dontwarn java.awt.*
  -keep class com.sun.jna.* { *; }
  -keepclassmembers class * extends com.sun.jna.* { public *; }
  -dontwarn android.support.**`
- Access the libunRootedAndroid.so file over JNA: 
  `public interface unRootedSample extends Library {
    public static final unRootedSample INSTANCE = Native.load("unRootedAndroid", unRootedSample.class);
    public int main (int fileDescriptor);
}`
  `unRootedSample.INSTANCE.main(usbDeviceConnection.getFileDescriptor());`

You can modify the source of the unRootedAndroid.c for performing other actions and recompile it again.
You also could include a method for JNI access.

I have uploaded a working sample: [https://github.com/Peter-St/unRootedSample](url)

If you have any improvements plz tell.

Peter

XanaduAPI, 2020-12-31:
@Peter-St 

Okay, I will try.  But now we have a holiday.  I will contact you if I have news.  thank you.

Peter-St, 2020-12-31:
> @Peter-St
> 
> Okay, I will try. But now we have a holiday. I will contact you if I have news. thank you.

--> happy holidays and a good start into the new year. 🎉

XanaduAPI, 2021-01-04:
> > @Peter-St
> > Okay, I will try. But now we have a holiday. I will contact you if I have news. thank you.
> 
> --> happy holidays and a good start into the new year. 🎉

For me, it's just an ordinary Festival, and the new year will come to the Spring Festival.🎉

XanaduAPI, 2021-01-04:
> Sure,
>
> * start from your applied code above and proceed with the following: --> I made a PullRequest for updating the doc for Android Libusb support (read through ..) [...]

I wonder if some documents are missing? I can't find the relevant implementation.
[image]

Peter-St, 2021-01-04:
Check your Plugins for Android-Studio.
I think it's the Android NDK Support Plugin.
A List of all my installed Plugins is attached below.

[image]

XanaduAPI, 2021-01-05:
hi

I checked the list of plug-ins and found nothing extra.

public interface unRootedSample extends Library {
    public static final unRootedSample INSTANCE = Native.load("unRootedAndroid", unRootedSample.class);
    public int main (int fileDescriptor);
}

What is in "unRootedAndroid"?

Is this your C code?

Xanadu

XanaduAPI, 2021-01-05:
hi
I compared your libusb with the official one. It was found that only the. MK file was modified.

In fact, I also used libusb and passed in descriptors, but for Apple devices, I can only open a specific PTP interface, but the multiplexing interface cannot.

Specific interface calls to claimInterface fail, whether it's libusb or Java code.

Peter-St, 2021-01-05:
Hello,

Plz check my commits from the Pull Request first.
https://github.com/libusb/libusb/pull/830/commits

If you were getting no compilation errors (because no error logs were attached), then set your Android Studio debugging Options from autodetect to Java Only. (menu.entry under Run or Build menu). In Android-Studio I can only get Debugging log from "Java Only" set from Native Code.

In my project I used the official Libusb with my posted PullRequst.

The unRootedAndroid.c   (in the Examples folder) is accessed over JNA.

Claiming the Interfaces on Java still does not work on some Android devices too, but once opend the Device on Java and reopened than with LibUsb Interface claiming allways works.

XanaduAPI, 2021-01-05:
I read your explain.
I found that you need to declare the interface on the Java side.

[image]

But I fail to call UsbDeviceConnection.claimInterfacein() Java

Peter-St, 2021-01-05:
On some Devices I'm having still problems finding the interfaces on Java, but could always find them with Libusb.

Your problem looks more like the Interfaces were obtained by other kernel drivers, which you can not detach.

Try to debug, if you can detach any drivers from the Usb Interfaces... this concern is not really pointing at this topic of this Issue any more I think, because here we start to talk about something else.

So far,

Peter

XanaduAPI, 2021-01-06:
> On some Devices I'm having still problems finding the interfaces on Java, but could always find them with Libusb.
>
> Your problem looks more like the Interfaces were obtained by other kernel drivers, [...]

My goal is to use usbmuxd to run on Android and connect to IOS devices, so the operations you provide can only open some other interfaces.

As you said, maybe I need more in-depth study.

Anyway, thank you very much.

Xanadu

Peter-St, 2021-01-06:
Ok,

in your case I would not use libusb for this behavior.

Usbmuxd uses tcp to communicate.
I would start with one of the 2 following examples:
https://www.source-code.biz/snippets/java/TelnetStdioRedirector/TelnetStdioRedirector.java
https://www.source-code.biz/snippets/java/SerialPortChannel/SerialPortChannel.java
From here:
https://www.source-code.biz/snippets/java/

Look at the Telnet-Server:
https://www.source-code.biz/snippets/java/TelnetStdioRedirector/
I got this samples running on Android with a Linux PC.

You will have to send data bytes to the usbmuxd filedescriptor. Not the one from dev/bus/usb...) 
I would first test it on a Pc. On Android a rooted devices could eventually be useful.

I don't own an ios devs, so I think I couldn't help you any more in your concern.

So far,

Peter

XanaduAPI, 2021-01-07:
I see that usbmuxd relies on libusb, and it needs to enumerate USB devices.

Of course, you don't need to help me. In fact, we've switched to other ways now.

By building a tiny Linux hardware system between Android and iPhone, these permission problems can be avoided.

So much for your help. I wish you a happy life.

Peter-St, 2021-01-07:
Ok, same too you and good luck in your issue!

xloem, 2021-02-12:
I submitted #874 to make libusb seamless for my usecase on android.  I'm new to libusb and didn't review the whole api, so if anybody has comments, advice, or can add what's missing onto my work, it's appreciated.

mcuee, 2021-06-07:
https://github.com/libusb/libusb/wiki/Android

I have also update the wiki to point to this issue #717, as well as #830 and #874.

mcuee, 2023-02-11:
For Android users, please check out #1164 for further improvement.
* #1164

ben1681, 2023-08-29:
@AiXanadu 

> Hello,I have also been studying libusb recently, and I want to realize the access to iphone on android devices. Now I am looking for a solution. What do you mean by "a tiny Linux hardware system between Android and iPhone"?

ben1681, 2023-08-29:
> I see that usbmuxd relies on libusb, and it needs to enumerate USB devices.
>
> Of course, you don't need to help me. In fact, we've switched to other ways now.
>
> By building a tiny Linux hardware [...]

Hello,I have also been studying libusb recently, and I want to realize the access to iphone on android devices. Now I am looking for a solution. What do you mean by "a tiny Linux hardware system between Android and iPhone"? @AiXanadu

AiXanadu, 2023-08-31:
@ben1681 
There are indeed ways to communicate with USB devices directly using Android, and it doesn't require Root. But this involves the company's code issues, so I can't reveal too much.

What I'm talking about is another method, using a small Linux system as middleware to connect Android and iPhone.

luoqiang110, 2024-01-19:
> 确实有方法可以直接使用Android与USB设备通信，而且不需要Root。但这涉及到公司的代码问题，所以不能透露太多。

hey,boy ,有没有联系方式，想跟你请教下这些问题  /ღ( ´･ᴗ･` )比心