libusb/libusb #718: udev: libusb reports devices before they are initialized

zfigura, 2020-04-23:
libusb reports udev devices as soon as they are connected, which may be before udev has initialized them, e.g. by applying permissions. This has the effect that an application using libusb will get "access denied" errors if trying to open the device as soon as it is connected.

This is somewhat of an enhancement request, I guess, since libusb doesn't guarantee that the device will be available, but I think it's reasonable to request this of libusb.

Unfortunately I'm not sure there's a way for udev to asynchronously report device initialization, but we can poll on udev_device_get_is_initialized().

dickens, 2020-04-23:
When we [create the udev monitor](https://github.com/libusb/libusb/blob/master/libusb/os/linux_udev.c#L55), we specify "udev" as the monitor name.  This gives us a monitor where events are received "after udev has finished its event processing, all rules have been processed, and needed device nodes are created." (quoting the libudev source here).

zfigura, 2020-04-24:
Hmm, I see. I'm definitely getting hotplug events from libusb before I have write access to the device, but it appears my original diagnosis was incorrect. I'll have to give this one a closer look.

zfigura, 2020-04-26:
Okay, I'm sorry for the noise, this was a user problem after all. I unwittingly had my rules and those shipped by the distribution in such a configuration that the default permissions (664) were taking precedence at "add" time, but not at "bind" time.

closed 2020-04-26 by zfigura