libusb/libusb #723: [darwin] race condition during reset device (another process has device opened for exclusive access)

mringwal, 2020-04-29:
Since 1.0.23, libusb re-enumerates a device when libusb_reset_device is called. During re-enumeration, I get this error:

libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access

I'm certain that only my process is accessing the USB device - a Bluetooth controller.

(Background: [BTstack](https://github.com/bluekitchen/btstack/) first calls libusb_open, then  libusb_reset_device,  then starts to configure it. libusb_reset_device does not return an error).

The error only happens maybe with 50%. 

Logs:
- [device-open-ok.txt](https://github.com/libusb/libusb/files/4550617/device-open-ok.txt)
- [device-opened-exclusive-access-fail.txt](https://github.com/libusb/libusb/files/4550618/device-opened-exclusive-access-fail.txt)

The only difference is the 'exclusive access error'.

I have no experience with IOKit or libusb internals, but after adding a 10 ms delay in darwin_reset_device() after the ' while (dpriv->in_reenumerate) { } ', it the error does not show up.

seanm, 2020-06-25:
Intermittent errors that go away adding sleeps could be a threading bug.  Could you trying building and testing with ThreadSanitizer?

mringwal, 2020-06-25:
@seanm Do you have a link to 'ThreadSanitizer'. I did use the AddressSanitizer in BTstack before to find memory corruptions. Is this similar?

As the error is highly reproducible, it's easy to argue that if there's a thread switch where I've added the 10 ms sleep, it would work. What do you expect from the ThreadSanitizer?

I still don't understand what process/thread could access the device that has just been opened. There's a polling while() that checks if the enumeration is done, maybe that exits to early (flag is set, but enum not done).

seanm, 2020-06-25:
Yes, it's similar to AddressSanitizer. See https://clang.llvm.org/docs/ThreadSanitizer.html

mcuee, 2021-01-16:
@mringwal Can you try the latest git to see if the issue is still there?

mringwal, 2021-01-19:
Thanks for the notification and the update.

With current master/1.0.24, the error does not show up anymore. Excellent!

closed 2021-01-19 by mringwal