libusb/libusb #725: Windows 10: crash of libusb when interface number is missing

reddevil156, 2020-05-03:
Hi,

I found a crash libusb under some circumstances.
I want to use libusb to control an ASUS Aura device for RGB lighting. This device hangs on a usb root hub and has 2 interfaces. The first interface is 0, the second 2. Number 1 is missing!

Claiming the interface works, the crash appears in line 486 of function "windows_assign_endpoints" in windows_winusb.c

The function is opened with iface 2, altsetting 0.

In line 483 the if_desc struct should get the correct interface:
`
if_desc = &conf_desc->interface[iface].altsetting[altsetting];
`

But iface[2] is empty, so libusb crashes. Since my device has no interface 1, the array is smaller than expected. In this case it should be iface[1].

I don't know how to properly fix. I could only verify that everything works if I set it manually correct, but this is only a workaround for me and shouldn't be in the code for everyone.

I attached the debug log and a log from USBView where the device can be seen.

[debug.log](https://github.com/libusb/libusb/files/4571455/debug.log)
[USBView.txt](https://github.com/libusb/libusb/files/4571456/USBView.txt)

diabolo38, 2020-05-04:
I think the device is out of spec 
interface number shall be 0 and 1 .

diabolo38, 2020-05-05:
did you mis-used "iInterface" field that is the interface string  index (if any) ?

reddevil156, 2020-05-05:
I don't understand your question properly.

This is the code (shorted):
`
libusb_get_device_descriptor(usb_dev, &desc);
libusb_get_active_config_descriptor(usb_dev, &conf_desc)
struct libusb_interface *intf = &conf_desc->interface[1];
const struct libusb_interface_descriptor *intf_desc;
intf_desc = &intf->altsetting[0];
libusb_open(usb_dev, &dev->device_handle);
libusb_claim_interface(dev->device_handle, intf_desc->bInterfaceNumber);
`

The rest does libusb. Claiming of the interface works but crashes in assigning the endpoint.

diabolo38, 2020-05-06:
so you used the right filed  (ther''s another one name iInterface 

> libusb_claim_interface(dev->device_handle, intf_desc->bInterfaceNumber);

can you post the device descriptor ?
 if bInterfaceNumber  is 2 (what you state at first)  then device is wrong. 
The usb spec says it's  0 base index so 1st inerface shalle be  0 and second 1 not 2.
The Lib could check that bInterfaceNumber  is wirhin range  but nothing much.

reddevil156, 2020-05-07:
So, this is the descriptor (&desc)

```
bDescriptorType	0x01
bDeviceClass	0x00
bDeviceProtocol	0x00
bDeviceSubClass	0x00
bLength		0x12
bMaxPacketSize0	0x40
bNumConfigurations 0x01
bcdDevice	256
bcdUSB		512
iManufacturer	0x01
iProduct		0x02
iSerialNumber	0x03
idProduct		6387
idVendor		2821
```

This is the active config descriptor (&conf_desc)
```
MaxPower	0x08
bConfigurationValue 0x01
bDescriptorType	0x02
bLength		0x09
bNumInterfaces	0x02
bmAttributes	0xa0
iConfiguration	0x00
wTotalLengh     43
```

altsetting from interface 0:
```
bAlternateSetting	0
bDescriptorType	4
bInterfaceClass	255
bInterfaceNumber	0
bInterfaceProtocol	255
bInterfaceSubClass 255
bLength		9
bNumEndpoints	0
```

altsetting from second interface with bInterfaceNumber 2:
```
bAlternateSetting	0x00
bDescriptorType	0x04
bInterfaceClass	0x03
bInterfaceNumber	0x02
bInterfaceProtocol	0x00
bInterfaceSubClass 0x00
bLength		0x09
bNumEndpoints	0x01
extra:
[0] 0x09
[1] 0x21
[2] 0x11
[3] 0x01
```

There seems to be a difference how Windows and Linux access the usb ports. I tried my code unter linux and I get access to this device with calling 
`
libusb_claim_interface(dev->device_handle, intf_desc->bInterfaceNumber);
`
where bInterfaceNumber is 2.

diabolo38, 2020-05-07:
ok it's simple 

> altsetting from second interface with bInterfaceNumber 2:

**IS OUT OF USB SPEC !!**
2nd interface number is 1

reddevil156, 2020-05-07:
Ok I see.
Then it would be nice if you would implement a simple check that bInterfaceNumber is in range and I don't get a crash.

mcuee, 2020-06-02:
You should contact the vendor to fix the broken device.

mcuee, 2020-09-15:
Close this issue as libusb is not able to accommodate non-complaint USB devices out there.

Same for #779 https://github.com/libusb/libusb/issues/779

closed 2020-09-15 by mcuee

fenugrec, 2023-09-28:
> Close this issue as libusb is not able to accommodate non-complaint USB devices out there.

There is a difference IMO between "support non-compliant USB devices" and "not crash with non-compliant / malicious USB device". I think the latter is important... Hasn't there been enough out-of-bounds CVEs already ?

mcuee, 2023-09-28:
> There is a difference IMO between "support non-compliant USB devices" and "not crash with non-compliant / malicious USB device". I think the latter is important... Hasn't there been enough [...]

If you think that libusb can do better, please raise a PR. Thanks.

tormodvolden, 2023-09-28:
I think this was already fixed in commit f5275f9 and commit bed8d30.

fenugrec, 2023-09-28:
> I think this was already fixed in commit [f5275f9](https://github.com/libusb/libusb/commit/f5275f9a87db898af56060958df182ef7e5c1240) and commit [bed8d30](https://github.com/libusb/libusb/commit/bed8d3034eac74a6e1ba123b5c270ea63cb6cf1a).

Thanks, indeed it looks it was addressed already.