libusb/libusb #729: Cannot seem to do a bulk transfer on macOS with FLIR One Pro combination camera/IR

inflex, 2020-05-15:
Trying to port software to macOS from linux;  the FLIR One Pro consists of two cameras (visual and infrared) but it's not actually a normal camera device, instead you have to extract and process the data from the camera frames from a single datablock (  this code works in linux just fine, I'm simply recompiling in macOS Mojave )

debug output run;  things seem to work fine until it tries to pull the bulk data containing the camera data.

> Initialising libusb...
> Finding FLIR One PRO...
> [timestamp] [threadID] facility level [function call] <message>
> --------------------------------------------------------------------------------
> [ 0.004278] [00000307] libusb: debug [libusb_get_device_list] 
> [ 0.005496] [00000307] libusb: debug [libusb_get_device_descriptor] 
> [ 0.005507] [00000307] libusb: debug [libusb_open] open 20.18
> [ 0.005674] [00000307] libusb: debug [darwin_open] device open for access
> Setting USB configuration for FLIR...
> [ 0.005690] [00000307] libusb: debug [libusb_set_configuration] configuration 3
> Claiming FLIR Interfaces...
> [ 0.005707] [00000307] libusb: debug [libusb_claim_interface] interface 0
> [ 0.006232] [00000307] libusb: debug [get_endpoints] building table of endpoints.
> [ 0.006263] [00000307] libusb: debug [get_endpoints] interface: 0 pipe 1: dir: 1 number: 1
> [ 0.006273] [00000307] libusb: debug [get_endpoints] interface: 0 pipe 2: dir: 0 number: 2
> [ 0.006316] [00000307] libusb: debug [darwin_claim_interface] interface opened
> [ 0.006322] [00000307] libusb: debug [libusb_claim_interface] interface 1
> [ 0.006883] [00000307] libusb: debug [get_endpoints] building table of endpoints.
> [ 0.006902] [00000307] libusb: debug [get_endpoints] interface: 1 pipe 1: dir: 1 number: 3
> [ 0.006912] [00000307] libusb: debug [get_endpoints] interface: 1 pipe 2: dir: 0 number: 4
> [ 0.006960] [00000307] libusb: debug [darwin_claim_interface] interface opened
> [ 0.006966] [00000307] libusb: debug [libusb_claim_interface] interface 2
> [ 0.007636] [00000307] libusb: debug [get_endpoints] building table of endpoints.
> [ 0.007668] [00000307] libusb: debug [get_endpoints] interface: 2 pipe 1: dir: 1 number: 5
> [ 0.007685] [00000307] libusb: debug [get_endpoints] interface: 2 pipe 2: dir: 0 number: 6
> [ 0.007736] [00000307] libusb: debug [darwin_claim_interface] interface opened
> Done
> stop interface 2 FRAME
> [ 0.007785] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7ffb7ae266a8
> [ 0.007804] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7ffb7ae266a8
> [ 0.010812] [00000307] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
> [ 0.010836] [00009607] libusb: debug [darwin_async_io_callback] an async io operation has completed
> [ 0.010858] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.010874] [00000307] libusb: debug [handle_events] poll fds modified, reallocating
> [ 0.010878] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
> [ 0.010897] [00000307] libusb: debug [handle_events] poll() returned 1
> [ 0.010900] [00000307] libusb: debug [handle_events] caught a fish on the event pipe
> [ 0.010902] [00000307] libusb: debug [darwin_handle_transfer_completion] handling control completion with kernel status 0
> [ 0.010905] [00000307] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ffb7ae266a8 has callback 0x10f49ff68
> [ 0.010908] [00000307] libusb: debug [sync_transfer_cb] actual_length=0
> [ 0.010931] [00000307] libusb: debug [libusb_free_transfer] transfer 0x7ffb7ae266a8
> stop interface 1 FILEIO
> [ 0.010947] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7ffb7ad3e6c8
> [ 0.010950] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7ffb7ad3e6c8
> [ 0.015015] [00009607] libusb: debug [darwin_async_io_callback] an async io operation has completed
> [ 0.015014] [00000307] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
> [ 0.015060] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.015066] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
> [ 0.015078] [00000307] libusb: debug [handle_events] poll() returned 1
> [ 0.015080] [00000307] libusb: debug [handle_events] caught a fish on the event pipe
> [ 0.015083] [00000307] libusb: debug [darwin_handle_transfer_completion] handling control completion with kernel status 0
> [ 0.015086] [00000307] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ffb7ad3e6c8 has callback 0x10f49ff68
> [ 0.015089] [00000307] libusb: debug [sync_transfer_cb] actual_length=0
> [ 0.015094] [00000307] libusb: debug [libusb_free_transfer] transfer 0x7ffb7ad3e6c8
> start interface 1 FILEIO
> [ 0.015102] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7ffb7ac79728
> [ 0.015105] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7ffb7ac79728
> [ 0.021952] [00000307] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
> [ 0.021974] [00009607] libusb: debug [darwin_async_io_callback] an async io operation has completed
> [ 0.021982] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.022036] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
> [ 0.022061] [00000307] libusb: debug [handle_events] poll() returned 1
> [ 0.022068] [00000307] libusb: debug [handle_events] caught a fish on the event pipe
> [ 0.022074] [00000307] libusb: debug [darwin_handle_transfer_completion] handling control completion with kernel status 0
> [ 0.022082] [00000307] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ffb7ac79728 has callback 0x10f49ff68
> [ 0.022088] [00000307] libusb: debug [sync_transfer_cb] actual_length=0
[... 86 lines cut ...]
> [ 7.992462] [00000307] libusb: debug [get_endpoints] building table of endpoints.
> [ 7.992485] [00000307] libusb: debug [get_endpoints] interface: 1 pipe 1: dir: 1 number: 3
> [ 7.992495] [00000307] libusb: debug [get_endpoints] interface: 1 pipe 2: dir: 0 number: 4
> [ 7.992546] [00000307] libusb: debug [darwin_claim_interface] interface opened
> [ 7.992551] [00000307] libusb: debug [darwin_restore_state] darwin/restore_state: re-claiming interface 2
> [ 7.993487] [00000307] libusb: debug [get_endpoints] building table of endpoints.
> [ 7.993504] [00000307] libusb: debug [get_endpoints] interface: 2 pipe 1: dir: 1 number: 5
> [ 7.993510] [00000307] libusb: debug [get_endpoints] interface: 2 pipe 2: dir: 0 number: 6
> [ 7.993539] [00000307] libusb: debug [darwin_claim_interface] interface opened
> [ 7.993542] [00000307] libusb: debug [darwin_restore_state] darwin/restore_state: device state restored
> [ 7.993549] [00000307] libusb: debug [libusb_close] 
> [ 7.993554] [00000307] libusb: debug [libusb_release_interface] interface 0
> [ 7.993698] [00000307] libusb: debug [libusb_release_interface] interface 1
> [ 7.993803] [00000307] libusb: debug [libusb_release_interface] interface 2
> [ 7.993934] [00000307] libusb: debug [libusb_exit] 
> [ 7.993940] [00000307] libusb: debug [libusb_exit] destroying default context
> [ 7.993943] [00000307] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
> [ 7.993946] [00000307] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 7.993950] [00000307] libusb: debug [handle_events] poll() 1 fds with timeout in 0ms
> [ 7.993962] [00000307] libusb: debug [handle_events] poll() returned 0
> [ 7.993966] [00000307] libusb: debug [libusb_unref_device] destroy device 20.18
> [ 7.993969] [00000307] libusb: debug [libusb_unref_device] destroy device 20.4
> [ 7.993973] [00000307] libusb: debug [libusb_unref_device] destroy device 20.2
> [ 7.993976] [00000307] libusb: debug [usbi_remove_pollfd] remove fd 6
> [ 7.994072] [00009607] libusb: debug [darwin_event_thread_main] darwin event thread exiting

inflex, 2020-05-16:
At least macOS sees the device and recognises it, unlike Windows which is blinder than a cat that refuses to see the food in front of it.

```
+-o FLIR ONE Camera@14200000  <class IORegistryEntry:IOService:IOUSBNub:IOUSBDevice, id 0x1000028d0, registered, matched, active, busy 0 (2 ms), retain$
  | {
  |   "sessionID" = 35442741980504
  |   "idProduct" = 6550
  |   "iManufacturer" = 1
  |   "bDeviceClass" = 0
  |   "bMaxPacketSize0" = 64
  |   "bcdDevice" = 264
  |   "iProduct" = 2
  |   "iSerialNumber" = 3
  |   "bNumConfigurations" = 1
  |   "Bus Power Available" = 250
  |   "USB Address" = 11
  |   "Built-In" = No
  |   "locationID" = 337641472
  |   "bDeviceSubClass" = 0
  |   "bcdUSB" = 512
  |   "USB Product Name" = "FLIR ONE Camera"
  |   "PortNum" = 2
  |   "non-removable" = "no"
  |   "AppleUSBAlternateServiceRegistryID" = 4294977742
  |   "bDeviceProtocol" = 0
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
  |   "kUSBCurrentConfiguration" = 3
  |   "Device Speed" = 2
  |   "USB Vendor Name" = "FLIR Systems"
  |   "idVendor" = 2507
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "F07H7L007A1"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
  | 
  +-o iAP Interface@0  <class IORegistryEntry:IOService:IOUSBNub:IOUSBInterface, id 0x1000028df, registered, matched, active, busy 0 (0 ms), retain 5>
  +-o com.flir.rosebud.fileio@1  <class IORegistryEntry:IOService:IOUSBNub:IOUSBInterface, id 0x1000028e0, registered, matched, active, busy 0 (0 ms), $
  +-o com.flir.rosebud.frame@2  <class IORegistryEntry:IOService:IOUSBNub:IOUSBInterface, id 0x1000028e2, registered, matched, active, busy 0 (0 ms), r$

```

mcuee, 2020-06-11:
Where is the code? Is it this one?
https://github.com/fnoop/flirone-v4l2

mcuee, 2020-06-11:
BTW, you can use WinUSB driver for the whole device and then see if you can run the code under Windows.

inflex, 2020-06-11:
Hello Mcuee,
  
   The software is a cut down version of that flirone-v4l2 (no v4l2 part), I've been looking at shifting over to using the macOS USB interface rather than libusb.  I don't my specific variant handy right now.

   As for windows, I've got someone building a WinUSB driver but there's been some complications due to the FLIR One Pro's compound nature, but there's some progress being made.

mcuee, 2020-06-11:
I see. 
For Windows, you can use Winusb to replace the USB Composite Parent driver (you can use Zadig to do that). Then you can try again to see if that works.

inflex, 2020-06-11:
Will see how things go over the course of the next week or two.  Things were getting "close" and the Zadig tool was being used but that side of matters is above my head for now (hence paying someone else to do it ;) ).

As for macOS, I'll have another attempt at it.  At least for linux, it's working perfectly.

hjelmn, 2020-06-12:
Will see if I can figure this one out.

The error is:
```
#define kIOUSBUnknownPipeErr                                                            iokit_usb_err(0x61)                                                                     // 0xe0004061  Pipe ref not recognized
```

hjelmn, 2020-06-12:
Seems odd. The endpoint finds a match. Not sure why macOS would be unhappy with the pipe ref. You will likely run into this same error code even if you use IOKit directly.

Can you post the complete descriptor for the device?

hjelmn, 2020-06-12:
Another note. The failure is on the call to GetPipeProperties. I will update libusb to use GetEndpointPropertiesV3 if it is available. Not sure if that will help but worth a try.

hjelmn, 2020-06-12:
Ok, pushed updates to translate that error code into a human-readable string and use GetEndpointPropertiesV3 on 10.9.x and newer. Can you give master a try and see if that helps?

inflex, 2020-06-12:
Thanks, just finished a long run of fixing macbooks under the microscope, I'll try the update and report back for you, sincerely appreciated.

hjelmn, 2020-06-17:
Just pushed a fix to my change. Let me know if you get a chance to try it.

inflex, 2020-06-17:
Thanks, struggling to catch a moment to get to the coding/recompile, definitely appreciate it.  Will try this evening.

mcuee, 2020-07-30:
@inflex Any updates on the testing results? Thanks in advance.

mcuee, 2021-01-25:
@inflex Please help to try. If not this issue will be closed due to inactivity. But you can always reopen once you get the time.

mcuee, 2021-02-26:
Close for now.

lnchan, 2024-01-16:
Hello,
I'm not sure this issue could be re-opened; but I suffer from the same error, on a different USB device. Running macOS Sonoma 14.2.1 with libusb v1.0.26.11724.
```
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = Out): pipe ref not recognized (code = 0xe0004061)
libusb: error [submit_bulk_transfer] bulk transfer failed (dir = In): pipe ref not recognized (code = 0xe0004061)
```
The device similarly causes issues on versions above Windows XP/Vista (refusing the device descriptors), but works on Linux without any issues.

It seems that IOUSB is refusing any access to the device's endpoints, calling the device descriptors "invalid".

```
default	11:48:02.861318+0100	kernel	IOUSBHostFamily::validateEndpointMaxPacketSize: USB 2.0 5.[5|7].3: endpoint 0x81 invalid wMaxPacketSize 0x0008
error	11:48:02.861325+0100	kernel	AppleT8103USBXHCI@00000000: AppleUSBHostController::createPipe: invalid descriptors
default	11:48:02.861332+0100	kernel	IOUSBHostFamily::validateEndpointMaxPacketSize: USB 2.0 5.[5|7].3: endpoint 0x02 invalid wMaxPacketSize 0x0008
error	11:48:02.861335+0100	kernel	AppleT8103USBXHCI@00000000: AppleUSBHostController::createPipe: invalid descriptors
default	11:48:02.861407+0100	kernel	IOUSBHostFamily::validateEndpointMaxPacketSize: USB 2.0 5.[5|7].3: endpoint 0x02 invalid wMaxPacketSize 0x0008
error	11:48:02.861407+0100	kernel	AppleT8103USBXHCI@00000000: AppleUSBHostController::createPipe: invalid descriptors
```

Sadly, I am not sure this would be fixable within libusb, as this looks like a IOUSB bug.

For reference, here is more information about the device:
```
libusb: debug [libusb_get_config_descriptor] index 0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x1
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x1

$ ioreg -p IOUSB -w0 -l
[...]
    +-o [...]@00100000  <class IOUSBHostDevice, id 0x10008a764, registered, matched, active, busy 0 (24 ms), retain 29>
        {
          "sessionID" = 12938481846587
          "USBSpeed" = 2
          "idProduct" = [...]
          "iManufacturer" = 1
          "bDeviceClass" = 0
          "IOPowerManagement" = {"PowerOverrideOn"=Yes,"DevicePowerState"=2,"CurrentPowerState"=2,"CapabilityFlags"=32768,"MaxPowerState"=2,"DriverPowerState"=0}
          "bcdDevice" = 257
          "bMaxPacketSize0" = 8
          "iProduct" = 2
          "iSerialNumber" = 0
          "bNumConfigurations" = 1
          "UsbDeviceSignature" = <[...]>
          "USB Product Name" = "[...]"
          "locationID" = 1048576
          "bDeviceSubClass" = 0
          "bcdUSB" = 272
          "USB Address" = 1
          "kUSBCurrentConfiguration" = 1
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "bDeviceProtocol" = 0
          "USBPortType" = 0
          "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
          "USB Vendor Name" = "[...]"
          "Device Speed" = 0
          "idVendor" = [...]
          "kUSBProductString" = "[...]"
          "IOGeneralInterest" = "IOCommand is not serializable"
          "kUSBAddress" = 1
          "kUSBVendorString" = "[...]"
        }
```

mcuee, 2024-01-16:
@lnchan

Issue reopened.

Please help to try git or libusb-1.0.27rc1 to see if that helps. Thanks.

lnchan, 2024-01-16:
Re-tried using a version compiled from Git, no difference whatsoever in behaviour.

lnchan, 2024-02-04:
Additional information: it is extremely probable that both Windows and macOS reject the device for being a low-speed device having bulk endpoints, which violates spec. Low-speed devices cannot have bulk endpoints as per the USB spec.

mcuee, 2024-02-04:
> Additional information (at least for my device): it is extremely probable that both Windows and macOS reject the device for being a low-speed device having bulk endpoints, which violates spec. Low-speed devices cannot have bulk endpoints as per the USB spec.

I see. In that case, this is not a valid issue. Latest version of Windows and macOS are more strict about USB Specs.

Example: avr-cdc does not work under Windows 8 and later.
http://www.recursion.jp/prose/avrcdc/

mcuee, 2024-02-04:
I will have to close this issue since libusb can not help if the OS does not support the device.

closed 2024-02-04 by mcuee