libusb/libusb #738: MacOS libusb_bulk_transfer with small timeout may lost packets

svilen18, 2020-05-28:
In main thread is send bulk command to USB device, as result it returns 4688 bytes of data, split in packets of 64 bytes - that is the maximum transfer size of the device. There is a thread for receiving the data with following code:
```
 int trn; 
 BYTE buff[8192], *b;
  while (loop)
    {
    int err = libusb_bulk_transfer(usbh, 0x81, buff, 8192, &trn, 5);
    if (!err || err == LIBUSB_ERROR_TIMEOUT)
      {
      //store the received data according to the trn amount
      }
    else qDebug("receive error %d", err);
    }

```
Under Windows all is working perfectly, but under MacOS often the beginning of the 4688 bytes data is missing. The amount of missing data is random between the failures, but is always multiple of 64.
If I change the timeout from 5ms to 50ms the failures occur very rare and if set to 500ms never noticed to occur, but that add a big lag to the software.
I suspect when a timeout occurs the received data is discarded or some internal buffers are reset. It can be in relation to using the command in another thread.
libusb: 1.0.23
macOS: 10.15.5

diabolo38, 2020-06-03:
even it's seam there's a real problem under MacOS for you , I also thinks your device or how you handle it is quite not the righ one 

If you exchange some sort of message  more than the usb maxpacket size  it shall be bound by a short or zlp packet. So you just push a large  ( assuming full length t is less the os limit size)  and it will complete when all data will be received without any extra pains or need to poll on small tiemout request .
This is the proper way  to design bulk messaging and overcome the "timeout issuse poll method "  it also minimize risk or loosing synch/and many over unused things that goes arround badly designed or handled bulks communication  etc ....

svilen18, 2020-06-05:
Thanks for clarify that.
Because the device never send more than 64 bytes messages, I reduced the buffer from 8192 to 64 bytes and there is never lost a data even with small timeout.
Even that is not a issue and not related to libusb, there is a differences how the OS-es handle the timeouts and bigger buffers and should be taken in account if same code need to work on different platforms.

diabolo38, 2020-06-05:
That's better and may avoid that what received get discarded (babble)....still not best design likely much slower and waste of cpu system load  but i guess you don't care it's fine

seanm, 2020-06-25:
Seems like this ticket could be closed?

closed 2020-06-25 by svilen18