libusb/libusb #739: Risk in discard urb that will cause data lost.

masc2008, 2020-06-09:
here is the risk:
libusb_interrupt_transfer will call discard_urbs in libusb via ioctl(fd, IOCTL_USBFS_DISCARDURB, urb) when timeout happen. 
In kernel it will call proc_unlinkurb in (drivers/usb/core/devio.c), if proc_unlinkurb returns 0, then libusb layer will set tpriv->reap_action = CANCELLED, and app level will not receive this packet. While proc_unlinkurb has risk that before it call usb_kill_urb, the urb receives data from device. The data will lost in such case. 
But from application level, the expectation is never lost any bus data no matter discard URB or not. 
The correct behavior is: when discard URB, it stop transfer from host controller and when submit another URB, it reads out data successfully.
Please refer below diff to fix this issue:

diff --git a/libusb/os/linux_usbfs.c b/libusb/os/linux_usbfs.c
index 082d726..5745ba2 100644
--- a/libusb/os/linux_usbfs.c
+++ b/libusb/os/linux_usbfs.c
@@ -2598,6 +2598,11 @@ static int reap_for_handle(struct libusb_device_handle *handle)

        usbi_dbg("urb type=%u status=%d transferred=%d", urb->type, urb->status, urb->actual_length);

+       if (tpriv->reap_action == CANCELLED) {
+               if ((urb->status == 0) ||
+                   (urb->status == -EREMOTEIO))
+                       tpriv->reap_action = NNORMAL;
+       }
        switch (transfer->type) {
        case LIBUSB_TRANSFER_TYPE_ISOCHRONOUS:
                return handle_iso_completion(itransfer, urb);

dickens, 2020-06-09:
Can you explain further how we end up losing data?  For all URB types, I do not see a path where the data would not be accounted for:

- control transfers: [here](https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L2525)
- bulk transfers: [here (when reap_action != NORMAL)](https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L2302) and [here otherwise](https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L2315)
- isochronous transfers: [here](https://github.com/libusb/libusb/blob/master/libusb/os/linux_usbfs.c#L2465)

What am I missing?

masc2008, 2020-06-10:
The cause is when  tpriv->reap_action != NORMAL, and  tpriv->reap_action ==
CANCELLED, then it will call  usbi_handle_transfer_cancellation(itransfer).
While when urb status is not -ENOENT, that stands a URB is not really
CANCELLED, and the we need further call usbi_handle_transfer_completion
otherthan usbi_handle_transfer_cancellation.

dickens, 2020-06-29:
Closing, since #740 is closed.

closed 2020-06-29 by dickens