libusb/libusb #741: Remove support of libusbk.sys and libusb0.sys, remove dependancy to libusbk.dll

mcuee, 2020-06-11:
Now that Windows XP is long gone and Windows 7 is also obsolete (or at least going down the obsolete route since it is no longer supported by Microsoft), the importance of libusbK.sys and libusb0.sys have gone down significantly. 

I think maybe it is a good idea to simplify the Windows libusb code base to remove libusbk.dll dependency and removethe support of libusbk.sys and libusb0.sys.

mcuee, 2020-06-11:
Why?
1) Windows XP/Vista and even Windows 7 are no longer supported by Windows
2) Only advantage of libusb0.sys is the filter support now. usbdk should be the way to go for filter support. libusb Windows does not support very well libusb0.sys anyway.
3) Only advantage of libusbk.sys compared to WinUSB is the support of isoc  transfer under Windows 7. WinUSB supports isoc transfer under Windows 8 and later.
3) libusb-win32 project is now in maintenance mode.
4) libusbk project is also not seeing much activity.
5) Removal of libusbk.dll will reduce the code complexity of Windows backend.

qmfrederik, 2020-06-11:
One of the reasons we used libusb0.sys was because it was the only back-end (at the time) which supported changing the device configuration. Is that still the case?

mcuee, 2020-06-12:
> One of the reasons we used libusb0.sys was because it was the only back-end (at the time) which supported changing the device configuration. Is that still the case?

Unfortunately it is still the case.

qmfrederik, 2020-06-16:
> Are you using libusb or libusb-win32 or libusbk with libusb0.sys? In case you use libusb Windows with libusb0.sys (it does not work well anyway), you can continue to use the 1.0.23 release if this proposal is accepted.

My main use case was porting [usbmuxd](https://github.com/libimobiledevice/usbmuxd) to Windows. At the time, I used libusb-win32 instead of libusb. I opened #473 in the hope I would be able to just use libusb.

Continuing to use the 1.0.23 release seems like a good middle ground. I find libusb to be very stable for my use cases.

codido, 2020-06-17:
>     3. Only advantage of libusbk.sys compared to WinUSB is the support of isoc  transfer under Windows 7. WinUSB supports isoc transfer under Windows 8 and later.

At least in the past, I believe WinUSB didn't have support for resetting devices while libusbk did. Unfortunately, this is a required feature for common use cases like DFU.

Is this supported for more recent versions of WinUSB?

mcuee, 2020-06-20:
Thanks for all the comments. Looks like libusbk and even libusb0.sys have some use cases.

mcuee, 2020-06-20:
> At least in the past, I believe WinUSB didn't have support for resetting devices while libusbk did. Unfortunately, this is a required feature for common use cases like DFU.
>
> Is this supported for [...]

Not supported by WinUSB.

However, the proper way to install reset in the DFU should be use a custom reset command in the device firmware side.
https://community.osr.com/discussion/252223/usb-reset-for-dfu-with-winusb

mcuee, 2020-06-20:
Work-around for WinUSB set configuration limitation: to use a lower KMDF filter driver.
https://nadavrub.wordpress.com/2014/10/16/winusb-with-non-default-configuration/

mcuee, 2020-06-20:
http://libusbk.sourceforge.net/UsbK3/usbk_comparisons.html
Some info about libusbk.sys, libusb0.sys and WinUSB.

codido, 2020-06-20:
> However, the proper way to install reset in the DFU should be use a custom reset command in the device firmware side.

Unfortunately, that's not always feasible to do so, especially for existing devices. AFAIU, some DFU devices simply do not have such special vendor commands and the DFU specification seems to rely on usb reset signalling (https://www.usb.org/sites/default/files/DFU_1.1.pdf).

In addition, there are other use cases (e.g. Android AOA) which rely on that functionality as well, though these are obviously less commonly used than DFU.

mcuee, 2020-06-20:
As for reset, the best case is to use IOCTL_USB_HUB_CYCLE_PORT IOCTL.
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbioctl/ni-usbioctl-ioctl_usb_hub_cycle_port?redirectedfrom=MSDN

Take note Windows Vista/7 do not support this.

Supported on Windows 8 and later versions of Windows, if the caller is running as Administrator. 

Supported on Microsoft Windows Server 2003, Windows XP-based versions of Windows. 

Not supported on Windows 7, Windows Vista, and Windows Server 2008.

mcuee, 2020-06-20:
Reference: https://docs.microsoft.com/en-us/previous-versions/windows/hardware/drivers/ff540046(v=vs.85)#winusb

codido, 2020-06-20:
@mcuee right, though as you mentioned, this would require administrator rights which aren't required when libusbk is used, and end users may be reluctant to approve that.

This is not a limitation that can't be worked around (e.g. by using UsbDk), but it does make things a bit more cumbersome.

diabolo38, 2020-06-20:
Libusbk enable to open a device multiple time in same or multiple process a  requirement  for most my usage: simple  user space driver as dll in multiple app with system wide exclusive device acess. 
that is impossible with winusb or hid device that would  require open close on each instance operation that is performance wise real bad (tested).

mcuee, 2020-06-24:
@diabolo38 This is a good point as well.

mcuee, 2020-06-24:
In the end, probably we have to keep status quo.

mcuee, 2020-07-29:
Reopen this to solicit more inputs.

mcuee, 2020-07-30:
Reference discussion:
https://github.com/libusb/libusb/issues/619

Pete has made a good comment about the HID backend and libusb0/libusbk support there, which can be of good reference.

mcuee, 2021-01-30:
All in all, the current Windows backend is pretty complicated and there is no other active Windows maintainer than Chris (who also takes care of Linux as of now). So please help to provide pull request if you want to improve the Windows backend, including the support of libusb0.sys and libusbk.sys.

hjelmn, 2021-01-30:
Indeed. I agree with Pete that it would be more useful to find another Windows backend maintainer. I have no love for the hid backend (it is responsible for many disagreements in this community) but if it is useful to someone and they can pitch in to help it should be fixed. Otherwise I don't know what else we can do. The same goes for any part of the Windows backend. Linux and macOS are near trivial to maintain but Windows takes a lot of effort 1) to understand the backend, and 2) keep it up to date. It is the curse of any complex code.

mcuee, 2021-07-05:
Close this issue since Travis is back to support libusbk development.

mcuee, 2021-07-05:
We will keep the Windows HID backend as well #758.

closed 2021-07-05 by mcuee