libusb/libusb #743: libusb_set_configuration() returns LIBUSB_ERROR_INVALID_PARAM

thierer, 2020-06-13:
Calling libusb_set_configuration() with a configuration value of 0 or -1 returns a LIBUSB_ERROR_INVALID_PARAM error on win 10 with libusb-1.0.23 (haven't tried any other windows or libusb version; it does work on linux, though).

From what I understand, libusb can't set the configuration on windows (libusb_set_configuration() calls with a configuration value > 0 seem to be silently ignored).

Wouldn't it be best to just ignore errors with configuration values 0 and -1 as well?

I can see why you would return an error (because the function can't do what it was asked to), but the same is true for configuration values > 0 and I find "invalid parameter" not very appropriate as the documentation for libusb_set_configuration() [explicitly mentions 0 and -1 as valid configuration numbers](http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga785ddea63a2b9bcb879a614ca4867bed):

> A configuration value of -1 will put the device in unconfigured state. The USB specifications state that a configuration value of 0 does this, however buggy devices exist which actually have a configuration 0.

mcuee, 2020-06-20:
Probably the documentation needs to be updated. -1 should not be a valid configuration and 0 is the unconfigured state. The buggy device may be really legacy devices which we should not care now. Anyway, I think the current Windows code behavior is correct.

thierer, 2020-06-20:
Please note that not only using -1 returns LIBUSB_ERROR_INVALID_PARAM, but also a configuration value of 0 (I'm not sure if your comment "_... and 0 is the unconfigured state_" is meant to express that you consider both -1 and 0 illegal values).

I suspect this https://github.com/libusb/libusb/blob/ad0473413edec14805e59599ff537eba0f5b8d33/libusb/os/windows_winusb.c#L2454-L2461 to be the code that returns the LIBUSB_ERROR_INVALID_PARAM error.

If there are "issues" to select the configuration from WinUSB, as the comment suggests, I think LIBUSB_ERROR_NOT_SUPPORTED (what `hid_submit_control_transfer()` uses) would be more appropriate. 

https://github.com/libusb/libusb/blob/ad0473413edec14805e59599ff537eba0f5b8d33/libusb/os/windows_winusb.c#L3657-L3664

mcuee, 2020-06-24:
-1 is an illegal value. 
0 means un-configured state. But you will not be able to achieve what you want with WinUSB or libusbk driver.

As for the error code, take note it is considered a warning and we do not want the code to fail. 

To me, the better way is probably like the following.
-1                                                                            -- LIBUSB_ERROR_INVALID_PARAM
0 or other number than the current configuration -- LIBUSB_ERROR_NOT_SUPPORTED.

I will let other maintainers to comment.

mcuee, 2020-06-24:
BTW, for legacy buggy device with configuration 0, I highly doubt it will even work under Windows 8.1/10 which are more stringent than Windows XP/7 in terms of USB compliance. I think it is an idea to update the documentation as well.

"A configuration value of -1 will put the device in unconfigured state. The USB specifications state that a configuration value of 0 does this, however buggy devices exist which actually have a configuration 0."

Do we really see codes which use "-1" to put the device in un-configured state? Is that even supported under macOS?

thierer, 2020-06-27:
The whole "use -1 as a substitute for 0 to deconfigure the device" seems to have originated on linux, with commit https://github.com/torvalds/linux/commit/3f141e2aed586c41c2666d49c70c1c1bbb6d6abd. That was 13 years ago, so I don't know how relevant it still is today, but at least there's some merit to allow both -1 and 0 as configuration values on linux.

On both windows and macos libusb seems to handle the "-1" case by just changing it to "0":

https://github.com/libusb/libusb/blob/ad0473413edec14805e59599ff537eba0f5b8d33/libusb/os/windows_common.c#L629-L633

https://github.com/libusb/libusb/blob/ad0473413edec14805e59599ff537eba0f5b8d33/libusb/os/darwin_usb.c#L1296-L1302

Which means that the LIBUSB_ERROR_INVALID_PARAM that I got for using "-1" as configuration value, as outlined in my original post, was in reality also just for "0".

Now, that [WinUsb_ControlTransfer()](https://docs.microsoft.com/en-us/windows/win32/api/winusb/nf-winusb-winusb_controltransfer) apparently interprets the message it is supposed to send and considers "0" an illegal configuration, suggests that de-configuring a device by setting configuration 0 isn't supported on Windows at all (don't know about MacOS, [the documentation I found](https://developer.apple.com/documentation/iokit/iousbdeviceinterface500/1559580-setconfiguration?language=occ) is a bit sparse).

So to summarize:

1. Allowing both -1 and 0 as configuration values is useful on linux, because the OS supports them and they are actually handled differently.
2. If libusb supports it for linux, it should be generally supported on all platforms and that means for me that it should not return INVALID_PARAM, because the parameters -1 or 0 are not really invalid in the scope of libusb, it's just that the notion they carry might not be supported on a platform. I consider NOT_SUPPORTED a better return code to communicate that fact.
3. The documentation for `libusb_set_configuration()` should be updated to both state which platforms actually support de-configuring a device and list LIBUSB_ERROR_NOT_SUPPORTED as an error code that's used in this case.

I could prepare a PR to show what I mean, it's just that I only ran into this as a user of libusb. I'm not an expert and can only really test on linux.

mcuee, 2020-06-29:
Thanks. A PR will be good.

commit 4261cbef (2020-07-16): winusb: setting a non-default config not supported

Just because winusb doesn't seem to support setting a different
configuration, that doesn't mean it's an invalid parameter.

It's generally supported by libusb, just not on this platform.

So return LIBUSB_ERROR_NOT_SUPPORTED instead of
LIBUSB_ERROR_INVALID_PARAM.

Closes #743
Closes #752
closed 2020-07-16 by dickens