libusb/libusb #745: calling libusb_open and then libusb_closed on my CDC USB devicee means that I cannot see it in the device list anymore (Windows only)

andrew-mclachlan, 2020-06-18:
I've got a simple function that initializes the context, sets the configuration to use UsbDk and then attempts to find the USB device information (based off the simple example with libUSB).

If I call the function n-times, then the first time I see the device in the list and I can open it and close it, but the second time, the device cannot be found.

This happens on Windows only.

Any ideas?

[ 0.041395] [00001444] libusb: debug [libusb_get_device_list]
[ 0.048306] [00001444] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.050546] [00001444] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.050847] [00001444] libusb: debug [libusb_get_device_descriptor]
...
[ 0.117727] [00001444] libusb: debug [libusb_get_device_descriptor]
 -- Found Vendor 0x44b, Product 0x1 --
[ 0.124651] [00001444] libusb: debug [libusb_open] open 2.2
[ 0.470739] [00001444] libusb: debug [libusb_close]
...
[ 0.503348] [00001444] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.504163] [00001444] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.504550] [00001444] libusb: debug [libusb_unref_device] destroy device 2.12
[ 0.507660] [00001444] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.509629] [00001444] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.509982] [00001444] libusb: debug [libusb_unref_device] destroy device 2.7
[ 0.510300] [00001444] libusb: debug [libusb_unref_device] destroy device 2.14
[ 0.511022] [00001444] libusb: debug [libusb_unref_device] destroy device 2.8
[ 0.511499] [00001444] libusb: debug [libusb_unref_device] destroy device 2.9
[ 0.511984] [00001444] libusb: debug [libusb_unref_device] destroy device 2.15
[ 0.512341] [00001444] libusb: debug [libusb_unref_device] destroy device 2.16
[ 0.512758] [00001444] libusb: debug [libusb_unref_device] destroy device 2.11
[ 0.513206] [00001444] libusb: debug [libusb_unref_device] destroy device 2.13
[ 0.513698] [00001444] libusb: debug [libusb_unref_device] destroy device 2.10
[ 0.514165] [00001444] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.514592] [00001444] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.514993] [00001444] libusb: debug [libusb_unref_device] destroy device 200.2
[ 0.515468] [00001444] libusb: debug [libusb_exit]
[ 0.518961] [00001444] libusb: debug [libusb_exit] destroying default context
[ 0.519866] [00001444] libusb: debug [usbi_remove_pollfd] remove fd 0
 -- again --
[ 0.546157] [00001444] libusb: debug [libusb_get_device_list]
[ 0.547007] [00001444] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.547757] [00001444] libusb: debug [libusb_get_device_descriptor]
...
[ 0.601196] [00001444] libusb: debug [libusb_get_device_descriptor]
 -- not found --
[ 0.603480] [00001444] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.603928] [00001444] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.604319] [00001444] libusb: debug [libusb_unref_device] destroy device 2.12
[ 0.606579] [00001444] libusb: debug [libusb_unref_device] destroy device 2.5
[ 0.607922] [00001444] libusb: debug [libusb_unref_device] destroy device 2.6
[ 0.609484] [00001444] libusb: debug [libusb_unref_device] destroy device 2.7
[ 0.611211] [00001444] libusb: debug [libusb_unref_device] destroy device 2.14
[ 0.611918] [00001444] libusb: debug [libusb_unref_device] destroy device 2.8
[ 0.612442] [00001444] libusb: debug [libusb_unref_device] destroy device 2.9
[ 0.613306] [00001444] libusb: debug [libusb_unref_device] destroy device 2.15
[ 0.613667] [00001444] libusb: debug [libusb_unref_device] destroy device 2.16
[ 0.614088] [00001444] libusb: debug [libusb_unref_device] destroy device 2.11
[ 0.614463] [00001444] libusb: debug [libusb_unref_device] destroy device 2.13
[ 0.614909] [00001444] libusb: debug [libusb_unref_device] destroy device 2.10
[ 0.618804] [00001444] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.619352] [00001444] libusb: debug [libusb_unref_device] destroy device 200.2
[ 0.619601] [00001444] libusb: debug [libusb_exit]
[ 0.620049] [00001444] libusb: debug [libusb_exit] destroying default context
[ 0.620407] [00001444] libusb: debug [usbi_remove_pollfd] remove fd 0

mcuee, 2020-06-20:
If that is the case, it is either device problem or usbdk problem. Try WinUSB driver to see if it happens with WinUSB or not.

andrew-mclachlan, 2020-06-20:
Discovered that this is an issue using UsbDk on Windows. On close, the device is disconnected and reconnected (to clear state etc).

The workaround is that if you’re using libUSB backed with UsbDk on Windows then you need to add a Sleep and/or Retries because the device may be reconnecting.

This issue can be closed as not a bug in libUSB, but a gotcha! using it and UsbDk on Windows.

mcuee, 2020-06-24:
Thanks for the updates.

closed 2020-06-24 by mcuee