libusb/libusb #746: Xbox Controller Dongle Issues

skrrrm, 2020-06-21:
Hi,
I am experiencing frequent disconnects between dongle and controllers - using recent versions of [ xow](https://github.com/medusalix/xow), but the issues only arose after a certain commit. I've opened an [issue](https://github.com/medusalix/xow/issues/90) with `xow`, where the developer [suggests](https://github.com/medusalix/xow/issues/90#issuecomment-647159542) that it might be an issue with `libusb`:
> It seems like there's a relation between my switch from the async libusb API to the sync API in commit [78cdb31](https://github.com/medusalix/xow/commit/78cdb31a735bcc0d1b5ce870585c44ce4b55c657) and the problems you are encountering.

***
**System info:**
```
OS:` Manjaro 20.0.3 Lysia
Kernel: x86_64 Linux 5.4.44-1-MANJARO
```
**libusb version:**
`1.0.23-2`

**lspci | grep USB:**
```
02:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] X370 Series Chipset USB 3.1 xHCI Controller (rev 02)
07:00.0 USB controller: ASMedia Technology Inc. ASM1143 USB 3.1 Host Controller
0d:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) USB 3.0 Host Controller
```
I believe I've been using one of the AMD Controllers since the ASMedia controller, would crash the whole USB stack when used with the dongle.

medusalix, 2020-06-21:
Some details (I'm `xow`'s maintainer):

- xow formerly used a mixture of `libusb`'s synchronous and asynchronous APIs for device communication, but a recent architectural change required me to switch to the synchronous API completely
- The USB device's protocol requires reading from two different endpoints at the same time, which shouldn't be a problem even with the sync API when used from multiple threads (please correct me if I'm wrong)

diabolo38, 2020-06-21:
disconnect is rarely caused by s/w unless it cause fault and reboot in the device f/w.
Am experiencing such issue  with some device (most are dwc3 based  otg h/w )  and with  with some  host controler/Cable . It is linked to LPM (Link power management) of USB 3.0 bus. The device h/w if faulty and rarely fail  on link training , but it occur so ofen whith LPM aggressive  host  that it come radily unusable as you start getting exhange.
You may overcome this problem by disabling PM on the root hub controller (if it work often it does nothing)   or try to use a hub in between as they tend to fully filter the power management. it work around the problem.

skrrrm, 2020-06-22:
I'll try do some testing on a hub.

skrrrm, 2020-06-23:
I'm still [experiencing](https://github.com/medusalix/xow/issues/90#issuecomment-648228428) the same type of disconnects with a usb-hub inbetween. I doubt my device is faulty since it's working fine on an older version and in Windows. I'll try around with Power Managment settings.

skrrrm, 2020-06-23:
I've disabled `USB_AUTOSUSPEND` through [TLP](https://linrunner.de/tlp/index.html) - same result.

mcuee, 2020-06-24:
In any case, this is not a libusb problem. This is either a kernel problem or a device problem.

mcuee, 2020-06-24:
libusb is a user space API. Most likely the async API triggered the inherent problem with the device or the kernel USB stack.

mcuee, 2020-06-24:
What you can do is to try a few things to rule out kernel problem.
1) Try to update the kernel.
2) Try to generate usbmon traces and ask in the linux-usb mailing list. They should be able to pinpont the problem.
https://www.kernel.org/doc/Documentation/usb/usbmon.txt

mcuee, 2020-07-01:
Close for now.

closed 2020-07-01 by mcuee