libusb/libusb #750: libusb_bulk_transfer no return!

snikeguo, 2020-07-02:
```
main()
{
libusb_init(ctx);
libusb_get_device_list(ctx, &devs);
int open_close_times=2;//
   while(open_close_times>0)
   {
     
     libusb_open(devs[i], &device_handle);
     libusb_claim_interface(device_handle,0);
     libusb_fill_bulk_transfer(read_transfer,
		device_handle,
		(2) | LIBUSB_ENDPOINT_IN,
		LibUsbRecBuffer,
		LibUsbRecBufferLen,
		LibUsbReadCallback,
		nullptr,
		-1);
     libusb_submit_transfer(Client->read_transfer);
     RunState = true;
     create_thread(PollThreadFunction);
     sleep(100);

   **libusb_bulk_transfer(device_handle,1,data,18,&actual_length,500);//send some data! first time (open_close_times ==2) ,the funtion return ok,everything is ok~.~but second time  (open_close_times ==1) ，the function is NOT return ,Why?**
   
     libusb_cancel_transfer(read_transfer);
     libusb_release_interface(device_handle,0);
     RunState = false;//flag is false ,exit the thread.
     libusb_close(this->device_handle);
     libusb_free_transfer(this->Client->read_transfer);
    open_close_times--;
   }   
}
libusb_free_device_list(devs, 1);//free device list
libusb_exit(ctx);
```
   **libusb_bulk_transfer(device_handle,1,data,18,&actual_length,500);//send some data! first time (open_close_times ==2) ,the funtion return ok,everything is ok~.~but second time  (open_close_times ==1) ，the function is NOT return ,Why?**

snikeguo, 2020-07-02:
[image]

snikeguo, 2020-07-02:
[image]

snikeguo, 2020-07-02:
[image]

diabolo38, 2020-07-02:
You are mixing assync and sync that require some care
 in your poll thread ( that we do not see)  you shall call  appropriate lib fucnt to check for completion and do callback using exclusive flag param to avoid races with the synch method

If the synch xfer doesn't return on time out you likely hitting deadlock due to your callback or assynch poll or above races mentioned races issue

closed 2020-07-08 by snikeguo