libusb/libusb #755: Visual Studio 2019: "error C2011: 'timespec': 'struct' type redefinition" in threads_windows.h

jpnielsen, 2020-07-25:
This library is included in  https://github.com/Ableton/push2-display-with-juce
i have cloned it to make it work with new JUCE and Visual Studio2019, seems Ableton have decided not to support it.
https://github.com/jpnielsen/push2-display-with-juce

When building in VS2019 community edition, i get this error:

`1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(900,28): warning C4099: 'libusb_device_handle': type name first seen using 'class' now seen using 'struct' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\Source\push2\Push2-UsbCommunicator.h(35): message : see declaration of 'libusb_device_handle' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(970,57): warning C4099: 'libusb_device_handle': type name first seen using 'class' now seen using 'struct' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(900): message : see declaration of 'libusb_device_handle' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(1191,23): warning C4099: 'libusb_transfer': type name first seen using 'class' now seen using 'struct' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\Source\push2\Push2-UsbCommunicator.h(34): message : see declaration of 'libusb_transfer' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(1202,74): warning C4099: 'libusb_transfer': type name first seen using 'class' now seen using 'struct' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(1191): message : see declaration of 'libusb_transfer' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(1210,24): warning C4099: 'libusb_transfer': type name first seen using 'class' now seen using 'struct' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\libusb.h(1210): message : see declaration of 'libusb_transfer' (compiling source file ..\..\Source\push2\Push2-Usb-Communicator.cpp)
1>libusb_platform_wrapper.c
1>C:\Users\jpn\source\repos\push2-display-with-juce\modules\libusb\libusb\os\threads_windows.h(40,17): error C2011: 'timespec': 'struct' type redefinition
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.18362.0\ucrt\time.h(39): message : see declaration of 'timespec'
1>Done building project "juce2push2.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========`

I added to \modules\libusb\libusb\os\threads_windows.h

```
line 29
#define HAVE_STRUCT_TIMESPEC
```
and the build finishes without errors

I found some references, but don't know how to fix proper.

> https://stackoverflow.com/questions/33557506/timespec-redefinition-error
> https://groups.google.com/forum/#!topic/kaldi-developers/NHZx4b2JzPs

mcuee, 2020-07-30:
What about Visual Studio 2017? Does this mean that there is a change between VS2017 and VS2019 in terms of timespec? 
(Sorry I only use MinGW under Windows and not VS.)

jpnielsen, 2020-07-30:
https://wimlib.net/forums/viewtopic.php?p=613#p613

It seems timespec was introduced in VS2015
C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h

#ifndef _CRT_NO_TIME_T
struct timespec
{
time_t tv_sec; // Seconds - >= 0
long tv_nsec; // Nanoseconds - [0, 999999999]
};
#endif

jpnielsen, 2020-07-30:
VS2019 Windows Kits\10\Include\10.0.18362.0\ucrt\time.h (head 43 lines)
```
//
// time.h
//
//      Copyright (c) Microsoft Corporation. All rights reserved.
//
// The C Standard Library <time.h> header.
//
#pragma once
#ifndef _INC_TIME // include guard for 3rd party interop
#define _INC_TIME

#include <corecrt.h>
#include <corecrt_wtime.h>

_CRT_BEGIN_C_HEADER

//-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
//
// Types
//
//-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
typedef long clock_t;

struct _timespec32
{
    __time32_t tv_sec;
    long       tv_nsec;
};

struct _timespec64
{
    __time64_t tv_sec;
    long       tv_nsec;
};

#ifndef _CRT_NO_TIME_T
    struct timespec
    {
        time_t tv_sec;  // Seconds - >= 0
        long   tv_nsec; // Nanoseconds - [0, 999999999]
    };
#endif
```

jpnielsen, 2020-07-30:
VS 2017  C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\ucrt\time.h 
(fair use claim, head 43 lines)
```

//
// time.h
//
//      Copyright (c) Microsoft Corporation. All rights reserved.
//
// The C Standard Library <time.h> header.
//
#pragma once
#define _INC_TIME

#include <corecrt.h>
#include <corecrt_wtime.h>

_CRT_BEGIN_C_HEADER

//-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
//
// Types
//
//-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
typedef long clock_t;

struct _timespec32
{
    __time32_t tv_sec;
    long       tv_nsec;
};

struct _timespec64
{
    __time64_t tv_sec;
    long       tv_nsec;
};

#ifndef _CRT_NO_TIME_T
    struct timespec
    {
        time_t tv_sec;  // Seconds - >= 0
        long   tv_nsec; // Nanoseconds - [0, 999999999]
    };
#endif
```

mcuee, 2020-07-31:
I have an installation of VS2017 community edition and it has no issues building libusb.

```
1>------ Build started: Project: getopt, Configuration: Debug Win32 ------
2>------ Build started: Project: libusb-1.0 (dll), Configuration: Debug Win32 ------
3>------ Build started: Project: hotplugtest, Configuration: Debug Win32 ------
4>------ Build started: Project: listdevs, Configuration: Debug Win32 ------
5>------ Build started: Project: stress, Configuration: Debug Win32 ------
6>------ Build started: Project: testlibusb, Configuration: Debug Win32 ------
7>------ Build started: Project: xusb, Configuration: Debug Win32 ------
1>getopt.c
1>getopt1.c
1>Generating Code...
2>core.c
1>getopt_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\lib\getopt.lib
8>------ Build started: Project: fxload, Configuration: Debug Win32 ------
5>stress.c
3>hotplugtest.c
6>testlibusb.c
4>listdevs.c
2>descriptor.c
8>ezusb.c
5>testlib.c
5>Generating Code...
2>hotplug.c
8>fxload.c
7>xusb.c
2>io.c
8>Generating Code...
2>poll_windows.c
2>strerror.c
3>hotplugtest_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\hotplugtest.exe
6>testlibusb_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\testlibusb.exe
4>listdevs_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\listdevs.exe
5>stress_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\tests\stress.exe
2>sync.c
8>fxload_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\fxload.exe
7>xusb_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\xusb.exe
2>threads_windows.c
2>windows_common.c
2>windows_usbdk.c
2>windows_winusb.c
2>Generating Code...
2>   Creating library C:\work\libusb\libusb\msvc\..\Win32\Debug\dll\libusb-1.0.lib and object C:\work\libusb\libusb\msvc\..\Win32\Debug\dll\libusb-1.0.exp
2>libusb_dll_2017.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\dll\libusb-1.0.dll
========== Build: 8 succeeded, 0 failed, 1 up-to-date, 0 skipped ==========
```

mcuee, 2020-07-31:
Just installed VS2019 and no issue either. Please upgrade to the latest VS2019 and latest libusb git and try again.

```
1>------ Build started: Project: libusb-1.0 (static), Configuration: Debug Win32 ------
2>------ Build started: Project: getopt, Configuration: Debug Win32 ------
3>------ Build started: Project: libusb-1.0 (dll), Configuration: Debug Win32 ------
3>core.c
2>getopt.c
1>core.c
2>getopt1.c
3>descriptor.c
2>Generating Code...
1>descriptor.c
3>hotplug.c
2>getopt_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\lib\getopt.lib
1>hotplug.c
3>io.c
1>io.c
3>poll_windows.c
1>poll_windows.c
3>strerror.c
1>strerror.c
3>sync.c
1>sync.c
3>threads_windows.c
1>threads_windows.c
3>windows_common.c
1>windows_common.c
3>windows_usbdk.c
1>windows_usbdk.c
3>windows_winusb.c
1>windows_winusb.c
3>Generating Code...
1>Generating Code...
1>libusb_static_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\lib\libusb-1.0.lib
4>------ Build started: Project: fxload, Configuration: Debug Win32 ------
5>------ Build started: Project: hotplugtest, Configuration: Debug Win32 ------
6>------ Build started: Project: listdevs, Configuration: Debug Win32 ------
7>------ Build started: Project: stress, Configuration: Debug Win32 ------
8>------ Build started: Project: testlibusb, Configuration: Debug Win32 ------
9>------ Build started: Project: xusb, Configuration: Debug Win32 ------
3>   Creating library C:\work\libusb\libusb\msvc\..\Win32\Debug\dll\libusb-1.0.lib and object C:\work\libusb\libusb\msvc\..\Win32\Debug\dll\libusb-1.0.exp
5>hotplugtest.c
4>ezusb.c
8>testlibusb.c
9>xusb.c
3>libusb_dll_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\dll\libusb-1.0.dll
4>fxload.c
4>Generating Code...
6>listdevs.c
7>stress.c
7>testlib.c
5>hotplugtest_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\hotplugtest.exe
7>Generating Code...
9>xusb_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\xusb.exe
8>testlibusb_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\testlibusb.exe
6>listdevs_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\listdevs.exe
4>fxload_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\examples\fxload.exe
7>stress_2019.vcxproj -> C:\work\libusb\libusb\msvc\..\Win32\Debug\tests\stress.exe
========== Build: 9 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
```

mcuee, 2020-07-31:
Reference: check your sdk version as well. This may be related to the Windows SDK version.

There was old report as well. 
https://github.com/libusb/libusb/issues/144

mcuee, 2020-07-31:
I think it is not related to libusb, but rather your solution file settings. But i am not an VS expert so I do not know how to fix your solution.

I will have the same problem if I build your solution.

1) Clone your repo
git clone --recurse-submodules https://github.com/Ableton/push2-display-with-juce.git
2) build with VS2019

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2011	'timespec': 'struct' type redefinition	juce2push2	C:\work\libusb\push2-display-with-juce\modules\libusb\libusb\os\threads_windows.h	40	
```

jpnielsen, 2020-07-31:
i am not an VS expert either, and the workaround is ok.

closed 2020-07-31 by jpnielsen