libusb/libusb #757: libusb_open to return LIBUSB_ERROR_NO_DEVICE

House-Men, 2020-07-27:
libusb Version 1.0.23
windows_winusb.c
static int hid_open(int sub_api, struct libusb_device_handle *dev_handle)
{
....
for (i = 0; i < USB_MAXINTERFACES; i++) {
		if ((priv->usb_interface[i].path != NULL)
				&& (priv->usb_interface[i].apib->id == USB_API_HID)) {
			hid_handle = CreateFileA(priv->usb_interface[i].path, GENERIC_WRITE | GENERIC_READ, FILE_SHARE_WRITE | FILE_SHARE_READ,
				NULL, OPEN_EXISTING, FILE_ATTRIBUTE_NORMAL | FILE_FLAG_OVERLAPPED, NULL);
			/*
			 * http://www.lvr.com/hidfaq.htm: Why do I receive "Access denied" when attempting to access my HID?
			 * "Windows 2000 and later have exclusive read/write access to HIDs that are configured as a system
			 * keyboards or mice. An application can obtain a handle to a system keyboard or mouse by not
			 * requesting READ or WRITE access with CreateFile. Applications can then use HidD_SetFeature and
			 * HidD_GetFeature (if the device supports Feature reports)."
			 */
			if (hid_handle == INVALID_HANDLE_VALUE) {
				usbi_warn(ctx, "could not open HID device in R/W mode (keyboard or mouse?) - trying without");
				hid_handle = CreateFileA(priv->usb_interface[i].path, 0, FILE_SHARE_WRITE | FILE_SHARE_READ,
					NULL, OPEN_EXISTING, FILE_ATTRIBUTE_NORMAL | FILE_FLAG_OVERLAPPED, NULL);
				if (hid_handle == INVALID_HANDLE_VALUE) {
					usbi_err(ctx, "could not open device %s (interface %d): %s", priv->path, i, windows_error_str(0));
					switch (GetLastError()) {
					case ERROR_FILE_NOT_FOUND: // The device was disconnected
						return LIBUSB_ERROR_NO_DEVICE;
					case ERROR_ACCESS_DENIED:
						return LIBUSB_ERROR_ACCESS;
					default:
						return LIBUSB_ERROR_IO;
					}
				}
				priv->usb_interface[i].restricted_functionality = true;
			}
			handle_priv->interface_handle[i].api_handle = hid_handle;
		}
	}
....
}

There are multiple USB Interface
The top index cannot be opened 
Other indexes that can be opened are ignored
Causes a return of LIBUSB_ERROR_NO_DEVICE

mcuee, 2020-07-29:
Please provide more details. Do you encounter real issues here with a real device?

And we list the limitations here.

House-Men, 2020-07-29:
There's a scan box device
Multiple interfaces  One of them couldn't be opened
There is no way to use libusb

Another interface can be opened successfully
Temporarily use CreateFile and readfile to successfully read and solve the problem

Similar issues https://github.com/libusb/libusb/issues/619
It is hoped that libusb can improve multiple interface problems of hid

mcuee, 2020-07-29:
Please post the device descriptors. If this is duplicate of #619, I will mark this as duplicate.

mcuee, 2020-07-29:
Please also take note improving the HID is not the focus of the project. In fact, we may want to remove the support of HID altogether and refer the user to HIDAPI.

closed 2020-07-29 by House-Men