libusb/libusb #758: RFC: remove Windows HID backend and refer users to HIDAPI

mcuee, 2020-07-29:
Now that HIDAPI project is alive and well again, maybe it is time to consider the removal of Windows HID backend to simply the Windows backend.

Reference: 
Some history and discussions here.
https://github.com/libusb/libusb/issues/619

mcuee, 2020-07-29:
https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions

- HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.
 
- Multiple HID top level collections are currently not supported (only the first top level collection will be used).

- The HID Report Descriptors provided by libusb are recomposed and may present minor differences from the actual ones, as the Windows HID API does not allow to read them directly from the device.

- Windows HID API does not support custom Control Transfer, everything needs to be done through report. Please check out the discussion here.

mcuee, 2020-07-29:
Historical discussions back in Jan 2011.
https://marc.info/?t=128347982600002&r=3&w=2

mcuee, 2021-01-30:
Some history and discussions here: #619

> Well, my opinion has always been the following:
> 
> Libusb is meant to be a generic library, so, in the absolute, it should provide agnostic access to as many devices as it can. Which means that if there's a somewhat generic native API that provides access to HID devices, or hubs, or whatever else, then, even if that access is limited, libusb should be designed to provide access to those devices. Now, I understand that some may be tempted to say _"but it's too limited"_, to which I would answer that it doesn't matter, because we do have limits with all the backends anyway (WinUSB does limit how generic an access we can provide too for instance), and, again, the design goal of the library should always to provide as much access to as many devices as we can.
> 
> Which brings us to the second part, the _"as we can"_. Obviously, if we are short of manpower and trying to achieve the goal above is going to add heavily to our workload, it's pretty obvious that we may want to be flexible with the established ideal above, be it by rejecting to add support for specific OSes (eg. older versions of Windows, or niche OSes), specific backends, or removing elements that we estimate are going to become a liability moving on.
> 
> Now, when I was maintainer of the Windows backend, and despite the push in the opposite direction from a specific maintainer, I didn't see much of an issue trying to support HID in the Windows backend, because I **did** have the time to address HID backend issues that were thrown at us, as part of the global Windows backend maintainership. And that is why I did push to keep HID in, because, technically, as long as you keep users aware of the limitations, there's no real problem with having HID in libusb.
> 
> But of course, with the lack of an active Windows backend maintainer, the situation is a bit different now, so it's really up to the active maintainership to decide what they want to do.
> 
> I'll just note that it may be worth estimating if the time and effort that may be spent removing the HID backend might not be greater than the time and effort spent simply fixing this and other HID issues as, if the reason we're removing the HID backend is _"so that we have to waste less time on it"_, and its removal results in current maintainers having to spend more time doing so than fixing the issue reported above, it does seem a bit quite counter productive to me.
> 
> Obviously, if you're going to remove HID, then you're going to have to test, on Windows, that you haven't broken anything (or at least I hope that someone here, which won't be me, will do so), which can't help to keep me a bit puzzled as to why testing a fix for the issue above is seen so drastically difficult.
> 
> Maybe time and effort would be more effectively spent trying to find an active Windows backend maintainer than starting to hack at the Windows backend and removing parts that we don't want to have to maintain. Considering the lack of volunteers for maintaining the Windows backend, I have to wonder how long it's going to be, before we decide that supporting libusb0 or libusbK is no longer worth it, because we have issues that nobody has volunteered to fix accumulating here too. Of course, it would also help if the people who report those issues, and expect the libusb team to fix it, realized that there currently isn't any active Windows maintainer, and that, if they do want something fixed, they **really** need to invest the time submitting Pull Requests in GitHub, rather than leave hints as to what **somebody else** should do.
> 
> So, to reiterate what I said above, and since I think I've made the points I wanted to make clear, I'll leave it entirely to the current maintainership to decide if they want to remove HID, noting that I'm not going to oppose that move if they decide to do so. But I'll also remind them that Open Source activity is often borne out of frustration which means that if libusb Windows users are getting frustrated that their backend is not properly being maintained, someone is bound to take matters in their own hand and step up to become its new maintainer, **eventually**...

mcuee, 2021-01-30:
Comment by Nathan.
> IMHO. It might be time to delete the Windows HID backend. It was controversial when it was added and HID users should be using hidapi anyway. The problem is we do not know how many code rely on the Windows HID support.

mcuee, 2021-01-30:
All in all, the current Windows backend is pretty complicated and there is no other active Windows maintainer than Chris (who also takes care of Linux as of now along with Nathan). So Chris has the say for this topic.

amezin, 2021-02-01:
Does it mean that if I'm writing an application interacting with USB HID device, I have to switch to hidapi (if I want it to run on Windows)? What if I want to use async interface (which, as far as I can tell, hidapi does not have)?

hjelmn, 2021-02-01:
libusb is not meant for HID and never way. It was a point of contention in the libusb community when the HID backend added for Windows.

The problem is that there is a well defined interface for HID devices and OSs (macOS for one) often grab exclusive access to the device. There are other device types that libusb doesn't necessarily support: mass storage, CDC, etc. For HID we strongly recommend that if you want to write a portable application that you use hidapi. It was designed specifically to support HID devices. If it is missing a non-blocking interface you have two options: open a PR to add one, or call into hidapi from another thread.

FYI, the HID windows backend was contentious when it was added. The reasoning for the addition of backend is simple and, IMHO, entirely the fault of Microsoft. With their USB stack it was easier to produce a fake-HID USB device (not really fitting the description of a human interface device) rather than write a Windows driver. This means there are a large number of non-HID HID devices out there. This is not a problem with any other platform as user-space drivers are not uncommon (libusb is a means to write a user-space driver). I joined this project after writing a user-space linux driver for a composite USB device.

mcuee, 2021-02-02:
@amezin: yes it is recommended to use HIDAPI. As for missing features from HIDAPI, you can request for the enhancement in hidapi github page.
https://github.com/libusb/hidapi/issues

mcuee, 2021-07-05:
Close this issue and we will keep HID backend for now, as per the discussion in 
https://github.com/libusb/libusb/issues/741

closed 2021-07-05 by mcuee