libusb/libusb #759: Isochronous transfer returns unsupported with WinUSB, version 1.0.23

csorvagep, 2020-07-29:
Wiki states that after version 1.0.23 WinUSB should support isochronous transfer. But if I set WinUSB as the driver in Zadig, the iso transfer returns with LIBUSB_ERROR_NOT_SUPPORTED. With libusbK transfer works fine.

Libusb was installed with vcpkg and shows version 1.0.23.
Windows version: Windows 10 1909
Compiler: MSVC 16.4.2 - amd64
WinUSB v6.1.7600.16385

What am I missing?

mcuee, 2020-07-29:
https://github.com/libusb/libusb/blob/master/ChangeLog
WinUSB isoc transfer was added in 1.0.23 release.

Please use the official version and not vcpkg and then try again.

mcuee, 2020-07-29:
vcpkg port file reference.
https://github.com/microsoft/vcpkg/tree/master/ports/libusb

csorvagep, 2020-07-29:
The commit ref matches with the 1.0.23 tag in the port file, but I'll check with the official as well.

mcuee, 2020-07-30:
If the official version does not work, then please post the debug log.

csorvagep, 2020-08-04:
I tried the official binary but it did not link, then I compiled the library myself and it works now.

Here's the related part of the log. I don't see much information related to the isochronous transfer, only the `RegisterIsochBuffer isn't available`. It is also interesting, that my code sends an interrupt transfer over 0x02 endpoint before the iso transfer, which appears here as a bulk transfer. I guess this is not a problem just a logging issue or something similar.

I hope it helps.

```
[ 2.054386] [00001fdc] libusb: debug [libusb_alloc_transfer] transfer 000001279D0804A8

[ 2.054437] [00001fdc] libusb: debug [libusb_submit_transfer] transfer 000001279D0804A8
[ 2.054483] [00001fdc] libusb: debug [usbi_add_pollfd] add fd 3 events 4

[ 2.054514] [00001fdc] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 02 with interface 0

[ 2.054541] [00001fdc] libusb: debug [winusbx_submit_bulk_transfer] writing 4 bytes

[ 2.054654] [00001fdc] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!

[ 2.054689] [00001fdc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.054713] [00001fdc] libusb: debug [handle_events] poll fds modified, reallocating

[ 2.054748] [00001fdc] libusb: debug [handle_events] poll() 3 fds with timeout in 100ms

[ 2.056698] [00001fdc] libusb: debug [handle_events] poll() returned 1

[ 2.056767] [00001fdc] libusb: debug [windows_handle_events] checking fd 2 with revents = 0000
[ 2.056797] [00001fdc] libusb: debug [windows_handle_events] checking fd 3 with revents = 0004
[ 2.056826] [00001fdc] libusb: debug [usbi_remove_pollfd] remove fd 3

[ 2.056855] [00001fdc] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 4

[ 2.056885] [00001fdc] libusb: debug [usbi_handle_transfer_completion] transfer 000001279D0804A8 has callback 00007FF60
2504700

[ 2.057549] [00001fdc] libusb: debug [libusb_submit_transfer] transfer 000001279D08CD28

[ 2.057610] [00001fdc] libusb: debug [usbi_add_pollfd] add fd 4 events 4
[ 2.057636] [00001fdc] libusb: debug [winusbx_submit_iso_transfer] matched endpoint 03 with interface 0

[ 2.057661] [00001fdc] libusb: debug [winusbx_submit_iso_transfer] RegisterIsochBuffer isn't available
[ 2.057691] [00001fdc] libusb: debug [usbi_remove_pollfd] remove fd 4

[ 2.058406] [00001fdc] libusb: debug [libusb_cancel_transfer] transfer 000001279D080E68

[ 2.058445] [00001fdc] libusb: debug [winusbx_abort_transfers] will use interface 0
[ 2.058510] [00001fdc] libusb: debug [libusb_release_interface] interface 0

[ 2.058638] [00001fdc] libusb: debug [libusb_close] 

[ 2.058669] [00001fdc] libusb: error [do_close] Device handle closed while transfer was still being processed, but the 
device is still connected as far as we know
[ 2.058701] [00001fdc] libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing
 the device handle
[ 2.058728] [00001fdc] libusb: debug [do_close] Removed transfer 000001279D080E68 from the in-flight list because devic
e handle 000001279D071370 closed

[ 2.058958] [00001fdc] libusb: debug [libusb_exit] 

[ 2.058991] [00001fdc] libusb: debug [libusb_exit] destroying default context
[ 2.059014] [00001fdc] libusb: warning [libusb_exit] some libusb_devices were leaked
[ 2.059036] [00001fdc] libusb: debug [usbi_remove_pollfd] remove fd 0

[ 2.062506] [00000604] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

mcuee, 2020-08-05:
Interesting. So libusb code is good but vcpkg binary is not good. Am I right to say that?

As for the log it is no problem. Interrupt and bulk transfer to libusb is the same.

csorvagep, 2020-08-05:
No, the result is the same in both versions (vcpkg and libusb code). After submitting the transfer the callback returns with status LIBUSB_ERROR_NOT_SUPPORTED. I did not find any difference between them.

I could not link with the precompiled library. Which VS copiler version vas used?

mcuee, 2020-08-11:
Please recompile and not use the pre-compiled binary if it does not work with VS2019.

mcuee, 2020-08-11:
@csorvagep However, in your debug log, the issue is not "LIBUSB_ERROR_NOT_SUPPORTED", rather it is the transfer does not work, right? You need to post more detailed info about your device. A dump of the USB device descriptor will be good.

mcuee, 2020-09-03:
@csorvagep Please update. If there are no updates within the next 10 days I will close this ticket. You can always reopen the ticket when you are able to provide new info.

csorvagep, 2020-09-03:
Sorry for the late response, and thanks for the reminder.

@mcuee If I submit the transfer the `libusb_submit_transfer` returns with error code -12 (LIBUSB_ERROR_NOT_SUPPORTED). But the log does not show any of this, only the `[winusbx_submit_iso_transfer] RegisterIsochBuffer isn't available` message. The libusb_cancel_transfer call at [ 2.058406] is caused by my code, because of the failed submission.

The device is a unique test device for education purposes, it does not belong to any USB Class, here's the dump from xusb:

```
Using libusb v1.0.23.11397

Opening device 0483:5555...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:5555
         bcdDevice: 0200
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 4
   Class.SubClass.Protocol: FF.00.00
       endpoint[0].address: 01
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0004
          polling interval: 0A
       endpoint[2].address: 82
           max packet size: 0004
          polling interval: 0A
       endpoint[3].address: 03
           max packet size: 00C0
          polling interval: 01
```
Endpoint 01 is a bulk out endpoint, endpoint 02 and 82 are interrupt out/in endpoints and 03 is an isochronous out one.

And here is the section of the libusbK log:
```
[ 6.086308] [00004050] libusb: debug [libusb_submit_transfer] transfer 0000017DAEA70FC8

[ 6.086361] [00004050] libusb: debug [usbi_add_pollfd] add fd 4 events 4
[ 6.086385] [00004050] libusb: debug [winusbx_submit_iso_transfer] matched endpoint 03 with interface 0
[ 6.086413] [00004050] libusb: debug [winusbx_submit_iso_transfer] writing 16 iso packets

[ 6.086536] [00004050] libusb: debug [libusb_submit_transfer] transfer 0000017DAEA93B88

[ 6.086580] [00004050] libusb: debug [usbi_add_pollfd] add fd 5 events 4
[ 6.086604] [00004050] libusb: debug [winusbx_submit_iso_transfer] matched endpoint 03 with interface 0

[ 6.086629] [00004050] libusb: debug [winusbx_submit_iso_transfer] writing 16 iso packets

[ 6.086702] [00004050] libusb: debug [libusb_free_transfer] transfer 0000017DAEA80C98

[ 6.086962] [00004050] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!

[ 6.087071] [00004050] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling

[ 6.087151] [00004050] libusb: debug [handle_events] poll fds modified, reallocating

[ 6.087235] [00004050] libusb: debug [handle_events] poll() 4 fds with timeout in 100ms

[ 6.106120] [00004050] libusb: debug [handle_events] poll() returned 1

[ 6.106188] [00004050] libusb: debug [windows_handle_events] checking fd 2 with revents = 0000
[ 6.106217] [00004050] libusb: debug [windows_handle_events] checking fd 4 with revents = 0004
[ 6.106243] [00004050] libusb: debug [usbi_remove_pollfd] remove fd 4
[ 6.106271] [00004050] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 3072

[ 6.106306] [00004050] libusb: debug [usbi_handle_transfer_completion] transfer 0000017DAEA70FC8 has callback 00007FF76
A2A29F5
```

mcuee, 2020-09-07:
Thanks for the updates. I have updated the title to make it a bit clearer.

mcuee, 2020-09-07:
Can you try the fixed in https://github.com/libusb/libusb/pull/749 as well? Not so sure if that helps though.

csorvagep, 2020-11-01:
Sorry, I totally remember as I've answered this, but it seems I didn't.
I was wrong earlier, because the libusb_submit_transfer returns immediately with -12 and not the callback status is set to that value. 
That's why I don't think the mentioned PR would help me, but I'll check it anyway.

pointcloudAI, 2021-03-02:
@csorvagep  Do you fix the problem with -12 (LIBUSB_ERROR_NOT_SUPPORTED). I got the same error when using libusb_claim_interface   and libusb_detach_kernal_driver 
I  am using  libusb-win32 (1.2.6.0) which installed from Zadig -2.4.exe

mcuee, 2021-03-02:
@pointcloudAI You issue is different. Please take note libusb-win32 is a different project. You should not use libusb0.sys with libusb 1.0 Windows. You should use WinUSB driver or libusbk driver when using Zadig.

tormodvolden, 2022-03-29:
I have tested isochronous streaming over both WinUSB and libusbK with 1.0.26-rc1 successfully, using libfreenect and an Xbox 360 Kinect camera. So we can close this bug, "does not work" doesn't apply any longer :)
It can surely be improved but that is the topic of #749 and others. Please file new reports if there are still issues.

closed 2022-03-29 by tormodvolden

mcuee, 2022-03-30:
> I have tested isochronous streaming over both WinUSB and libusbK with 1.0.26-rc1 successfully, using libfreenect and an Xbox 360 Kinect camera. So we can close this bug, "does not work" doesn't [...]

This is great to hear. I do not have good means to test libusb under Windows as my known-good FW are tested using libusbK kBench or Cypress FX3 USB SDK, Cypress FX3 USB SDK is using libusb under Linux and macOS, but not under Windows.

Therefore I was using pyusb but apprarent pyusb does not work properly under Windows for isoc transfer.

mcuee, 2022-03-30:
Let's continue on #749.

I belive libusbk isoc and WinUSB isoc  may have some issues with SuperSpeed isoc transfer. Travis has done some improvement on the libusbk release last year. I will create a new ticket once I figured out how to test that. One possibility is to port Cypress FX3 SDK linux/macOS libusb implementation back to Windows. But that may be too difficult for me -- I believe it can be done but I am just not good at coding.

csorvagep, 2022-03-31:
This was really long ago, but I can try to reproduce with #749