libusb/libusb #761: libusb_get_string_descriptor_ascii() returns always LIBUSB_ERROR_NOT_FOUND

pazourek, 2020-07-30:
The libusb_get_string_descriptor_ascii uses internally a function libusb_control_transfer() which always returns -5 (LIBUSB_ERROR_NOT_FOUND). The wireshark in USB mode showed me there wasn't sent any USB packet (URB).

Tested on WIN10 (MSW 10 Pro, 10.0.18362 N/A Build 18362) with libusb-win32 drivers(1.2.6.0). An old libusb-0.1 works as expected with this driver.
Replacing the libusb-win32 driver by a WinUSB (6.1.7600.16385) driver solved this issue. From my point of view it seems as some kind of incompatibility to the libusb-win32.

My application uses following pseudo-code:
libusb_init(NULL);
libusb_get_device_list(NULL, &_list);
for(all_interested_devices){
  libusb_get_device_descriptor(_libusb_device, &_descriptor)
  libusb_open(_libusb_device, &_handle)
  unsigned char _str[256];  
  libusb_get_string_descriptor_ascii(_handle, _descriptor.iSerialNumber, _str, sizeof(_str));
  libusb_close(_handle);
}
libusb_free_device_list(_list, 1);

pazourek, 2020-07-30:
The tested libusb was in version 1.0.23

mcuee, 2020-07-30:
If WinUSB driver works fine but not libusb0.sys, then this can be considered part of the following ticket. 
https://github.com/libusb/libusb/issues/94

Take note it is not recommended to use libusb0.sys with libusb Windows. It is mentioned in the wiki.
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

Recommended: Use the most recent version of Zadig, an Automated Driver Installer GUI application for WinUSB (recommended), libusb-win32 (not working well, not recommended) and libusbK (only if you hit WinUSB limitations).

mcuee, 2020-07-30:
Still please provide the debug log when running with libusb0.sys (libusb-win32 device driver). Thanks.

pazourek, 2020-07-30:
`libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_2846&PID_7808&MI_03\6&163BC91D&0&0003'
libusb: info [winusb_get_device_list] libusb will not be able to access it
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface`

The log was made together with an USB Composite Device, an interface 2 had got the libusb0 driver installed. 

The issue has also been reproducible with non-Composite USB Device.

pazourek, 2020-07-30:
```
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 1 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 2 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 3 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 4 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 5 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 6 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 7 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 8 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 9 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 10 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 11 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 12 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 13 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 14 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 15 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 16 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 17 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 18 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 19 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 20 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 21 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 22 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 23 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 24 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 25 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 26 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 27 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 28 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 1 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 2 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 3 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 4 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 5 with WinUSB): [87] The parameter is incorrect.
[... cut ...]
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
```
Another log on another PC with  non-Composite USB Device.

mcuee, 2020-07-31:
Try claim the interface first to see if that helps.

pazourek, 2020-07-31:
I tried to claim the interface right after opening the device. Unfortunately it didn't help.
```
libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_2846&PID_7808&MI_03\6&163BC91D&0&0003'
libusb: info [winusb_get_device_list] libusb will not be able to access it
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [winusbx_claim_interface] could not access interface 2: [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
```

diabolo38, 2020-08-01:
what kind of device is this ? they are so many interfaces :( 
Could some of the interfaces be hid keyboard/mice ? that you can''t claim (window system devices)
In such case installing libusbk back end  driver  for the whole device could work, but you wan't get anymore the kbd/mice and stanadrd class interfaces drivers support if any (i have such device and i can still use system device but with custom inf file ).

posting the descriptor could give us some hints

pazourek, 2020-08-02:
The Composite device is proprietary printer (interface 2) combined with bar-code reader (CDC-ACM, iface 0,1). There is also a DFU iface (5) for FW upgade.
The Non-composite device has only iface 0 that is to be used with libusb.

mcuee, 2020-08-03:
For your composite device, since an USB CDC-ACM interface union is unsed, looks like IAD is involved. In that case, there is a limitation from libusb side.

pazourek, 2020-08-06:
Since the non-composite device has the same issue and the libusb0.1 together with libusb0.sys works as expected I don't think I caught the libusb limitation as mentioned.

pazourek, 2020-08-06:
Log with non-composite device with only one interface (nr. 0, vendor specific) . The iface 0 has been claimed right after opening the device.

```
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 1 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 2 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 3 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 4 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 5 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 6 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 7 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 8 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 9 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 10 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 11 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 12 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 13 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 14 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 15 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 16 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 17 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 18 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 19 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 20 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 21 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 22 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 23 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 24 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 25 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 26 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 27 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 28 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 1 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 2 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 3 with WinUSB): [87] The parameter is incorrect.
[... cut ...]
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 28 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [87] The parameter is incorrect.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [87] The parameter is incorrect.
libusb: error [auto_claim] could not auto-claim any interface
```

pazourek, 2020-08-06:
The libusb_claim_interface() itself returns -3 (LIBUSB_ERROR_ACCESS) with a log:
```
libusb: error [winusbx_claim_interface] could not access interface 0: [87] The parameter is incorrect.
```

mcuee, 2020-08-19:
Please post the USB descriptor of your non-composite device. Thanks.

pazourek, 2020-08-19:
USB descriptors dumped by a USBTreeView utility are as follows. The device operate only on Full-speed so a qualifier descriptor reuqest is stalled. 
```
 ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0xA600
idProduct                : 0xE360
bcdDevice                : 0x0100
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "ShiftingControll"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "AutoBlipper"
iSerialNumber            : 0x03 (String Descriptor 3)
 Language 0x0409         : "123456789.123456789.123456789.12"
bNumConfigurations       : 0x01 (1 Configuration)

    ------------------ Configuration Descriptor -------------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x02 (Configuration Descriptor)
wTotalLength             : 0x0019 (25 bytes)
bNumInterfaces           : 0x01 (1 Interface)
bConfigurationValue      : 0x01 (Configuration 1)
iConfiguration           : 0x00 (No String Descriptor)
bmAttributes             : 0xC0
 D7: Reserved, set 1     : 0x01
 D6: Self Powered        : 0x01 (yes)
 D5: Remote Wakeup       : 0x00 (no)
 D4..0: Reserved, set 0  : 0x00
MaxPower                 : 0x32 (100 mA)

        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x00
bAlternateSetting        : 0x00
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0xFF (Vendor Specific)
bInterfaceSubClass       : 0x00
bInterfaceProtocol       : 0x00
iInterface               : 0x00 (No String Descriptor)

        ----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x81 (Direction=IN EndpointID=1)
bmAttributes             : 0x03 (TransferType=Interrupt)
wMaxPacketSize           : 0x0040 (64 bytes)
bInterval                : 0x80 (128 ms)

    ----------------- Device Qualifier Descriptor -----------------
Error                    : ERROR_GEN_FAILURE
```

mcuee, 2021-07-24:
@pazourek I tend to think this issue has been sorted out by 1.0.24 release version. Please help to test again. Thanks.

Ref: https://github.com/libusb/libusb/issues/954

mcuee, 2021-07-24:
If you use libusb0.sys driver in filter mode, then it will mostly not work.
Ref: https://github.com/libusb/libusb/issues/94

mcuee, 2021-07-24:
> @pazourek I tend to think this issue may be a bit better now with 1.0.24 release version or latest git head if you use libusb0.sys in device driver mode. Please help to test again. Thanks.
> 
> Ref: #954

Sorry no, it still does not work with latest git head.

tormodvolden, 2021-10-09:
@pazourek : So that I can better understand your problem with winusbx_claim_interface, can you please apply this patch to latest master and attach (not paste) a debug log? (Make sure you get debug level messages, above I only see info, warning and error.)
[libusb0_filter_debug.patch.txt](https://github.com/libusb/libusb/files/7315597/libusb0_filter_debug.patch.txt)

In #1006 there is OTOH no winusbx_claim_interface problem so there might be multiple issues.

tormodvolden, 2021-10-29:
I wonder if it is an updated libusbk.dll that made things work this time, compared to the original post? The connection is libusb -> libusbk.dll -> libusb0.sys where the latter is the libusb-win32 kernel driver, and every possible combination of the versions adds to a labyrinth of bugs and a small (and not necessary continuous) path of success. 

Maybe you can try upgrading to libusb0.sys 1.2.7.2, and see if everything still is fine, then upgrade libusb in smaller steps (obviously 1.0.24 and latest git, but if you can git bisect it would be great).

tormodvolden, 2021-11-01:
Please try latest git of libusb with the patch from https://github.com/libusb/libusb/pull/1016 applied on top.

mcuee, 2021-11-02:
No issues with libusb0.sys and #1016.

Run log with #1016 using testlibusb.exe example.
```
C:\work\libusb\libusb_test\x64\Release\examples [test1016 +1 ~0 -0 !]> .\testlibusb.exe
Dev (bus 1, device 1): 046D - C534 speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 1, device 5): 8087 - 0AAA speed: 12M
Dev (bus 1, device 2): 047F - C056 speed: 12M
  Manufacturer:              Plantronics
  Product:                   Plantronics Blackwire 3220 Series
Dev (bus 1, device 0): 8086 - A36D speed: 5G
Dev (bus 1, device 3): 0A5C - 5842 speed: 480M
Dev (bus 2, device 0): 8086 - 15DB speed: 5G
Dev (bus 1, device 14): 0925 - 7001 speed: 12M
  Manufacturer:              Lakeview Research
  Product:                   Generic HID
Dev (bus 1, device 4): 0BDA - 58FD speed: 480M
```

Run log with libusb-1.0.24 release (not working for 0925:7001)

```
C:\work\libusb\libusb-1.0.24_bin\examples\bin64> .\testlibusb.exe
Dev (bus 1, device 1): 046D - C534 speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 1, device 5): 8087 - 0AAA speed: 12M
Dev (bus 1, device 2): 047F - C056 speed: 12M
  Manufacturer:              Plantronics
  Product:                   Plantronics Blackwire 3220 Series
Dev (bus 1, device 0): 8086 - A36D speed: 5G
Dev (bus 1, device 3): 0A5C - 5842 speed: 480M
Dev (bus 2, device 0): 8086 - 15DB speed: 5G
Dev (bus 1, device 14): 0925 - 7001 speed: 12M
```

mcuee, 2021-11-02:
@pazourek Please try again with libusb git head and #1016. You can use libusb0.sys 1.2.6.0 release version.

pazourek, 2021-11-02:
I tried to reproduce the issue again by replacing the libusbk by older version (3.0.7.0) instead of the latest one (3.1.0.0). No issue observed.

tormodvolden, 2021-11-02:
If #1016 works, please try #1018 as well.

mcuee, 2021-11-09:
@pazourek Just wondering if you have some time to try #1016 first. If that works, please try #1018 as well. Thanks.

But if you have no time, I will close this issue as the original issue with 1.0.23 can not be reproduced.

mcuee, 2021-11-09:
@pazourek - #1016 has been merged. Please try the latest git head to see if that works. Then try - #1018. Thanks,

pazourek, 2021-11-10:
I made a test of the 32bit build of the latest git head (683e3cf21ed37d4492c20cea8de810c4d95ae8b6). Results:
1) Non-composite device with libusb0.sys (1.2.6.0) driver: Calling the libusb_set_configuration() failed with LIBUSB_ERROR_IO.
```
[40.558450] [00001564] libusb: debug [libusb_set_configuration] configuration 1
[40.559186] [00001564] libusb: debug [libusb_alloc_transfer] transfer 09A0B5E0
[40.559415] [00001564] libusb: debug [libusb_submit_transfer] transfer 09A0B5E0
[40.559415] [00001564] libusb: debug [usbi_add_pollfd] add fd 2 events 4
[40.560415] [00001564] libusb: debug [libusb_claim_interface] interface 0
[40.560415] [00001564] libusb: debug [winusbx_claim_interface] claimed interface 0
[40.560415] [00001564] libusb: debug [windows_assign_endpoints] no endpoints found for interface 0
[40.560415] [00001564] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[40.560415] [00001564] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[40.561412] [00001564] libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [87] The parameter is incorrect.
[40.561412] [00001564] libusb: debug [usbi_remove_pollfd] remove fd 2
[40.562409] [00001564] libusb: debug [libusb_free_transfer] transfer 09A0B5E0`
```

pazourek, 2021-11-10:
2. Non-composite device with winusb.sys (10.0.19041.1) driver: All tests passed.

pazourek, 2021-11-10:
3. Non-composite device with libusb0.sys (1.2.7.2) driver: All tests passed.

pazourek, 2021-11-10:
After these tests, deleting the old drivers from Windows folder and installing the new drivers via Zadig 2.4, I am able to reproduce the original issue.
[dbg_211110.txt](https://github.com/libusb/libusb/files/7515475/dbg_211110.txt)
Non-composite device with libusb0.sys (1.2.6.0)/libusbk.dll (3.0.7.0).

tormodvolden, 2021-11-10:
Please apply this debug patch, run with LIBUSB_DEBUG=4 and attach the /full/ log.
[libusb0_debug3.patch.txt](https://github.com/libusb/libusb/files/7516165/libusb0_debug3.patch.txt)

tormodvolden, 2021-11-10:
For libusb_set_configuration() you must have libusbk.dll >= 3.0.8.0 (to have libusbk commit ad9cacd included). Since 3.1.0.0 has been released you should use this.

pazourek, 2021-11-11:
Non-composite device with libusb0.sys (1.2.6.0)/libusbk.dll (3.0.7.0), libusb from latest git head (683e3cf), the libusb0_debug3.patch.txt applied. For now the latest libusb is not able to open the device at all. The older version cannot read the string descriptors.

[log761_211111_libusb1.0.23.11397.txt](https://github.com/libusb/libusb/files/7518038/log761_211111_libusb1.0.23.11397.txt)
[log761_211111_libusb1.0.24.11675_githead.txt](https://github.com/libusb/libusb/files/7518039/log761_211111_libusb1.0.24.11675_githead.txt)

pazourek, 2021-11-11:
2nd round, run with LIBUSB_DEBUG=4, on WIN10 (MSW 10 Pro, 20H2 Build 19042.1237) as the previous last runs.
[log761_211111_libusb1.0.23.11397_2.txt](https://github.com/libusb/libusb/files/7518165/log761_211111_libusb1.0.23.11397_2.txt)
[log761_211111_libusb1.0.24.11675_githead_2.txt](https://github.com/libusb/libusb/files/7518166/log761_211111_libusb1.0.24.11675_githead_2.txt)

pazourek, 2021-11-11:
The 3rd round, the drivers replaced by libusb0.sys (1.2.6.0)/libusbk.dll (3.1.0.0) via Zadig 2.6.
[log761_211111_libusb1.0.23.11397_3.txt](https://github.com/libusb/libusb/files/7518372/log761_211111_libusb1.0.23.11397_3.txt)
[log761_211111_libusb1.0.24.11675_githead_3.txt](https://github.com/libusb/libusb/files/7518373/log761_211111_libusb1.0.24.11675_githead_3.txt)

mcuee, 2021-11-11:
@pazourek You did not install libusbk correctly as your debug log says that libusbk.dll is not present.
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000038] [0000306c] libusb: debug [libusb_init] created default context
[ 0.000051] [0000306c] libusb: debug [libusb_init] libusb v1.0.24.11675
[ 0.000061] [0000306c] libusb: debug [usbi_add_event_source] add HANDLE 00000104 events 0
[ 0.000076] [0000306c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000077] [0000306c] libusb: debug [usbi_add_event_source] add HANDLE 00000108 events 0
[ 0.000110] [0000306c] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000112] [0000306c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003290] [0000306c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.003395] [0000306c] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.005199] [0000306c] libusb: info [windows_init] UsbDk backend is not available
```

mcuee, 2021-11-11:
If you are having issues with Zadig, you can also use libusbk-inf-wizard. You can install libusbk 3.1.0.0 through the installer here.
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0
Run libusbK-3.1.0.0-setup.exe and then you will have the libusbK-inf-wizard application. It may be more intuitive to use than Zadig,

pazourek, 2021-11-11:
Yes your'e right, I didn't install the libusbk driver because I selected the libusb0.sys in the Zadig. I don't have troubles with Zadig, it reports  that the driver was instaled sucessfully. I supposed if the libusbk is needed for the libusb0.sys it will be installed automatically.
[Zadig_2.6.log](https://github.com/libusb/libusb/files/7518735/Zadig_2.6.log)

pazourek, 2021-11-11:
I replaced the libusb0.sys driver by libusbk and vice versa by Zadig to force the  libusbk installation. After that the device works as expected, no issue observed neither  with libusb1.0.23.11397 nor with latest git head version.

tormodvolden, 2021-11-11:
Then after having installed libusbk (so that libusbk.dll is installed), you can switch driver to libusb0.sys again.

pazourek, 2021-11-11:
Yes, I have done it, now it works as expected. So from my point of view, the root cause of this issue seems to be the missing libusbk.dll which was not installed by Zadig. Is it possible?

mcuee, 2021-11-11:
> Yes, I have done it, now it works as expected. So from my point of view, the root cause of this issue seems to be the missing libusbk.dll which was not installed by Zadig. Is it possible?

But I think Zadig may not include libusbk.dll by default if you install libusb0.sys. It will include libusbk.dll and libusb0.dll if you install libusbk driver.

The easies way is to install libusbk.dll is to use the official release of libusbk.
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0

FYI: I happen to be the non-developer admin of libusb, libusb-win32 and libusbk project...

mcuee, 2021-11-11:
@pazourek But your install log shows that Zadig has libusb0.dll and libusbk.dll included in the driver package.

```
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [wdi_prepare_driver] K driver available - adding the libusbK DLLs to the libusb-win32 inf
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Petr\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Petr\usb_driver\AutoBlipper.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\amd64\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\amd64\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\amd64\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\amd64\libusbk.dll'
libwdi:debug [AddFileHash] 'libusbk_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\amd64\libusbk_x86.dll'
libwdi:debug [AddFileHash] 'autoblipper.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\autoblipper.inf'
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\x86\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\x86\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\x86\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Petr\usb_driver\x86\libusbk.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\Petr\usb_driver\AutoBlipper.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_A600&PID_E360 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_A600&PID_E360 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_A600&PID_E360 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_A600&PID_E360 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_A600&PID_E360 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\Petr\usb_driver\AutoBlipper.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
```

mcuee, 2021-11-11:
@pazourek If you still have issues with Zadig 2.7, raise an issue with libwdi here.
https://github.com/pbatard/libwdi/issues

mcuee, 2021-11-11:
@pazourek BTW, it is confirmed that libusb0.sys 1.2.7.2 snapshot release does not work under Windows 7.
mcuee/libusb-win32#24

mcuee, 2021-11-12:
@pazourek I will close this issue but you can continue the discussion here or in libwdi.

closed 2021-11-12 by mcuee

mcuee, 2021-11-13:
@pazourek BTW, libusb0.sys 1.2.7.3 release might work with Windows 7, please give it a try if you can.
https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20211113/

It will be greatly appreciated that you can provide the feedbacks here if you got the time.
https://github.com/mcuee/libusb-win32/issues/24

mcuee, 2021-11-15:
From https://github.com/mcuee/libusb-win32/issues/24, the new snapshot release works under Windows 7.

Snapshot release including Zadig.
https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.2.7.3