libusb/libusb #765: How to compile with -fPIC on static library?

TomzBench, 2020-08-12:
Hello, I want to compile my libusb.a file with -fPIC but i'm not sure if that is supported?  I am using `--with-pic=ON` However I get these errors when linking:
```
/usr/bin/ld: ../../install/lib/libusb-1.0.a(core.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(descriptor.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(hotplug.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(io.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(strerror.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(sync.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(poll_posix.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(threads_posix.o): relocation R_X86_64_TPOFF32 against `tid.5187' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(linux_usbfs.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../install/lib/libusb-1.0.a(linux_udev.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
```

TomzBench, 2020-08-12:
did not pass arg correctly i think

closed 2020-08-12 by TomzBench

pwmorrison, 2020-11-18:
@TomzBench Did you solve this problem? I'm getting similar errors, when using `--with-pic`.