libusb/libusb #766: Video stream and still photos to iPhone App

Marijusaj, 2020-08-14:
Hello,
 
We are developing an accessorie to iPhone which would stream videos and still images to iOS APP and Android APP

It looks that we are able to compile the libraries and write the wrapper in Android OS.
 
While reading description we found libusb is compatible with MacOS by using Darwin commands backend.
But it is not clear for us does those libraries are compatible with iOS?
Can we use libusb in connection with libuvc to transfer video stream and still photos to iPhone App?
Is it possible to compile them in C++ and make a wrapper with Kotlin?
 
Thank You in advance for Your answer.

mcuee, 2020-08-15:
No. libusb is not compatible with iOS.

closed 2020-08-18 by mcuee

Qamar4P, 2022-03-15:
@osy says it's working with jailbreak iPhone. https://github.com/libusb/libusb/pull/1088#issuecomment-1059927916

@osy  @Marijusaj and @mcuee Did you guys found any "USB info" like app for iOS?

osy, 2022-03-15:
If you compile libusb for iOS, you can run the example apps over SSH.

mcuee, 2022-03-17:
I will say officially libusb is not supported on iOS.

catgod321, 2025-02-27:
> I will say officially libusb is not supported on iOS as of now and probably in a foreseeable future. Majority of the users will not jailbreak there iPhones as well.
> 
> But if you are using a jailbreak iPhone, then you can always try.

Hello, I would like to ask if you still do not support libusb to connect to iOS or iPadOS, so if iOS wants to connect to UVC camera devices, is there a way to refer to it?

mcuee, 2025-02-27:
iOS is not supported.

Even under macOS, libusb and libuvc are not the right library to use for UVC cameras.