libusb/libusb #767: [Errno 5]: USBError Accessing Configurations

jstiefel, 2020-08-17:
I'm facing a problem when trying to get/ set configurations on an Aim TTi CPX400SP lab power supply using PyUSB together with libusb-1.0-0 on Ubuntu 18.04. Executing `dev.set_configuration()` (and also `dev.get_active_configuration())` fails with:

```
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 757, in get_configuration_descriptor
    config, byref(cfg)))
  File "/home/user/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
```
I'm executing the script as sudo, but also printing out the device instance object's bNumConfigurations fails:

```
sudo python3 cpx_test.py 
DEVICE ID 103e:0460 on Bus 003 Address 010 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x2 Communications Device
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x103e
 idProduct              : 0x0460
 bcdDevice              :  0x200 Device 2.0
 iManufacturer          :    0x4 TTi
 iProduct               :    0xc CPX Series PSU
 iSerialNumber          :   0x2a 434839
 bNumConfigurations     :    0x1 USBError Accessing Configurations
```
I first posted this issue on [PyUSB](https://github.com/pyusb/pyusb/issues/316). @jonasmalacofilho helped me to figure out the source of the problem and we are not sure if it is caused by the device itself or libusb.

Here are the Debug outputs:

`dev.set_configuration()`:

```
sudo python3 cpx_test.py
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000001] [000018fe] libusb: debug [libusb_init] created default context
[ 0.000022] [000018fe] libusb: debug [libusb_init] libusb v1.0.21.11156
[ 0.000043] [000018fe] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000059] [000018fe] libusb: debug [op_init] bulk continuation flag supported
[ 0.000065] [000018fe] libusb: debug [op_init] zero length packet flag supported
[ 0.000079] [000018fe] libusb: debug [op_init] sysfs can relate devices
[ 0.000084] [000018fe] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000240] [00001900] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.002617] [000018fe] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.002627] [000018fe] libusb: debug [linux_get_device_address] scan usb3
[ 0.002663] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.002670] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.002674] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.002744] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.002750] [000018fe] libusb: debug [linux_get_device_address] scan 3-2
[ 0.002770] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=7
[ 0.002774] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 7 session_id 775
[ 0.002778] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/7 (session 775)
[ 0.002797] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x26aea50 (3-2) has parent 0x26b33c0 (usb3) port 2
[ 0.002846] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-6 detached: 0
[ 0.002850] [000018fe] libusb: debug [linux_get_device_address] scan 3-6
[ 0.002870] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.002873] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.002877] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.002895] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x2652b40 (3-6) has parent 0x26b33c0 (usb3) port 6
[ 0.002941] [000018fe] libusb: debug [linux_get_device_address] getting address for device: 3-6.1 detached: 0
[ 0.002945] [000018fe] libusb: debug [linux_get_device_address] scan 3-6.1
[ 0.002965] [000018fe] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.002968] [000018fe] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.002972] [000018fe] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.002990] [000018fe] libusb: debug [linux_get_parent_info] Dev 0x2673f40 (3-6.1) has parent 0x2652b40 (3-6) port 1
[... 252 lines cut ...]
[ 0.083404] [00001926] libusb: debug [libusb_close] 
[ 0.083416] [00001926] libusb: debug [usbi_remove_pollfd] remove fd 9
[ 0.083474] [00001926] libusb: debug [libusb_exit] 
[ 0.083485] [00001926] libusb: debug [libusb_exit] destroying default context
[ 0.083490] [00001926] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.083499] [00001926] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.083509] [00001926] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[ 0.083514] [00001926] libusb: debug [handle_events] poll() returned 0
[ 0.083518] [00001926] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.083523] [00001926] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.083528] [00001926] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.083533] [00001926] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.083538] [00001926] libusb: debug [libusb_unref_device] destroy device 4.3
[ 0.083541] [00001926] libusb: debug [libusb_unref_device] destroy device 4.2
[ 0.083546] [00001926] libusb: debug [libusb_unref_device] destroy device 4.1
[ 0.083550] [00001926] libusb: debug [libusb_unref_device] destroy device 3.5
[ 0.083555] [00001926] libusb: debug [libusb_unref_device] destroy device 3.4
[ 0.083559] [00001926] libusb: debug [libusb_unref_device] destroy device 3.3
[ 0.083564] [00001926] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.083568] [00001926] libusb: debug [libusb_unref_device] destroy device 3.7
[ 0.083573] [00001926] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.083576] [00001926] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.083589] [00001926] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 0.083613] [00001928] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

dickens, 2020-08-17:
The problem is highlighted here:

```
[ 0.004809] [00001926] libusb: warning [seek_to_next_config] config length mismatch wTotalLength 67 real 60
[ 0.004813] [00001926] libusb: warning [seek_to_next_config] config length mismatch wTotalLength 67 real 60
[ 0.004818] [00001926] libusb: error [parse_endpoint] short endpoint descriptor read 0/2
[ 0.004821] [00001926] libusb: error [raw_desc_to_config] parse_configuration failed with error -1
```

It seems that the descriptors on this device are broken.  What does `lsusb -v` list for this device?

jstiefel, 2020-08-17:
> The problem is highlighted here:
>
> ```
> [ 0.004809] [00001926] libusb: warning [seek_to_next_config] config length mismatch wTotalLength 67 real 60
> [ 0.004813] [00001926] libusb: warning [...]

```
Bus 003 Device 008: ID 103e:0460  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x103e 
  idProduct          0x0460 
  bcdDevice            2.00
  iManufacturer           4 
  iProduct               12 
  iSerial                42 
  bNumConfigurations      1
Couldn't get configuration descriptor 0, some information will be missing
Couldn't get configuration descriptor 0, some information will be missing
```
Does this mean an issue in the firmware and nothing I can solve?

dickens, 2020-08-17:
> Does this mean an issue in the firmware and nothing I can solve?

This is very likely the case.

It slipped my mind that `lsusb` uses libusb, so we don't have any additional information than when we started.  Could you provide the output of `dmesg` after plugging in the device?

mcuee, 2020-08-18:
@jstiefel Other than dmesg, can you run "sudo lsusb -vvv" to see if that provide more info.

mcuee, 2020-08-18:
@jstiefel If you can not get "lsusb -vvv" works under Linux, you can try a Windows computer to use usbview to get the descriptors.

usbview:
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview

Smaller download -- FTDI version
https://www.ftdichip.com/Support/Utilities.htm#MicrosoftUSBView

mcuee, 2020-08-18:
http://resources.aimtti.com/manuals/CPX400D+DP_Instruction_Manual-Iss13.pdf

From the user manual, it is suppoed to be a CDC-ACM USB serial port device.

mcuee, 2020-08-18:
From the debug log, the device descriptor got problems.
```
[ 0.004809] [00001926] libusb: warning [seek_to_next_config] config length mismatch wTotalLength 67 real 60
[ 0.004813] [00001926] libusb: warning [seek_to_next_config] config length mismatch wTotalLength 67 real 60
[ 0.004818] [00001926] libusb: error [parse_endpoint] short endpoint descriptor read 0/2
[ 0.004821] [00001926] libusb: error [raw_desc_to_config] parse_configuration failed with error -1
```

mcuee, 2020-08-18:
https://github.com/pyusb/pyusb/issues/316

jstiefel, 2020-08-18:
> > Does this mean an issue in the firmware and nothing I can solve?
> 
> This is very likely the case.
> 
> It slipped my mind that `lsusb` uses libusb, so we don't have any additional information than when we started. Could you provide the output of `dmesg` after plugging in the device?

```
[62277.607481] usb 3-2: new full-speed USB device number 10 using xhci_hcd
[62277.760682] usb 3-2: config index 0 descriptor too short (expected 67, got 64)
[62277.760686] usb 3-2: config 1 has an invalid descriptor of length 7, skipping remainder of the config
[62277.760692] usb 3-2: config 1 interface 1 altsetting 0 has 1 endpoint descriptor, different from the interface descriptor's value: 2
[62277.761647] usb 3-2: New USB device found, idVendor=103e, idProduct=0460, bcdDevice= 2.00
[62277.761651] usb 3-2: New USB device strings: Mfr=4, Product=12, SerialNumber=42
[62277.761653] usb 3-2: Product: CPX Series PSU
[62277.761655] usb 3-2: Manufacturer: TTi
[62277.761657] usb 3-2: SerialNumber: 434839
[62277.763396] cdc_acm: probe of 3-2:1.0 failed with error -22
```

> @jstiefel Other than dmesg, can you run "sudo lsusb -vvv" to see if that provide more info.

Gives nearly the same as the output above with `lsusb`:

```
Bus 003 Device 010: ID 103e:0460  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x103e 
  idProduct          0x0460 
  bcdDevice            2.00
  iManufacturer           4 TTi
  iProduct               12 CPX Series PSU
  iSerial                42 434839
  bNumConfigurations      1
Couldn't get configuration descriptor 0, some information will be missing
Couldn't get configuration descriptor 0, some information will be missing
Device Status:     0x0000
  (Bus Powered)
```

dickens, 2020-08-18:
Yes, your device is broken.  Closing this as this isn't a libusb issue.

closed 2020-08-18 by dickens