libusb/libusb #769: Returns error status code <LIBUSB_ERROR_NOT_SUPPORTED> !!!

kiv62, 2020-08-19:
Why at an instance <testlibusb.c> for some devices 
ret = libusb_open (dev, &handle)
returns error status code <LIBUSB_ERROR_NOT_SUPPORTED>, and for others <LIBUSB_SUCCESS>? 

static int print_device (libusb_device *dev, int level)
{
struct libusb_device_descriptor desc;
libusb_device_handle *handle = NULL;
char description [260], string [256];
int ret;
uint8_t i;

	ret = libusb_get_device_descriptor (dev, &desc);
	if (ret <0) {
		fprintf (stderr, "failed to get device descriptor");
		return-1;
	}

	ret = libusb_open (dev, &handle);
	if (LIBUSB_SUCCESS == ret) {
	}
	else {//error!!!
		snprintf (description, sizeof (description), "Err: %04X - %04X", desc.idVendor, desc.idProduct);
	}
...................
}

I use Win7 x64.

mcuee, 2020-08-19:
Please use the mailing list for technical support.
And have you read the wiki and install a supported driver?

kiv62, 2020-08-19:
I created the project in LabWindows_CVI for a test example <testlibusb.c> c connection libusb-1.0.dll, lib from a folder/Win32.

At start produces the following: 
Dev (bus 1, device 0): Err =-12> 8086 - 1E26
Dev (bus 2, device 0): Err =-12> 8086 - 1E2D
Dev (bus 2, device 5): Err =-5> 0403 - 6010
Dev (bus 3, device 1): Err =-12> 0781 - 5567
Dev (bus 2, device 3): A4TECH - USB Device
Dev (bus 2, device 2): Err =-12> 1A40 - 0101
Dev (bus 2, device 4): SIGMACHIP - USB Keyboard
Dev (bus 2, device 1): Err =-12> 8087 - 0024
Dev (bus 1, device 1): Err =-12> 8087 - 0024
Dev (bus 3, device 0): Err =-12> 8086 - 1E31

Prompt, something else can it is required for usage <libusb>?

mcuee, 2020-08-19:
Yes, you have to use WinUSB driver or other supported driver.
https://github.com/libusb/libusb/wiki/Windows#Driver_Installation

diabolo38, 2020-08-20:
Winusb wan't let you open or claim device multiple time à device only libusbk back end let you do that (requires driver insalll or use zadig) 
If you  use appropriate set option for libsub to use usbdk then no specific driver install is required.

kiv62, 2020-08-20:
I have a task to get s / n for flash-disk (SanDisk). For this I tried to use <libusb>
Can you tell me how to do this without installing a device driver, because windows
he already sees it as a device and allows you to work with it.

mcuee, 2020-08-20:
@kiv62 I will close this ticket. This is not a libusb issue. You should not use libusb for this task.
If you want to continue, please ask in libusb mailing list. Thanks.

closed 2020-08-20 by mcuee