libusb/libusb #770: deadlock in handle_iso_completion after uvc_stream_stop

ccdcn, 2020-08-21:
Hi, 
I am running libuvc on my android device.
Sometimes I have a dead lock in
usbi_mutex_lock(&itransfer->lock);  
in entering handle_iso_completion,  after calling uvc_stream_stop,
bcz uvc_stream_stop free the lock in libusb_cancel_transfer right after set running flag = false,
and then the lock seems can't be obtained in handle_iso_completion.
Any idea to fix this ?

diabolo38, 2020-08-21:
Lock come invalid as freed look like  theirs a racy or bad use of the flag 
BTW is the handle event called from the application and in it s for w/o the flags? If so then that is likely  the reason why

mcuee, 2020-08-21:
Please post the code snippets and debug log.

ccdcn, 2020-08-24:
Got it.
I should have report this is issue in libuvc project.
Thanks.

closed 2020-08-24 by ccdcn