libusb/libusb #774: Valgrind error when calling libusb_exit()

TomzBench, 2020-08-24:
I recently pulled from master and noticed a valgrind error started breaking my tests  I reduced my code to simple init/exit and can reproduce with the code below.

```
#include "libusb-1.0/libusb.h"
static void
test_libusb_context(void** context_p)
{
    libusb_context* context;
    libusb_init(&context);
    libusb_exit(context);
}
```
(Summary of the valgrind error follows...)
```
==1975== Conditional jump or move depends on uninitialised value(s)
==1975==    at 0x467D09: handle_events.isra.0 (io.c:2241)
==1975==    by 0x468BE3: libusb_handle_events_timeout_completed (io.c:2343)
==1975==    by 0x4651FA: libusb_exit (core.c:2433)
==1975==    by 0x40B242: main (main.c:134)
==1975== 
==1975== Conditional jump or move depends on uninitialised value(s)
==1975==    at 0x467D0D: handle_events.isra.0 (io.c:2250)
==1975==    by 0x468BE3: libusb_handle_events_timeout_completed (io.c:2343)
==1975==    by 0x4651FA: libusb_exit (core.c:2433)
```

LudovicRousseau, 2020-08-24:
Try again with this patch:
```patch
diff --git a/libusb/io.c b/libusb/io.c
index 48e964d..7d1c7ce 100644
--- a/libusb/io.c
+++ b/libusb/io.c
@@ -2189,7 +2189,7 @@ out:
  * doing the same thing. */
 static int handle_events(struct libusb_context *ctx, struct timeval *tv)
 {
-	struct usbi_reported_events reported_events;
+	struct usbi_reported_events reported_events = {0};
 	int r, timeout_ms;
 
 	/* prevent attempts to recursively handle events (e.g. calling into
```

TomzBench, 2020-08-24:
Yes this patch works

commit 83772692 (2020-08-24): core: Ensure that reported event bits are initialized

Closes #774
closed 2020-08-24 by dickens