libusb/libusb #776: WIN32: Crash when async transfer is active when device is removed.

a-v-s, 2020-08-27:
libusb version 1.0.23 crashes on the WIN32 platform when using the asynchronous API, and a transfer is active while the USB device is unplugged.

I have created a sample to trigger the crash https://github.com/a-v-s/libusb-win32-async-crash

This sample runs libusb_handle_events_completed in a thread, initiated a receive transfer and a transmit transfer.
When data has been received, this data will be transmitted back to the device. The device does the same, thus creating a loop of sending and transmitting. This will cause there will always be a transmission active. Unplugging the device will trigger a crash in libusb. The crash occurs in windows_winusb.c line 1890 (winusb_get_transfer_fd) 

The example can be accompanied by running https://github.com/a-v-s/ucdev/tree/master/demos/usbd/stm32f103xB on a Blue Pill.

dickens, 2020-08-27:
Please try the latest source from Github.  The Windows code has recently
been refactored.

a-v-s, 2020-08-31:
I have made a build of the latest head (4b64ecc). Using Microsoft Visual Studio 2019, using the Debug x64 settings.   (Microsoft C/C++ Optimizing Compiler Version 19.25.28614 for x64) 

Using this build I am still getting crashes. (Read Access Violation)

`io.c line 1384, in the function arm_timer_for_next_timeout , **cur_ts** was 0xFFFFFFFFFFFFFFFF`
This occurs most of the time. This is called from the libusb_handle_events_completed thread.
This occurs when the device has been removed, but before I get a DBT_DEVICEREMOVECOMPLETE callback from Windows.

`io.c line 1746. in the function usbi_signal_transfer_completion, **dev_handle** was 0xFFFFFFFFFFFFFFBF`
This is called from windows_iocp_thread. This rarely occurs

`io.c line 1438, in the function add_to_flying_list, **cur_ts** was 0xFFFFFFFFFFFFFFFF`
occurs when trying to send data to the removed device.

a-v-s, 2020-08-31:
Addidional, I have received a transfer callback with what appears to be a corrupted transfer:

```
transfer	0x00000264e0470a98 
	dev_handle		0x0000000000000000
	flags			101 'e'	
	endpoint		0 '\0'	
	type			110 'n'	
	timeout			4718695	
	status			1152616060	
	length			25345	
	actual_length	-532244816	
	callback		0x00000264e046fd10	
	user_data		0x004d002c00430054
	buffer			0x006f007200630069
	num_iso_packets	7274611	
	iso_packet_desc	0x00000264e0470ad4 

```

dickens, 2020-09-01:
This seems likely to be caused by the transfer being freed or stomped on while in flight.  Can you reproduce this with debug logging enabled?

dickens, 2020-09-11:
Ping.  I'm trying not to let issues get stale or forgotten.

a-v-s, 2020-09-11:
My apologies, I've been put on a different task at work.
However, I believe I have found an issue in the example code I provided, causing (at least part) of the problem, I've been resubmitting a transfer that got the cancelled status.

mcuee, 2020-09-15:
In this case, we will close the issue for now.

closed 2020-09-15 by mcuee