libusb/libusb #777: Misc: To release 1.0.25

mcuee, 2020-09-07:
Windows hotplug support may be delayed to 1.0.25

mcuee, 2021-01-16:
With #838 and #831 and a few other issues resolved, probably it is good to prepare for a maintainence release 1.0.25.

mcuee, 2021-01-16:
I think #850 will be fixed soon as well.

mcuee, 2021-01-16:
Pull request $833 should be in as well.
https://github.com/libusb/libusb/pull/833

yorickvP, 2021-02-06:
Hi! Great work fixing these issues! I just ran into #831 and spent a lot of time debugging until someone pointed me to the issue. It might be worth it to release 1.0.25 so nobody ends up using 1.0.24 by accident (it's the default in some distros).

mcuee, 2021-02-07:
I think the following will be good to be included in the 1.0.25 release:
#833 (pending review), #856 (reviewed, pending minor modification of commit message) and #850 (pending review of #851).

tormodvolden, 2021-02-07:
Please consider #847 (pending review of #852) as well, it appears as a regression.

mcuee, 2021-02-09:
Thanks to Chris. #833 and #850 have been closed.

mcuee, 2021-02-09:
@hjelmn Please help to review #852 and then see if you want to consider that for the release. Thanks.

EmptyVisual, 2021-04-14:
@mcuee Can we do a partial maintenance release of the Linux regressions in #856 and #858 whilst we wait for the Darwin issues to be reviewed and merged separately?  Most of us Linux users have been forcibly sitting on 1.0.23.0 for months now because of the bugs in the 1.0.24.0 release.

mcuee, 2021-04-14:
I think #856 needs to be closed for the release to happen.

tormodvolden, 2021-04-14:
+1 for high time to get a maintenance release with regression fixes out. I asked about #852 (Darwin) here two months ago because it is relatively trivial, but this is not a regression fix so please don't let it delay any release. Indeed it can seem like the release is hanging on #856, where the suggested fix is a massive commit with a mix of code refactoring and actual changes, API changes even. IMO it is not a critical bug fix (it was discussed whether it was a bug at all) so it could wait until after 1.0.25 and bake longer in the tree before going into a release.

mcuee, 2021-04-14:
#856 fixed #858. So I think #856 is still good to be merged. Nathan already reviewed it and he mentioned just needs to change the commit message. So I think it is better to merge it before the release.

mcuee, 2021-04-14:
@dickens Please weigh in as well. Thanks.

mcuee, 2021-06-01:
As far as I am concerned, only #856 needs to be merged and we should be ready for an RC.

mcuee, 2021-06-05:
#856 has been merged.

mcuee, 2021-06-05:
BTW, welcome to join the maintainer team. Right now only Nathan and Chris are actively handling the release.

Release since 2015.
libusb 1.0.24: by Chris on 11 Dec 2020
libusb 1.0.23 by Nathan on on 29 Aug 2019
libusb 1.0.22 by Chris on 25 Mar 2018
libusb 1.0.21 by Nathan on 26 Oct 2016
libusb 1.0.20 by Chris on 14 Sep 2015

mcuee, 2021-06-05:
The main difficulty with libusb is probably on Windows backends. As of today, there are 48 issues and half of them are related to Windows. And one of the main missing feature under Windows is Hotplug. 

But never mind if you do not want to handle Windows, you can still help to handle Linux and/or macOS.

mcuee, 2021-06-06:
For those who are interested to see the 1.0.25 release coming soon, please help to carry out tests on git head. Thanks.

hjelmn, 2021-06-06:
I identify why the Darwin backends is failing the stress test. Will have a fix today.

hjelmn, 2021-06-06:
Identified the issue with Linux and the stress test. Fixing.

hjelmn, 2021-06-06:
Interesting question. I have always assumed this is true but we don't document it. We can call:

```
libusb_set_option(NULL, ...);
```

before the default context is created, right? If not the stress test is wrong. Otherwise I have a fix that will allow this. Will hold off of this change until I get a consensus.

hjelmn, 2021-06-06:
Opened #923 as a prototype. Please comment.

mcuee, 2021-06-07:
#923 seems to be good to me.

mcuee, 2021-06-07:
> I identifed why the Darwin backend is failing the stress test. Will have a fix today.
> 
> The problem is in the ordering in libusb_exit caused by destroying the cached device list when the init count goes to 0. Still not sure I agree with this behavior change. I used a library destructor function for a reason... I can fix the ordering issue but this code is somewhat more fragile than I would like.

Commit 676076c helps macOS stress test to have the same result as Linux and Windows. Then #923 should be able to fix the remaining segfault issues with Stress test.

patstew, 2021-06-07:
Any chance of getting #867 in? And my #312 too if there aren't any objections.

mcuee, 2021-06-08:
@patstew I am fine with #867. But #312 will need review from Chris.

hjelmn, 2021-06-08:
Same here. Do we know if Chris has cycles to take a look?

mcuee, 2021-06-08:
@hjelmn Looks like Chris is busy. We can take the risk to merge #312 if we do not hear from Chris. I think the risk is not high.

hjelmn, 2021-06-08:
I think we can take a risk on #312. As I mentioned in that PR it is a pretty big missing feature in the backend.

mcuee, 2021-06-08:
@hjelmn I agree with you. Let's merge #312 as well.

hjelmn, 2021-06-08:
Done.

hjelmn, 2021-06-08:
I can't comment at all on #867 but it seems safe. We could include it and see if any issues come up in the RC. I don't know how much testing it will get though.

mcuee, 2021-06-08:
#867 should be safe.

hjelmn, 2021-06-08:
Done. Will merge my PR as well then I think we are good to go.

hjelmn, 2021-06-08:
Ok, ready to test again.

mcuee, 2021-06-08:
Great. I will test the current git head again this evening.

mcuee, 2021-06-08:
I tend to think #924 may not be a big issue and should not prevent RC1. So we are probably good to go for RC1 now.

mcuee, 2021-06-08:
But if we want to take a bit more risks, I am also fine with #895, #900 and #925. Tormod Volden has reviewed #895 and he is the author of #900 and #925. I think he knows libusb pretty well and the fixes all seem to be reasonable to take.

mcuee, 2021-06-08:
@hjelmn The other common complaints about libusb is for Android which I can not test myself due to lack of knowledge and equipment. But I think #830 and #874 are in the right direction to improve Android. #874 is apparently not ready to be merged. As for #830 I think we at least can merge the improved README file for Android which does not hurt anything.

mcuee, 2021-06-08:
As for #912, I found it to be there already in 1.0.24 and libusbk project is now considered unmaintained, so I am okay with not fixing it in 1.0.25. We will encourage people to use WinUSB whenever possible.

mcuee, 2021-06-08:
Looks like #925 needs some more discussions.

tormodvolden, 2021-06-08:
- #896 I commented on it now. It sounds fine, but the code details are quite hard to review. I am sure the author understands it well and tested it though.
- #900 Would be nice to have @dickens' blessing, but I think it is a safe improvement.
- #925 The pull request is safe. The question is is whether to already introduce a new option name, and obsolete the old. This just get worse the longer we leave the old option name in and supported.
- #852 Is also ready from my side, @hjelmn MIA? Also a safe improvement IMO.

It seems like the prospect of an RC is bringing up activity on a lot of long-time, low-hanging fruit, and we should profit from the momentum to get this through. With the fire-fighting after #856 we should probably let it cool down a couple weeks before final release though.

mcuee, 2021-06-08:
@tormodvolden Thanks a lot for the help. I am fine with #852 as well.

And yes, after RC1 we will sure have two weeks or more cool down time before release.

mcuee, 2021-06-08:
@hjelmn @tormodvolden I have narrowed down #924 on external hub. Hopefully you can take a look and find something from the run logs.

hjelmn, 2021-06-08:
@mcuee I will take a look. I have a couple of external hubs I can test with. Will be later today or in the evening.

mcuee, 2021-06-09:
@hjelmn Looks like #924 may already be present in 1.0.24 and may not be a common issue. So I think it is okay to proceed with RC1 without resolving it.

mcuee, 2021-06-09:
@hjelmn @tormodvolden Please take a look at #912 to see if the test code is valid or not. It is related to Windows and libusbk driver, but you do not need to care for Windows, just to see if the example code is valid or not.

mcuee, 2021-06-09:
With regard to #830, I think it is safe to include the updated README file but not the example.

mcuee, 2021-06-10:
BTW, I have done some simple tests under Linux, macOS and Windows, it seems to work fine other than the issues raised here.

I also tested using libusbdotnet Benchmarkcon example under Linux, macOS and Windows, it indirectly test libusb as well and it works fine.

hjelmn, 2021-06-11:
#924 should be fixed now.

hjelmn, 2021-06-11:
Please test and ensure I didn't break anything.

mcuee, 2021-06-11:
#900 and #852 have been merged. I will test the latest git under macOS and Linux. It seems to work fine under Windows 10.

mcuee, 2021-06-11:
So remaining question is #896 review and #925 improvement.

mcuee, 2021-06-11:
Somehow #924 becomes a different issue now with the latest git.

mcuee, 2021-06-12:
#929 can be merged as well. Somehow it was missed in 1.0.23 release due to a misunderstanding.

mcuee, 2021-06-12:
#924 is now resolved.

mcuee, 2021-06-17:
Now only #929 and #927 and #935 need to be merged.

mcuee, 2021-06-23:
@hjelmn Just wondering if you have got the time to merge the above and then create RC1. Thanks.

mcuee, 2021-06-24:
I have done further testing of #912 and it seems to be a bit complicated. Since it is already in 1.0.24 and only concerns libusbk and limited test cases, I will say just live with it unless Chris or others got the time to look into it.

hjelmn, 2021-06-24:
Been a bit busy the last couple of weeks. Will take a look today.

[50 more replies]

closed 2022-01-31 by tormodvolden