libusb/libusb #778: `darwin_handle_transfer_completion` crash when plug & unplug.

miniLV, 2020-09-08:
Thread 14 Crashed:: Dispatch queue: com.xxx
0   com.ringcentral.rooms         	0x0000000109ba5769 darwin_handle_transfer_completion + 103
1   com.ringcentral.rooms         	0x0000000109babd68 handle_events + 992
2   com.ringcentral.rooms         	0x0000000109bab720 libusb_handle_events_timeout_completed + 262
3   com.ringcentral.rooms         	0x0000000109ba0a1f libusb_exit + 230
4   com.ringcentral.rooms         	0x0000000109baa6e5 uvc_exit + 48

mcuee, 2020-09-09:
Please post the debug log by setting LIBUSB_DEBUG=4.

dickens, 2020-09-11:
This would only happen if the transfer is no longer valid (e.g. use-after-free, memory corruption).

miniLV, 2020-09-14:
This problem may occur after multiple **hot-plugging** of the camera device, and the probability is very low.
I suspect that the device is still in use after hot plugging.

I used work around, before entering the assert, I judged in advance and returned.

closed 2020-09-18 by miniLV