libusb/libusb #779: Handling of non compliant usb devices

fesc2000, 2020-09-09:
Hi,

i try to operate a USB audio DAC with libusb. The code is actually working in general, but a specific DAC (Dayton DAC01) is giving me problems. It starts with linux dmesg reporting this:

~~~
[180160.276589] usb 1-3.3: new full-speed USB device number 62 using xhci_hcd
[180160.397837] usb 1-3.3: config 1 has an invalid interface number: 3 but max is 2
[180160.397849] usb 1-3.3: config 1 has an invalid interface number: 3 but max is 2
[180160.397857] usb 1-3.3: config 1 has an invalid interface number: 3 but max is 2
[180160.397865] usb 1-3.3: config 1 has an invalid interface number: 3 but max is 2
[180160.397873] usb 1-3.3: config 1 has no interface number 2
[180160.415296] usb 1-3.3: New USB device found, idVendor=262a, idProduct=9023
[180160.415309] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[180160.415317] usb 1-3.3: Product: DAC01\xc3\xbf\xc3\xbf\xc3\xbf\xc3\xbf\xc3\xbf\xc3\xbf\xc3\xbf\xc3\xbf
[180160.415325] usb 1-3.3: Manufacturer: DAC01
~~~

After some searching i found this is a spec violation:  
From "Universal Serial Bus Specification Revision 2.0" section "9.2.3 Configuration":  
Interfaces are numbered from zero to one less than the number of concurrent interfaces supported by the configuration.

It has been discussed on the LKML some years ago. So basically the device works with ALSA (which i can't use for several reasons) or under windows, but not with libusb. I started modifying code in descriptor.c, but so far could not get it to work (kernel driver always reports back ENOENT for SUBMITURB ioctl). And i am also no expert in USB protocol. 

So my question is, is it worth adding support for such devices in libusb. Maybe some expert can give a quick assessment what would have to be changed.

Thanks,
Felix

mcuee, 2020-09-10:
Do you know how it works under Linux? Do they just add a quirk or things like that?

Personally I think no point to support non-compliant USB devices in libusb, unless you provide the fix and the fix is not intrusive and not affect the proper functions of libusb.

fesc2000, 2020-09-11:
Linux just prints the warning message in the configuration parser and moves on. So it can handle it.

The problem in libusb is that although it parses interface descriptors from 0..bNumInterfaces-1, the actual interface descriptor for this DAC contains a device descriptor with bInterfaceNumber >= bNumInterfaces (which is not spec. compliant).

So even if i increase the size of the allocated usb_interface[] array (and adjust bNumInterfaces) to be able to cope with this, it would break the libusb API in the sense that the usb_interface[] array would now contain holes. In this case, the device has existing devices 0 and 1, then a hole, and device 3.

Apart from that, even if i change my application to deal with it, for some reason the SUBMITURB ioctl fails. 

A solution might be to fake a contiguous usb_interface[] array to an application, and "correct" it for internal communication with the kernel.

If i have time and nerves i might try to fix it and provide a patch. I just thought that maybe someone else hass seen this and has a good idea. But (luckily) such devices seem rare (i already complained to the vendor, but don't expect much).

dickens, 2020-09-11:
I really don't think we should care about buggy/non-compliant devices.  There are too many corner cases to consider.  I'm not in favor of complicating the library to accommodate such devices.

mcuee, 2020-09-15:
In this case, we will close this issue.

closed 2020-09-15 by mcuee