libusb/libusb #780: Loop starvation bug

mildsunrise, 2020-09-10:
I'm using libusb integrated with a custom event loop (Node.js). That is, if USB events arrive fast enough then `libusb_handle_events_timeout` never returns, preventing the rest of the event loop tasks from running.

I still need to verify, but this code looks like the culprit, `reap_for_handle` is repeatedly called in an infinite loop:

https://github.com/libusb/libusb/blob/4b64ecc6409ff3533b00153c65e02099c6ca55a9/libusb/os/linux_usbfs.c#L2671-L2673

dickens, 2020-09-11:
I suppose if the transfer callback does a lot of work this could happen.  I don't know if this is really a bug.

mildsunrise, 2020-09-11:
this can happen anytime events arrive faster than processed (be it because of heavy callbacks, CPU scarceness or high traffic)

to prevent loop starvation, it's common to place a rough limit on the amount of iterations, see [an example](https://github.com/libuv/libuv/blob/aa4fcc49a29bb55928d5d519c179a1ba3f9d71b7/src/unix/udp.c#L258)

commit 006ca0fb (2020-09-13): Guard against getting stuck while handling events

Alba Mendez (user mildsunrise) reports that a thread can get stuck at a
specific point while handling events, thus preventing other events from
being handled. This change addresses two such points in the code:

  1) When processing completed transfers in handle_event_trigger(), the
     function wll loop for as long as the completed_transfers list is
     not empty. Since the event data lock is released and reacquired for
     each completed transfer, it is possible for the completed_transfers
     list to never be emptied. Address this by cut [... cut]
closed 2020-09-13 by dickens