libusb/libusb #782: libusb_exit Crash at `darwin_handle_transfer_completion`

miniLV, 2020-09-15:
Thread 15 Crashed:: Dispatch queue: com.test.cameracontrol Thread 15 Crashed:: Dispatch queue: com.test.cameracontrol0  

com.test.abc         0x0000000105c43aec darwin_handle_transfer_completion + 1181 

com.test.abc         0x0000000105c4a0f0 handle_events + 9922   

com.test.abc         0x0000000105c49aa8 libusb_handle_events_timeout_completed + 2623   

com.test.abc         0x0000000105c3ed93 libusb_exit + 2304  

com.test.abc         0x0000000105c48a6d uvc_exit + 485  

com.test.abc         0x0000000105c3c1f2 -[CompatibleTestController closeDevice] + 1506   

com.test.abc         0x0000000105c3b2d8 -[CompatibleTestController startWithCameraID:] + 9297   

com.test.abc         0x0000000105c4a980 -[LogitechUVCTestController startWithCameraID:] + 448   

com.test.abc         0x0000000105c3a029 -[CameraTestController initCurrentDevice:] + 1009   

com.test.abc         0x0000000105c39f02 -[CameraTestController prepareCurrentDevice:] + 12310  

libdispatch.dylib              0x00007fff7184f6c4 _dispatch_call_block_and_release + 1211  

libdispatch.dylib              0x00007fff71850658 _dispatch_client_callout + 812  

libdispatch.dylib              0x00007fff71855c44 _dispatch_lane_serial_drain + 59713  libdispatch.dylib              

0x00007fff718565d6 _dispatch_lane_invoke + 36314  libdispatch.dylib             

0x00007fff7185fc09 _dispatch_workloop_worker_thread + 59615  libsystem_pthread.dylib     

0x00007fff71aaaa3d _pthread_wqthread + 29016  libsystem_pthread.dylib        0x00007fff71aa9b77 start_wqthread + 15

hjelmn, 2020-09-15:
Will look into this now.

hjelmn, 2020-09-15:
Is this the same as #778?

miniLV, 2020-09-15:
is different with #778, this crash user don't unplug camera.

hjelmn, 2020-09-15:
It might be a threading issue. Not sure. The only thing I can see in that function that might cause an issue is accessing the transfer OS data. I pushed a commit to check for NULL. Might be worth a try to see what happens.

Can you post a gist of the output of the reproducer with the debug level set to 10? Would help to determine what exactly is going on.

hjelmn, 2020-09-15:
If there is a threading problem it is very possible it is in libusb.

miniLV, 2020-09-16:
It is very difficult for me to use debug mode here, because I use pod to integrate into the project. 

Or do I have a way to add judgment protection before calling `libusb_exit`.

miniLV, 2020-09-16:
I found that this library has a similar issue(https://github.com/jetperch/pyjoulescope/issues/5).

Can the reference of this change help you? 

Maybe it needs a longer sleep time? 

And It seems that the next task was cancelled after the unplugging was processed. 
Maybe it can solve the timeout problem?

miniLV, 2020-09-18:
```		
static int handle_events(struct libusb_context *ctx, struct timeval *tv){
           ...

            /* complete any pending transfers */
		while (ret == 0 && !list_empty(&ctx->completed_transfers)) {
			itransfer = list_first_entry(&ctx->completed_transfers, struct usbi_transfer, completed_list);
			list_del(&itransfer->completed_list);
            
            // add device is close judge.
            if (!ctx->device_close) {
			usbi_mutex_unlock(&ctx->event_data_lock);
			ret = usbi_backend.handle_transfer_completion(itransfer);
			if (ret)
				usbi_err(ctx, "backend handle_transfer_completion failed with error %d", ret);
                usbi_mutex_lock(&ctx->event_data_lock);
            }
    }		
}```

If I add `if (!ctx->device_close)` judge, when call `handle_transfer_completion`.

It may solve the problem I encountered, is it risky to the main function?

mcuee, 2021-01-16:
Just set LIBUSB_DEBUG=4 and then post the debug log.

mcuee, 2021-01-16:
Please also try the latest git.

mcuee, 2021-01-25:
@miniLV Please help to try the latest git and post the debug log. Thanks.

mcuee, 2021-06-05:
No updates from the OP so I will close this for now.
@miniLV Please reopen if you find the time to post the debug log.

closed 2021-06-05 by mcuee