libusb/libusb #787: libusb no response after reopen

benfounder, 2020-09-26:
libusb_init(NULL)
libusb_get_device_list(NULL, &list)

loop:
libusb_open(list, &handle)
libusb_claim_interface(handle, 0)

start_once another thread and loop on libusb_handle_events_timeout_completed() in the new thread;

req = libusb_alloc_transfer(0)
libusb_fill_bulk_transfer(req, ...)
libusb_submit_transfer(req)
wait for callback
libusb_free_transfer(req)

libusb_release_interface(handle, 0)
libusb_close(handle)

goto loop

I followed the above procedure in a program of Windows10. It worked on first loop, but in second loop, no callback got be called in libusb_handle_events_timeout_completed(). I did the test with wireshark dumping and I saw the usb in / out messages sent out and related acks. But callback didn't be called.

So I am wondering where the problem lies in.

mcuee, 2020-09-28:
Usually the issue is with the device. Please post the details of the device and the debug log.

benfounder, 2020-10-01:
@mcuee Thanks for your response.
I found the root cause of problem.  The actual procedure is like this:

```
libusb_get_device_list(NULL, &list)

loop:
libusb_open(list, &handle)
libusb_claim_interface(handle, 0)

start_once another thread and loop on libusb_handle_events_timeout_completed() in the new thread;

req = libusb_alloc_transfer(0)
libusb_fill_bulk_transfer(req, ...)
libusb_submit_transfer(req)
wait for callback
libusb_free_transfer(req)

**req = libusb_alloc_transfer(0)
libusb_fill_bulk_transfer(req, ..., callback,...)
libusb_submit_transfer(req)**

libusb_release_interface(handle, 0)
libusb_close(handle)

goto loop
```
There is a submitted(pending) transfer not canceled before libusb_release_interface() and following libusb_close(). So on the second loop, the reopened interface (same device) would not work. I wonder whether it would be better for libusb_release_interface() to returns an error if there are pending tranfers.

So the  solution  for my application is like this:

```
libusb_get_device_list(NULL, &list)

loop:
libusb_open(list, &handle)
libusb_claim_interface(handle, 0)

start_once another thread and loop on libusb_handle_events_timeout_completed() in the new thread;

req = libusb_alloc_transfer(0)
libusb_fill_bulk_transfer(req, ...)
libusb_submit_transfer(req)
wait for callback
libusb_free_transfer(req)

req = libusb_alloc_transfer(0)
libusb_fill_bulk_transfer(req, ..., callback,...)
libusb_submit_transfer(req)

**libusb_cancel_transfer(req)
wait for callback
libusb_free_transfer(req)**

libusb_release_interface(handle, 0)
libusb_close(handle)

goto loop
```
Once more thing I want to emphasize, is that after libusb_cancel_transfer() you must wait for callback done. If no, the following libusb_free_transfer() may cause   libusb_handle_events_timeout_completed() thread crash. Or if you libusb_free_transfer() inside callback func, the following libusb_release_interface() may release interface before callback done. The callback would never be called, then the transfer still pending there, so that second loop still doesn't work.

@mcuee thanks again for your response.

closed 2020-10-01 by benfounder