libusb/libusb #789: win10, run libusbTest.exe, crash, report "Unhandled exception thrown: write access violation. entry->next was nullptr."

hiquokka, 2020-09-30:
The another debug report : https://github.com/zarthcode/Libusbpp/issues/9.
-------------------------------
working environment:   10.0.19042.541, vs2019 16.7.4,  ZPL Priter: USB\VID_1FC9&PID_2016\, and or USB\VID_154F&PID_1300.
If I run the exe which depend on  a  dll of libusb wrapper,  it crahed. 
 The complier tell: 
Unhandled exception thrown: write access violation. entry->next was nullptr.
and stoped at: rv = libusb_bulk_transfer(g_usb_handle, user_device.bOutEndpointAddress, cmd, cmd_length, &length, 0);
 at the line : entry->prev = head->prev;  compiler report: Exception Unhandled, Unhandled exception thrown: write access violation.entry->**next** was nullptr.
context: 
static inline void list_add_tail(struct list_head *entry,
	struct list_head *head)
{
	entry->next = head;
	entry->prev = head->prev;

	head->prev->next = entry;
	head->prev = entry;
}

the call stack: 
	usb_print_d.dll!list_del(list_head * entry) Line 195	C
 	usb_print_d.dll!handle_event_trigger(libusb_context * ctx) Line 2116	C
 	usb_print_d.dll!handle_events(libusb_context * ctx, timeval * tv) Line 2234	C
 	usb_print_d.dll!libusb_handle_events_timeout_completed(libusb_context * ctx, timeval * tv, int * completed) Line 2341	C
 	usb_print_d.dll!libusb_handle_events_completed(libusb_context * ctx, int * completed) Line 2440	C
 	usb_print_d.dll!sync_transfer_wait_for_completion(libusb_transfer * transfer) Line 49	C
 	usb_print_d.dll!do_sync_bulk_transfer(libusb_device_handle * dev_handle, unsigned char endpoint, unsigned char * buffer, int length, int * transferred, unsigned int timeout, unsigned char type) Line 194	C
 	usb_print_d.dll!libusb_bulk_transfer(libusb_device_handle * dev_handle, unsigned char endpoint, unsigned char * data, int length, int * transferred, unsigned int timeout) Line 278	C
>	usb_print_d.dll!usb_print::USBPrinter::send_cmd(unsigned char * cmd, unsigned int cmd_length) Line 328	C++
 	usb_print_d.dll!usb_print::USBPrinter::send_cmd(const std::string & cmd) Line 314	C++
 	usb_print_d.dll!usb_print::USBPrinter::print(const std::vector<usb_print::print_data_t,std::allocator<usb_print::print_data_t>> & set) Line 447	C++
 	demo_d.exe!test() Line 34	C++
 	demo_d.exe!main() Line 40	C++
 	[External Code]	
 	kernel32.dll![Frames below may be incorrect and/or missing, no symbols loaded for kernel32.dll]	Unknown

---------------
       Then I compilered the libusbpp(https://github.com/zarthcode/Libusbpp) project,  the same errors occured again.  It stoped at:
(libusbi.pp) : entry->next->prev = entry->prev;
      the context(libusbi.h): 
static inline void list_del(struct list_head *entry)
{
	entry->next->prev = entry->prev;
	entry->prev->next = entry->next;
	entry->next = entry->prev = NULL;
}
the call stack: 
>	LibusbTest.exe!list_del(list_head * entry) Line 203	C
 	LibusbTest.exe!handle_event_trigger(libusb_context * ctx) Line 2116	C
 	LibusbTest.exe!handle_events(libusb_context * ctx, timeval * tv) Line 2234	C
 	LibusbTest.exe!libusb_handle_events_timeout_completed(libusb_context * ctx, timeval * tv, int * completed) Line 2341	C
 	LibusbTest.exe!libusb_handle_events_completed(libusb_context * ctx, int * completed) Line 2440	C
 	LibusbTest.exe!LibUSB::TransferImpl::Start() Line 155	C++
 	LibusbTest.exe!LibUSB::Transfer::Start() Line 122	C++
 	LibusbTest.exe!main(int argc, char * * argv) Line 114	C++
 	[External Code]	
 	kernel32.dll![Frames below may be incorrect and/or missing, no symbols loaded for kernel32.dll]	Unknown

hiquokka, 2020-09-30:
The exe crashed only when added a third party dll of libusb wrapped.
If the exe depond on libusb, everything is OK.

mcuee, 2020-10-10:
Then this is not libusb issue. Please contact the wrapper author.

closed 2020-10-10 by mcuee