libusb/libusb #790: Linux does not allow interrupt based transfers after a control transfer.

ulao, 2020-09-30:
I have not had this issue with any Windows system using usb-windows.c but on all version of linux using usb-libusb.c I seem to have this same result. 

I see that when using my USB driver the 

while (!usbInterruptIsReady())
{
usbPoll();
}
usbSetInterrupt((void *) reportBuffer, 8); 
 

persistently returns true after a control transfer in linux. I do not see this in windows. I made sure to release my device after.

err = usbOpenDevice(&dev, VENDOR, NULL, ID, NULL, 1);
	if (err) return 0; else return 1;
	usbCloseDevice(dev);

I can of course continue to receive further control transfers as I'm in a loop that constantly polls my USB but the loop never exists as the interrupt is never ready so my interrupt transfers do not process.

mcuee, 2020-10-10:
You need to provide more details of the device and your code.

ulao, 2020-10-10:
I was able to track down the issue further. Looks like as soon as I run 
usbOpenDevice(&dev, VENDOR, NULL, ID, NULL, 1);
linux stops all traffic and only allows the device for control transfers. 

Bus 001 Device 073: ID 16d0:0d05 MCS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x16d0 MCS
  idProduct          0x0d05
  bcdDevice            5.00
  iManufacturer           1 BLISS-BOX
  iProduct                2 4-PLAY(GP)PORT.2
  iSerial                 3 2000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x00
      (Missing must-be-set bit!)
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength    1260
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)

mcuee, 2020-10-22:
If you use HID device, you can try HIDAPI using the hidraw backend to see if that helps you.

Unfortunately, in order for libusb to work, you have to detach the kernel driver. Linux happens to have that facility so that there are codes using libusb instead of HIDAPI. Still it is better you use HIDAPI for HID device, It has native API for Windows and macOS. It has two backends under Linux -- hidraw and libusb.

If your device is not meant for the kernel HID driver, submit a patch to linux kernel to blacklist it. I have done it last time for PICkit 2.

```
commit f4971031f4acd98423a2903c6517fb3ef1aea8dc
Author: Xiaofan Chen
Date: Tue May 13 17:11:59 2008 +0200

HID: add Microchip PICKit 1 and PICkit 2 to blacklist

Microchip PICkit 1 and PICKit 2 USB Programmers are USB HID class of device but
they are not real HID device. They are now supported by libusb based programs
like the following programs.

pk2 and pk2cmd Linux port:
http://home.pacbell.net/theposts/picmicro/
usb_pickit:
http://tfc.duke.free.fr/pickit.html
usb_pickit original version:
http://charm.cs.uiuc.edu/users/olawlor/projects/2003/microchip/

Therefore it is preferred to blacklist them.

```

mcuee, 2020-12-07:
Close this for now.

closed 2020-12-07 by mcuee