libusb/libusb #791: WinUSB: Functionality broken if there is a "Device Descriptor Request Failed" device in the system

bohdan-tymkiv, 2020-09-30:
The 'libusb_get_device_list' stops enumerating devices and returns -1 if such device is present in the system.
[image]

I've narrowed down the issue to commit https://github.com/libusb/libusb/commit/d21956dc

Simple change of this line 
https://github.com/libusb/libusb/blob/d21956dc3357bb40cde6d47eaf2497caf8d0a2de/libusb/os/windows_winusb.c#L844
to 
```
if (conn_info.ConnectionStatus == NoDeviceConnected || \
    conn_info.DeviceDescriptor.bNumConfigurations == 0) {
```
when applied on top of https://github.com/libusb/libusb/commit/d21956dc makes the issue go away.

However, such change leads to Segmentation fault when applied to current HEAD (https://github.com/libusb/libusb/commit/eee6998395184d87bd8e9c07ce2637caed1207f4).
I suspect that all special handling of LIBUSB_ERROR_NO_DEVICE is broken in current HEAD.

VitaSokhatska, 2020-10-26:
any updates here? We'd like to understand whether it is going to be addressed somehow in the nearest releases

Youw, 2020-11-02:
Confirm: seeing same issue on Windows 10.0.19041.572.
The patch suggested above does work for me too.

Youw, 2020-11-02:
> However, such change leads to Segmentation fault when applied to current HEAD (eee6998).

Don't see any issues with c3deb6dd6440fd467ce12965742f2d7a228b3bbe.

commit 851ff404 (2020-11-10): Windows: ignore all broken devices during enumeration

- explicitly check/log info, when a device has broken descriptor;

Closes #791
Closes #806
closed 2020-11-10 by dickens