libusb/libusb #792: usb4java - LibUsb.open returns Access denied (insufficient permissions) when accessed too slow

tzah4748, 2020-10-08:
I am using usb4java which uses libusb, I've opened a new issue at their GitHub but because the "bug" originates from the LibUsb library i think i will get more help here.

**Environment:**
 - OS: Windows 10
 - Java version:
```
java --version
openjdk 13.0.2 2020-01-14
OpenJDK Runtime Environment (build 13.0.2+8)
OpenJDK 64-Bit Server VM (build 13.0.2+8, mixed mode, sharing)
```
 - Android device: xiaomi-pocophone_f1
 - usb4java version 1.3.0

**Bug description**
I want to detect my Android device's S/N when Debugging is Enabled on the device settings.
When my program manages to access the LibUsb.open()/device fast enough, i am able to get the S/N.
But if i give a 1 second (or even a bit less) before trying to get the S/N i will get an exception of -
`USB error 3: Can't open device Bus 001 Device 060: ID 18d1:4ee7: Access denied (insufficient permissions)`

**Its worth mentioning that i succeed in getting the S/N only when i replug the device and not when its already plugged in.**

**Reproduction**
[DeviceUtils.zip](https://github.com/usb4java/usb4java/files/5340049/DeviceUtils.zip)
**Added DeviceUtils class as an attachment**
```
public static void main(String[] args) throws UsbException {
        // Get the USB services and dump information about them
        final UsbServices services = UsbHostManager.getUsbServices();
        services.addUsbServicesListener(serviceListener());
        while(true){}
}

    public static UsbServicesAdapter serviceListener() {
        return new UsbServicesAdapter() {

            @Override
            public void usbDeviceAttached(UsbServicesEvent event) {
                // Read the string descriptor indices from the device descriptor.
                // If they are missing then ignore the device.
                handleDevice(event.getUsbDevice());
            }

            private void handleDevice(UsbDevice device) {
                if (device.isUsbHub()) {
                    final UsbHub hub = (UsbHub) device;
                    for (Object child : hub.getAttachedUsbDevices()) {
                        handleDevice((UsbDevice) child);
                    }
                } else {
                    DeviceUtils.Platform platform = DeviceUtils.getPlatform(device.getUsbDeviceDescriptor().idVendor());
                    if (platform == DeviceUtils.Platform.Android) {
                        System.out.println("Device platform - " + platform);
                        boolean isDebuggingEnabled = DeviceUtils.isDebuggingEnabled(device);
                        System.out.println("Is Debugging Enabled? " + isDebuggingEnabled);
                        if (isDebuggingEnabled) {
                            try {
                                TimeUnit.SECONDS.sleep(0);
                                // TimeUnit.SECONDS.sleep(1);
                                System.out.println("S/N: " +device.getSerialNumberString());
                            } catch (UsbException | InterruptedException | UnsupportedEncodingException e) {
                                e.printStackTrace();
                            }
                        }
                    }
                }
            }
        };
    }
```

**Expected behavior**
A consistent behavior at least, if the problem is really permissions so why can i still get the S/N when i replug the device?
If the problem is not really permissions i would expect to get the S/N and not get this exception!

Many thanks for everyone who can help

mcuee, 2020-10-10:
You need to post the debug log to see what happens.

mcuee, 2020-10-10:
And if you can try out other OS as well like Linux.

tzah4748, 2020-10-12:
Thanks for the fast answer.

**Attaching additional logs when LIBUSB_DEBUG=3**
The below code is when i start my application.
As mentioned before, replugging the device shows no issues.
```
libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
libusb: info [windows_init] UsbDk backend is not available
USB Service Implementation: usb4java
Implementation version: 1.2.0
Service API version: 1.0.2

0x1038
0x05AC
0x0951
0x1532
0x18D1
Device platform - Android
class: -1 subclass: 66 protocol: 1
Is Debugging Enabled? true
libusb: error [winusbx_open] could not open device \\.\USB#VID_18D1&PID_4EE7#2040C3#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [5] Access is denied.
javax.usb.UsbPlatformException: USB error 3: Can't open device Bus 001 Device 020: ID 18d1:4ee7: Access denied (insufficient permissions)
	at org.usb4java.javax.ExceptionUtils.createPlatformException(ExceptionUtils.java:39)
	at org.usb4java.javax.AbstractDevice.open(AbstractDevice.java:226)
	at org.usb4java.javax.AbstractDevice.getLanguages(AbstractDevice.java:538)
	at org.usb4java.javax.AbstractDevice.getUsbStringDescriptor(AbstractDevice.java:507)
	at org.usb4java.javax.AbstractDevice.getString(AbstractDevice.java:526)
	at org.usb4java.javax.AbstractDevice.getSerialNumberString(AbstractDevice.java:314)
	at org.usb4java.javax.Main$1.handleDevice(Main.java:59)
	at org.usb4java.javax.Main$1.usbDeviceAttached(Main.java:40)
	at org.usb4java.javax.ServicesListenerList.usbDeviceAttached(ServicesListenerList.java:39)
	at org.usb4java.javax.Services.usbDeviceAttached(Services.java:110)
	at org.usb4java.javax.AbstractDevice.setParentUsbPort(AbstractDevice.java:293)
	at org.usb4java.javax.Port.connectUsbDevice(Port.java:79)
	at org.usb4java.javax.Ports.connectUsbDevice(Ports.java:123)
	at org.usb4java.javax.RootHub.connectUsbDevice(RootHub.java:254)
	at org.usb4java.javax.RootHub.connectUsbDevice(RootHub.java:29)
	at org.usb4java.javax.DeviceManager.scanNewDevices(DeviceManager.java:162)
	at org.usb4java.javax.DeviceManager.scan(DeviceManager.java:194)
	at org.usb4java.javax.DeviceManager.scan(DeviceManager.java:281)
	at org.usb4java.javax.DeviceManager$1.run(DeviceManager.java:372)
	at java.base/java.lang.Thread.run(Thread.java:830)
```

**Attaching additional logs when LIBUSB_DEBUG=4**
[LibUsb-log-level-4.txt](https://github.com/libusb/libusb/files/5364276/LibUsb-log-level-4.txt)

mcuee, 2020-10-12:
What if you reinstall the driver using Zadig?
Zadig: https://zadig.akeo.ie/

Please post the USB descriptors as well. 
If your device is a USB composite device, Zadig has the option to install WinUSB as the driver of the whole USB device (including other interfaces, and then you can try again as well).

tzah4748, 2020-10-12:
```
Device Descriptor:
  bLength                 18
  bDescriptorType          1
  bcdUSB                2.10
  bDeviceClass             0 Per Interface
  bDeviceSubClass          0
  bDeviceProtocol          0
  bMaxPacketSize0         64
  idVendor            0x18d1
  idProduct           0x4ee7
  bcdDevice             4.09
  iManufacturer            1
  iProduct                 2
  iSerial                  3
  bNumConfigurations       1
```

I used Zadig to "Downgrade" the WinUSB driver from v10.something to v6.1.something
It seems like it is really working now...

Assume i want to use this code for a very wide range of devices, is this the only way to solve the issue?

mcuee, 2020-10-28:
Is it possible that you use a simple C program to demonstrate the problem and not using usb4java?

diabolo38, 2020-10-28:
what kind of device is that ? can you post the full descriptor including all interlaces  not  just the device one 
reason to get acess denies :
 the interface that could be a system keyboard/mice 
Win PC system rules prevent usage of some device  like Mass Storage , network , serial adapter ...

mcuee, 2021-01-23:
Reference:
https://github.com/usb4java/usb4java/issues/81

mcuee, 2021-01-23:
@tzah4748 Please help provide the debug log by setting LIBUSB_DEBUG=4.

mcuee, 2021-03-12:
Close for now. Please reopen when you are able to provide the info and debug log.

closed 2021-03-12 by mcuee